micro-manager/mmCoreAndDevices #2: Missing BOOST_FILESYSTEM_LIB variable

tomhanak, 2021-04-12:
There are missing some ```.m4``` files for prebuilt Windows Boost libraries, namely:
- ```ax_boost_filesystem.m4```
- ```ax_boost_log.m4```
- ```ax_boost_log_setup.m4```
- ```ax_boost_regex.m4```
- ```ax_boost_timer.m4```

On the other hand there is ```ax_boost_asio.m4``` but prebuilt library is missing.

On Linux, I use in Makefile ```$(BOOST_SYSTEM_LIB)``` and ```$(BOOST_THREAD_LIB)``` but have to use ```-lboost_filesystem``` because ```$(BOOST_FILESYSTEM_LIB)``` variable expands to empty string.

So what Boost libraries are officially allowed to use on Micro-Manager project?

nanthony21, 2021-04-12:
@tomhanak , I'm not sure I understand. For Windows, we build using Ant and a Visual Studio solution so I don't think .m4 files ever get used.

As to your question about which libraries are supported, I'm not sure. I'm guessing that support for libraries in the Make files has been added only as needed over the years.

As a side note, I am currently working towards a cross-platform Meson build system for all of the c++ components which will replace the Visual Studio solution as well as the Autotools files and will hopefully make everything easier to understand. However, there is a lot of work to be done so it probably won't be ready for a long while.

tomhanak, 2021-04-12:
There are some prebuilt Boost binaries for Windows in ```3rdpartypublic``` repo. Because I don't want to build Boost on my own, I use only those available. For example, I'm extending PVCAM adapter now and need to create folder structure. So I've used Boost.Filesystem and everything works on Windows without any change in Visual Studio project. But when building the same code on Linux, I have to link with new library. That's of course expected, so I extended Makefile and added third Boost library the same way as other two by adding ```$(BOOST_FILESYSTEM_LIB)```, but this variable is not available and expands to empty string...

Adding a few more ```.m4``` files that can be downloaded without modifications should be sufficient to fix this inconsistency.

I know about your task on new build system and really appreciate it (although I'd rather see CMake than Meson and keep Visual Studio). Ant and Autotools is a PITA.

nanthony21, 2021-04-13:
I see. It looks like the only place that Boost Filesystem is currently used is in the Andor device adapter. Here is the relevant line from the makefile:
`libmmgr_dal_Andor_la_LIBADD = $(MMDEVAPI_LIBADD) -lboost_filesystem -lboost_date_time`

I'm guessing that better support was just never added to the Autotools system. I might be wrong, but since we hope to move away from Autotools anyway this is something that probably won't be worked on.

Meson can easily be used to generate Visual Studio files so it won't mean that Visual Studio can't be used any more. It just means new device adapter will specify their build process in a `meson.build` file instead of a `.vcxproj` for Windows and a `Makefile` for Linux/Mac.

tomhanak, 2021-04-13:
I know that and have it as workaround. But as you wrote, new build system "won't be ready for a long while"...

In the mean time, all I'm asking you is please:
- download the [```ax_boost_filesystem.m4```](http://git.savannah.gnu.org/gitweb/?p=autoconf-archive.git;a=blob_plain;f=m4/ax_boost_filesystem.m4) from this page https://www.gnu.org/software/autoconf-archive/ax_boost_filesystem.html
- place it to ```m4``` folder
- Add ```AX_BOOST_FILESYSTEM``` to [micro-manager/configure.ac](https://github.com/micro-manager/micro-manager/blob/master/configure.ac) around line 30 (needed until ```mmCoreAndDevices``` is fully detached from ```micro-manager``` build system)

Optionally, do the same for other four libraries from the list above. I'm not sure about ```chrono``` library, should be on the list too, but maybe this one is configured as header-only.

I would create a pull request but the changes are needed in two projects, in ```micro-manager``` and this sub-module, and you would wonder where did I take those ```.m4``` files from.

nanthony21, 2021-04-13:
I just added ax_boost_filesystem.m4 to this repository's m4 folder. I use Windows and don't have a convenient way to test this change so please go ahead and submit a pull request for the changes to https://github.com/micro-manager/micro-manager/blob/master/configure.ac once you've had a chance to verify that it's working.

Also, feel free to just go straight to submitting a pull request in the future if you have a proposed solution for an issue.

tomhanak, 2021-04-26:
I think this issue can be closed because Boost.Filesystem has been fully integrated.

closed 2021-04-26 by tomhanak