micro-manager/mmCoreAndDevices #10: Move MMCorePy_wrap to pymmcore?

marktsuchida, 2021-05-11:
The up-to-date way to use `MMCorePy.i` (with Python >= 3.6) is to run pymmcore (which now has mmCoreAndDevices as a subproject, but does not use the latter's build system, only source code).

The MMCorePy build that remains in this repository is for Python 2.7, but Python 2.7 has been [end-of-life](https://www.python.org/doc/sunset-python-2/) for over a year.

Distributing MMCorePy or pymmcore with the Micro-Manager installer is problematic also because installation via `pip` is much more convenient and because it is hard to correctly follow Python's build requirements (e.g. compiler version) as part of a larger build.

So I propose that we remove MMCorePy from all parts of the build and distribution, and move `MMCorePy.i` to the pymmcore repo. It will be one fewer thing to maintain in this repo's build system. After the change, pymmcore will still have mmCoreAndDevices as a subproject, but will only use the `MMDevice` and `MMCore` sources.

If there are no objections, I'll create a PR (here and in micro-manager). Cc: @nanthony21 @nicost.

nicost, 2021-05-11:
Great plan!

marktsuchida, 2021-05-11:
@nanthony21 Would appreciate it if you could take a look at the two PRs, since you've been deep into the autotools build most recently.

In micro-manager/micro-manager#1233, I did not change the mmCoreAndDevices commit that micro-manager's submodule points to, so I think the two PRs can be merged in either order without breaking the build (unless I made some other mistake).

I realized that I don't know how to update the submodule commits in micro-manager's `master` and `privateMain` without causing a problem. If I update `master` first, that change will get merged into `privateMain`, which would then point to a `main` branch commit of mmCoreAndDevices, losing the private adapters. If I update `privateMain` first (or second, to readd the private adapters), a merge conflict will occur the next time `updatePrivateMain.yml` runs. (Not sure what happens when a merge conflict occurs -- maybe we just manually resolve it and everything will be ok?)

nanthony21, 2021-05-12:
@marktsuchida I am not currently set up to run the AutoTools build and probably won't be able to get that set up in the near future. However, the changes you've made look reasonable to me, I think we should merge this and if there's an issue then the nightly mac build should let us know.

FYI, we have so far not been  worrying about updating the submodule commits. Instead, at build time we use the `--remote` option of `git submodule update` to make sure we are at the most recent commit of the default branch. This works well for our current approach of always using the most recent version of all code. In the future we will hopefully have a more structured approach for release versions of various modules  and we can then do things in a more standard way.

You raise a good point about the potential issue we will run into with the `main`/`privateMain` branches if we start referencing specific submodule commits instead our current approach. I'm still wondering if there is a better way for us to deal with the situation of having public code mixed with private.

nanthony21, 2021-05-12:
I can merge the `mmCoreAndDevices` PR but I don't have the permissions to do it on the main repository. Can someone else merge this?

marktsuchida, 2021-05-12:
I went ahead and merged both.

Regarding the submodule updating -- that strategy works for our build, but not for other people building, of course. Probably can be solved by a better GitHub Action, though not sure off the top of my head.

PR #11 (2021-05-12): Remove MMCorePy
It is now superseded by pymmcore.
Closes #10.

Merge should be coordinated with micro-manager/micro-manager#1233.
closed 2021-05-12 by marktsuchida