micro-manager/mmCoreAndDevices #18: Can not clone  SecretDeviceAdapters  and build micro-manager failed

lwxGitHub123, 2021-06-22:
Dear,
     when I  move Git bash into the mmCoreAndDevices submodule: cd mmCoreAndDevices
Change to the "privateMain" branch: git checkout privateMain ;then, git submodule update --init --recursive --remote
it reject me. can  anybody help me  please ? 
     what's  more; when i build micro-manager,there are many C++ files missing,such as BFApi.h,LightEngineAPI.h,pylon/PylonIncludes.h,PvInterface.h,mvIMPACT_CPP/mvIMPACT_acquire.h,FlyCapture2.h,toupcam.h,NIDAQmx.h,pdl2000.h,ximc.h,USMCDLL.h,TMCLWrapperRS232.h,oasis4i.h,APTAPI.h,PiperApiErrors.h,biostep\EI_SDK 1.0\EagleIceSDK.h,cbw.h,libfli.h,   tl_camera_sdk.h,NIDAQmx.h,obsROE_Device.h,ShamrockCIF.h,ITC18.h,flexmotn.h,Jai_Factory.h,master.h,ALC_REV.h,MexExl.h,olmem.h,sencam.h,atmcd32d.h,atmcd32d.h,LightEngineAPI.h...
      thank you very much for help me .

nicost, 2021-06-22:
Hi @lwxGitHub123, why do you want to switch to the privateMain branch?  That branch contains code that we are legally not allowed to share publicly.  Maybe we did not make this clear in the instructions?

The missing include files are all vendor provided headers, some of which are in our 3rdpartypublic subversion repository, others are in the non-public 3dparty repository.  See: https://micro-manager.org/wiki/Building_Micro-Manager_Device_Adapters#Directory_layout.

lwxGitHub123, 2021-06-23:
> Hi @lwxGitHub123, why do you want to switch to the privateMain branch? That branch contains code that we are legally not allowed to share publicly. Maybe we did not make this clear in the [...]

sorry! it's my fault. I didn't  notice it .

closed 2021-06-23 by nicost