micro-manager/mmCoreAndDevices #21: TriggerScope DAC sequencing does not work on Falling edge

ieivanov, 2021-07-09:
The TriggerScope Volts property can be sequenced, and the user can specify whether the DAC state changes on the Rising or Falling edge of the input TTL signal. Sequencing on the Rising edge works well. Changing Sequence Trigger Edge to Falling has no effect - the triggering still happens on the positive edge.

@nicost could you please look into that too? I think this will be our main mode of operation. I'm happy to test and confirm that everything works well afterwards. Thanks!

nicost, 2021-07-10:
Started looking at this but do not see obvious problems in the code (but still need to actually hook up equipment).  Do you happen to know if the Trigger direction property works correctly for the TTL outputs?

ieivanov, 2021-07-10:
Thank you! I haven't checked that, I'll try and I'll let you know what I find out

nicost, 2021-07-10:
No longer needed, found the bug, which this time was in the firmware;).  The bug was present in both the V3 and V4 code and both are fixed now (although I can not test the V4 code).  Update your firmware with the latest version on Github, and let me know if things now work correct for you.  Bug fix description is here https://github.com/micro-manager/TriggerScopeMM/pull/7

closed 2021-07-10 by nicost

ieivanov, 2021-07-10:
Haha, perfect, thanks! I'll test and confirm that it works on V4 as well.