micro-manager/mmCoreAndDevices #33: Add `bytesInBuffer` to `Serial` API

nanthony21, 2021-08-02:
Providing some way for device adapters to determine how many bytes are in the RX serial buffer would be very useful. Currently, as far as I'm aware, there is no way to determine if a command may be waiting in the buffer is to try to `read`, if nothing is there you end up waiting for the function to timeout before it returns. 

Other useful functionality would be `peek` allowing you too examine the contents of the serial buffer without removing it from the buffer.

Other examples of useful general purpose serial functionality may be found here: https://www.arduino.cc/reference/en/language/functions/communication/serial/

jvieira88, 2023-09-07:
It would be really great to have this feature available.
I'm new to micro manager and I'm not quite sure what's the underlying library used to handle serial comms but the same type of method to check if anything has arrived over serial (or even better, how many bytes are ready to be read) is available in python, c++, c#... Could you please expose this?

For anyone else with a similar problem, I found myself giving the hardware a little breath room after transmitting a message (10 ms using the `Sleep` method), followed by continuously calling the `ReadFromComPort` and always checking the `read` value, until all bytes are read. Wrap this whole thing with a timeout check just to be safe.

marktsuchida, 2023-09-07:
> or even better, how many bytes are ready to be read

This, I think, is the right way to describe it, because it is the only information we have (and buffering should remain an implementation detail).

Many device adapters that call `ReadFromComPort()` have a loop with a timeout, similar to what you describe. Many of them look like the result of copy-paste programming, and I suspect some are subtly buggy. I'm wondering if it would be even more useful to have a `ReadFromComPort()` that blocks (with timeout) until the requested number of bytes have been read.