micro-manager/mmCoreAndDevices #35: Improve interface for DA/AD devices

nicost, 2021-08-02:
From a small group email thread May 2020.  Suggestions from Jon Daniels that make a lot of sense to me:

I like the idea of breaking the interface/device into 2 separate device types, one for analog input and the other for analog output.  In the ASITiger device adapter we have both input and output devices.
 
Instead of simply converting the existing SignalIO interface to output-only, maybe it would be better to create two new device types and keep the existing one as-is for backward compatibility?  Then device adapters which are actively maintained could be converted to use the new device types and the old device type marked as deprecated.
 
I also concur with Nico's suggestion of renaming things but I suggest slightly different names.  How about AOSignal for analog output and then AISignal for analog input?  That matches with the NI convention of AO and AI, and also specifies that it's an analog signal.  I think it would be best to avoid "DA" in the method names; people with electronics background immediately recognize "DA" as "digital to analog" meaning it's an analog output, but that is far from obvious to everyone. 
 
So I am suggesting API methods could be

void setAOSignal(const char* signalILabel, double volt) throw (CMMError);
double getAOSignal(const char* signalILabel); // returns the voltage that was actually set on the device, which could differ from the previously set voltage
double getAOLowerLimit(const char* signalILabel) throw (CMMError);
double getAOUpperLimit(const char* signalILabel) throw (CMMError);

The sequence functions could look like:

long getAOSequenceMaxLength(const char* signalILabel) throw (CMMError);
void startAOSequence(const char* signalILabel) throw (CMMError);
void stopAOSequence(const char* signalILabel) throw (CMMError);
void loadAOSequence(const char* signalILabel,  std::vector<double> voltSequence) throw (CMMError);
 
 
Then with the analog input there would be at minimum the API method (and maybe others aren't needed)
 
double getAISignal(const char* signalILabel); // returns the voltage read from the hardware

jondaniels, 2021-08-02:
I'm on vacation until late Monday Aug 2.  If you require immediate help please contact ***@***.*** and/or ***@***.***  I will read your email when I return.
 
Best Regards,
Jon Daniels
Applied Scientific Instrumentation (ASI)

artmeln, 2022-06-29:
I'd like to revive this discussion and propose an interim solution. It seems to me that modifying the existing DA/AD device prototype (or adding another data acquisition device type) would entail a lot of work, and the details of input/output devices still need to be ironed out. In the meantime, I think it would be useful to add functionality to GenericDevice that would allow it to receive arbitrary data. It can be similar to retrieving a camera image but without specifying what kind of data it is or what to do with it (this would be left to the user to implement in the spirit of keeping the device generic). Specifically, the following calls would be necessary:

CollectData - tell the device to start collecting; after receiving this call the device should start filling its internal buffer with data.
GetDataBufferSize - bytes of data available for reading, should be 0 unless the device is ready to send data.
GetDataBuffer - get n bytes of data from the device where n has been populated by a previous call to GetDataBufferSize
StopDataCollection - stop collecting data
GetDataOverflow - a flag that indicates the internal device buffer overflowed before data could be retrieved
ResetDataOverflow - reset overflow flag

I tried to name these methods so that they are self-explanatory but also can't be confused with calls to other devices that already exist. I think these additions would make GenericDevice more versatile by letting it receive any kind of data but also allowing it to serve as analog input device if need be.

If this seems like a good idea, I'd be happy to implement this change. Since we are talking about modifying an existing device, I think the disturbance to MM code would be minimal. Implementation of these calls by a device adapter can be optional, and the change will be entirely back compatible.

artmeln, 2022-06-30:
Here are a couple of actual devices that illustrate the case:
1. A board for collecting fluorescence lifetime data. The board collects photons for a specified amount of time and produces as an output a set of data points (time on the X axis, number of events on the Y axis).
2. A hardware correlator used in dynamic light scattering and fluorescence correlation spectroscopy. Again, the board collects photons for a predetermined amount of time and produces several sets of data with time on the X axis and correlation values on the Y axis.
3. A generic photon counting card. Each detected photon has a record (arrival time, optical channel, lifetime information, etc), and the card streams hundreds of thousands such records per second. The exact data format depends on the manufacturer of the card.

I think these examples illustrate the need to retrieve data from hardware but I doubt that any one of them warrants a new device category in MM. My understanding was that GenericDevice was reserved for any hardware that did not fit in any other MM device category, and it seemed natural to make GenereicDevice more versatile by giving it capability to receive data. By buffer I meant any data stored in the hardware waiting to be retrieved by the device adapter which would then pass it on to upper layers of software, just the way a camera image is handled.

I believe an AD converter can take advantage of this capability as well since it’s not practical to get data one point at a time from hardware, and right now there is no way in MM to pass chunks of data.

I hope my motivation makes more sense now, I just want a device adapter that will let me get arbitrary data from hardware.

marktsuchida, 2022-06-30:
Thanks for the examples; indeed we were thinking about very different things.

I'm familiar with TCSPC cards and these are all great things to be able to support, but they are not AD devices in the usual sense. If the interface only produces "raw data" without a way for user programs to determine the type or meaning of the data, it will not be easy to write, say, code that requires an analog input device but works with any such device.

I think there is a more urgent need to support hardware-clocked acquisition/generation with "regular" AD/DA devices (voltage, maybe current), and that is what this issue was intended for. I'm not at all against supporting devices that acquire fixed-size 1D arrays (your use cases 1 &2; also possibly spectrometers and oscilloscopes) and devices that produce vendor-specific byte streams (or word streams) (your use case 3), but I think these should be addressed separately -- or at least should not prevent us from having a straightforward interface dedicated to AD/DA devices.

It does make sense to think of the AD device as a special case of a byte (or word) stream acquisition device, but having the latter alone does not satisfy the need for a dedicated AD device type that user code can use knowing that the samples are (convertible to) voltages.

I think I now see why you propose to expand GenericDevice -- it may seem like a nicely simplifying idea from the perspective of implementing device adapters. The problem is that it will then offer no hardware abstraction from the perspective of user (application) code: If I need a voltage input device, am I supposed to check every loaded device and see if it has that capability?

Of course there are ways to make that easier (maybe _all_ devices should be "generic" but implement one or more functional "profiles"), but the current way we have is to have specific device types. I would almost certainly avoid this design if starting over from scratch, but we're stuck with it for now, and I'm opposed to having a mix of very different design principles in the interface in the absence of a convincing plan to unify (we already have plenty of inconsistency problems).

To be clear, I'm not opposed to supporting arbitrary byte streams as long as the interface accomplishes the goal of making it easy to implement and use AD devices. Provided that it doesn't delay us in coming up with a good solution for AD/DA.

marktsuchida, 2022-06-30:
Actually I don't think that's stupid. The only problem is that all existing user code expects properties to be one of the known types, so I'm not sure how we would deal with adding a new one. Next time anybody designs property-like things, the API should be such that it forces user code to handle "unknown" types....

artmeln, 2022-07-01:
Thank you for your responses!

Based on this discussion, I think I’ll concentrate my efforts on the problem of byte streaming for now and not on the separate problem of implementing AD/DA devices (I have a more immediate need for the former and unfortunately I’m less familiar with the latter).

I need to educate myself on MM architecture first, so that I can better contribute to this discussion. If this sounds like a reasonable plan, in a few days I’ll open a new issue to continue this discussion with specific proposals for how byte streaming functionality could be implemented.

marktsuchida, 2022-07-01:
@artmeln That sounds great!
Depending on what that looks like, we might be able to support AD devices as a special case, but feel free to not worry about that for now.