micro-manager/mmCoreAndDevices #37: MMCoreJ doesn't build with Swig 4.0

marktsuchida, 2021-08-02:
This is a continuation of micro-manager/micro-manager#838.

Quoting from the [Swig Changelog](http://www.swig.org/Release/CHANGES), in 4.0.0 there is the following entry:
```
2019-02-28: wsfulton
            [Java] std::vector improvements for types that do not have a default constructor.

            The std::vector wrappers have been changed to work by default for elements that are
            not default insertable, i.e. have no default constructor. This has been achieved by
            not wrapping:

              vector(size_type n);

            Previously the above had to be ignored via %ignore.

            If the above constructor is still required it can be added back in again via %extend:

              %extend std::vector {
                vector(size_type count) { return new std::vector< T >(count); }
              }

            Alternatively, the following wrapped constructor could be used as it provides near-enough
            equivalent functionality:

              vector(jint count, const value_type& value);

            *** POTENTIAL INCOMPATIBILITY ***
```
This suggests that it might be possible to have code that works across Swig versions.

JiangXL, 2021-08-04:
May I ask how it is going with Swig 4.0?

marktsuchida, 2021-08-04:
Let me try to at least make a to-do list for this.

Below are the relevant changes (swig <4 to swig 4) that we need to deal with (generated using [japicmp](https://github.com/siom79/japicmp) and filtered).

They result from two changes to SWIG's Java support (from the changelog linked above):
- 2017-05-26 "Extend from java.util.AbstractList<> and implement java.util.RandomAccess for std::vector wrappers"
- 2019-02-28 "std::vector improvements for types that do not have a default constructor"

Of the API changes, we can accept the ones labeled NEW INTERFACE, MODIFIED SUPERCLASS (from `Object`), NEW CONSTRUCTOR, and NEW METHOD. We can also accept the single NEW CLASS (`StrMap$Iterator`). (But it does mean we should switch to _requiring_ SWIG 4 once we enable it, or else the API will become incompatible between copies of `MMCoreJ.jar` built by different SWIG versions.)

Among the backward-incompatible changes, there is not much we can do about the MODIFIED METHODs, where the return value of `get()` got changed to the boxed types and the return value of `size()` got changed from `long` to `int`. These can break existing user code, but hopefully not that frequently; the only failure mode is a compile error, so people will easily see how their code needs to change.

That leaves us with the REMOVED CONSTRUCTORs and REMOVED METHODs. More about these in my next comment.

```
***! MODIFIED CLASS: PUBLIC mmcorej.BooleanVector  (not serializable)
	+++  NEW INTERFACE: java.util.RandomAccess
	+++  NEW INTERFACE: java.util.List
	+++  NEW INTERFACE: java.util.Collection
	***  MODIFIED SUPERCLASS: java.util.AbstractList (<- java.lang.Object)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) BooleanVector(long)
	+++  NEW CONSTRUCTOR: PUBLIC(+) BooleanVector(int, boolean)
	+++  NEW CONSTRUCTOR: PUBLIC(+) BooleanVector(java.lang.Iterable)
	+++  NEW CONSTRUCTOR: PUBLIC(+) BooleanVector(mmcorej.BooleanVector)
	+++  NEW CONSTRUCTOR: PUBLIC(+) BooleanVector(boolean[])
	---! REMOVED METHOD: PUBLIC(-) void add(boolean)
	+++  NEW METHOD: PUBLIC(+) boolean add(java.lang.Boolean)
	+++  NEW METHOD: PUBLIC(+) void add(int, java.lang.Boolean)
	***! MODIFIED METHOD: PUBLIC java.lang.Boolean (<-boolean) get(int)
	+++  NEW METHOD: PUBLIC(+) java.lang.Boolean remove(int)
	+++  NEW METHOD: PROTECTED(+) void removeRange(int, int)
	---! REMOVED METHOD: PUBLIC(-) void set(int, boolean)
	+++  NEW METHOD: PUBLIC(+) java.lang.Boolean set(int, java.lang.Boolean)
	***! MODIFIED METHOD: PUBLIC int (<-long) size()
***! MODIFIED CLASS: PUBLIC mmcorej.CharVector  (not serializable)
	+++  NEW INTERFACE: java.util.RandomAccess
	+++  NEW INTERFACE: java.util.List
	+++  NEW INTERFACE: java.util.Collection
	***  MODIFIED SUPERCLASS: java.util.AbstractList (<- java.lang.Object)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) CharVector(long)
	+++  NEW CONSTRUCTOR: PUBLIC(+) CharVector(int, char)
	+++  NEW CONSTRUCTOR: PUBLIC(+) CharVector(mmcorej.CharVector)
	+++  NEW CONSTRUCTOR: PUBLIC(+) CharVector(java.lang.Iterable)
	+++  NEW CONSTRUCTOR: PUBLIC(+) CharVector(char[])
	---! REMOVED METHOD: PUBLIC(-) void add(char)
	+++  NEW METHOD: PUBLIC(+) boolean add(java.lang.Character)
	+++  NEW METHOD: PUBLIC(+) void add(int, java.lang.Character)
	***! MODIFIED METHOD: PUBLIC java.lang.Character (<-char) get(int)
	+++  NEW METHOD: PUBLIC(+) java.lang.Character remove(int)
	+++  NEW METHOD: PROTECTED(+) void removeRange(int, int)
	---! REMOVED METHOD: PUBLIC(-) void set(int, char)
	+++  NEW METHOD: PUBLIC(+) java.lang.Character set(int, java.lang.Character)
	***! MODIFIED METHOD: PUBLIC int (<-long) size()
***! MODIFIED CLASS: PUBLIC mmcorej.DoubleVector  (not serializable)
	+++  NEW INTERFACE: java.util.RandomAccess
	+++  NEW INTERFACE: java.util.List
	+++  NEW INTERFACE: java.util.Collection
	***  MODIFIED SUPERCLASS: java.util.AbstractList (<- java.lang.Object)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) DoubleVector(long)
	+++  NEW CONSTRUCTOR: PUBLIC(+) DoubleVector(mmcorej.DoubleVector)
	+++  NEW CONSTRUCTOR: PUBLIC(+) DoubleVector(int, double)
	+++  NEW CONSTRUCTOR: PUBLIC(+) DoubleVector(double[])
	+++  NEW CONSTRUCTOR: PUBLIC(+) DoubleVector(java.lang.Iterable)
	---! REMOVED METHOD: PUBLIC(-) void add(double)
	+++  NEW METHOD: PUBLIC(+) boolean add(java.lang.Double)
	+++  NEW METHOD: PUBLIC(+) void add(int, java.lang.Double)
	***! MODIFIED METHOD: PUBLIC java.lang.Double (<-double) get(int)
	+++  NEW METHOD: PUBLIC(+) java.lang.Double remove(int)
	+++  NEW METHOD: PROTECTED(+) void removeRange(int, int)
	---! REMOVED METHOD: PUBLIC(-) void set(int, double)
[... 54 lines cut ...]
	+++  NEW METHOD: PUBLIC(+) void add(int, java.lang.String)
	+++  NEW METHOD: PUBLIC(+) java.lang.String remove(int)
	+++  NEW METHOD: PROTECTED(+) void removeRange(int, int)
	***! MODIFIED METHOD: PUBLIC java.lang.String (<-void) set(int, java.lang.String)
	***! MODIFIED METHOD: PUBLIC int (<-long) size()
***! MODIFIED CLASS: PUBLIC mmcorej.UnsignedVector  (not serializable)
	+++  NEW INTERFACE: java.util.RandomAccess
	+++  NEW INTERFACE: java.util.List
	+++  NEW INTERFACE: java.util.Collection
	***  MODIFIED SUPERCLASS: java.util.AbstractList (<- java.lang.Object)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) UnsignedVector(long)
	+++  NEW CONSTRUCTOR: PUBLIC(+) UnsignedVector(java.lang.Iterable)
	+++  NEW CONSTRUCTOR: PUBLIC(+) UnsignedVector(mmcorej.UnsignedVector)
	+++  NEW CONSTRUCTOR: PUBLIC(+) UnsignedVector(long[])
	+++  NEW CONSTRUCTOR: PUBLIC(+) UnsignedVector(int, long)
	---! REMOVED METHOD: PUBLIC(-) void add(long)
	+++  NEW METHOD: PUBLIC(+) boolean add(java.lang.Long)
	+++  NEW METHOD: PUBLIC(+) void add(int, java.lang.Long)
	***! MODIFIED METHOD: PUBLIC java.lang.Long (<-long) get(int)
	+++  NEW METHOD: PUBLIC(+) java.lang.Long remove(int)
	+++  NEW METHOD: PROTECTED(+) void removeRange(int, int)
	---! REMOVED METHOD: PUBLIC(-) void set(int, long)
	+++  NEW METHOD: PUBLIC(+) java.lang.Long set(int, java.lang.Long)
	***! MODIFIED METHOD: PUBLIC int (<-long) size()
```

marktsuchida, 2021-08-04:
Here are the changes that we probably want to avoid (and probably can) (extracted and reordered from above):
```
***! MODIFIED CLASS: PUBLIC mmcorej.BooleanVector  (not serializable)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) BooleanVector(long)
***! MODIFIED CLASS: PUBLIC mmcorej.CharVector  (not serializable)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) CharVector(long)
***! MODIFIED CLASS: PUBLIC mmcorej.DoubleVector  (not serializable)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) DoubleVector(long)
***! MODIFIED CLASS: PUBLIC mmcorej.LongVector  (not serializable)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) LongVector(long)
***! MODIFIED CLASS: PUBLIC mmcorej.StrVector  (not serializable)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) StrVector(long)
***! MODIFIED CLASS: PUBLIC mmcorej.UnsignedVector  (not serializable)
	---! REMOVED CONSTRUCTOR: PUBLIC(-) UnsignedVector(long)

***! MODIFIED CLASS: PUBLIC mmcorej.BooleanVector  (not serializable)
	---! REMOVED METHOD: PUBLIC(-) void add(boolean)
	---! REMOVED METHOD: PUBLIC(-) void set(int, boolean)
***! MODIFIED CLASS: PUBLIC mmcorej.CharVector  (not serializable)
	---! REMOVED METHOD: PUBLIC(-) void add(char)
	---! REMOVED METHOD: PUBLIC(-) void set(int, char)
***! MODIFIED CLASS: PUBLIC mmcorej.DoubleVector  (not serializable)
	---! REMOVED METHOD: PUBLIC(-) void add(double)
	---! REMOVED METHOD: PUBLIC(-) void set(int, double)
***! MODIFIED CLASS: PUBLIC mmcorej.LongVector  (not serializable)
	---! REMOVED METHOD: PUBLIC(-) void add(int)
	---! REMOVED METHOD: PUBLIC(-) void set(int, int)
***! MODIFIED CLASS: PUBLIC mmcorej.UnsignedVector  (not serializable)
	---! REMOVED METHOD: PUBLIC(-) void add(long)
	---! REMOVED METHOD: PUBLIC(-) void set(int, long)

***! MODIFIED CLASS: PUBLIC mmcorej.StrMap  (not serializable)
	---! REMOVED METHOD: PUBLIC(-) void del(java.lang.String)
	---! REMOVED METHOD: PUBLIC(-) boolean empty()
	---! REMOVED METHOD: PUBLIC(-) java.lang.String get(java.lang.String)
	---! REMOVED METHOD: PUBLIC(-) boolean has_key(java.lang.String)
	---! REMOVED METHOD: PUBLIC(-) void set(java.lang.String, java.lang.String)
```

So the remaining job is to reintroduce the above constructors and methods by editing `MMCoreJ.i`, and to switch all of our automated builds to use SWIG 4.x exclusively. We should also cause the build to fail when attempting to use SWIG <4.

marktsuchida, 2022-02-11:
Something I do not address above is whether the API is equivalent between a Windows build and macOS/Linux build. This should also be checked.

marktsuchida, 2022-06-15:
Right. But I think we can solve the `*Vector` problem once the `long` problem is solved: we just need to create a Java `StrVector` that implements `List<String>` but also has the same methods as the SWIG 3 `StrVector`. Then,
- Wrapped functions taking a `List<String>` parameter will have no problem when passed a `StrVector`
- We need to make wrapped functions that return `std::vector<std::string>` return `StrVector` for backward compatibility, but user code can start treating the return value as `List<String>`. This is annoying but I think doable.