micro-manager/mmCoreAndDevices #44: Ti2 driver crashes reproducibly with latest Nikon driver on Windows 10

shenker, 2021-08-02:
Hi all,

This problem has been mentioned a couple times already:
- https://micro-manager.org/wiki/NikonTi2
- https://sourceforge.net/p/micro-manager/mailman/micro-manager-general/thread/CALvGY9RJi2YnkgweXxomxDvf%2BSCQd9qoyO0KV45-jSt41zGTPA%40mail.gmail.com/#msg36931563
- http://micro-manager.3463995.n2.nabble.com/Nikon-TI2-on-64-bit-MicroManager-td7590876.html

I wanted to collect information here so we can drive this towards a resolution.

Steps to reproduce (on latest build of Windows 10 64-bit):
1. Install MicroManager 2.0gamma nightly dated March 5, 2020 (I've tried this with a handful of 2.0 and 1.4 nightlies going back many months, they all show the same behavior).
2. Install Ti2 Control Ver2.20 (64bit) from https://www.nikon.com/products/microscope-solutions/support/download/software/biological/
3. Add Ti2 in Hardware Configuration Wizard. If you add any other Ti2 device, it works fine. If you add the Ti2 Nosepiece, it will crash immediately.
4. Run the Ti2 Control Ver2.20 installer to uninstall it.
5. Install Ti2 Control Ver2.00 and try adding the Ti2 Nosepiece in the Hardware Configuration Wizard. It will work.

Ver2.20 and Ver2.10 crash, Ver2.00 works, the MicroManager wiki link above implies that Ver1.20 works as well.

I'd be happy to submit any log files or run any debug builds if it would help, I don't have Windows development experience so hoping others can tell me what would be most helpful.

nicost, 2022-07-19:
I believe this is fixed now (at least, it works now with the latest/current version of Ti2Control).  Would be great if anyone else can confirm, and I'll close this issue.

bwaclaw, 2022-07-21:
Hello,

I just installed the latest Ti2 SDK that I could find on Nikon's website for developers (version 2.1.0.141), and Micromanager (version 2.0.1, built 21 July) and I can confirm that MM works with the latest Ti2 driver.

nicost, 2022-07-21:
Awesome!  Closing this issue.

closed 2022-07-21 by nicost