micro-manager/mmCoreAndDevices #58: "Olympus/Objective: objective turret" device

JYeste, 2021-08-19:
The device adapter "Olympus/Objective: objective turret" is not detected using a BX51WI microscope and a BX-UCB control unit. Since the rest of devices work well and the control unit recognize the objective turret, I assume it is an issue related to the commands that the device adapter code use to detect the device.

JYeste, 2021-08-30:
Thanks. I have created a problem report as you suggested. This is the report at the error and the full report:
"
2021-08-30T17:03:32.935640 tid11224 [IFO,dev:Objective] ExecuteQuery-Sending: "2POS?"
2021-08-30T17:03:32.935640 tid11224 [dbg,dev:COM4] SetCommand -> 2POS?\r\n
2021-08-30T17:03:32.935640 tid11224 [dbg,dev:Objective] FetchResponse
2021-08-30T17:03:33.436292 tid11224 [IFO,dev:Objective] FetchResponse-Timeout(2POS)
2021-08-30T17:03:33.436292 tid11224 [IFO,dev:Objective] Error 17, Objective, ExecuteQuery-FetchResponse
2021-08-30T17:03:33.436292 tid11224 [ERR,Core:dev:Objective] Error in device "Objective": Device not detected (10003)
"
[Problem Report.txt](https://github.com/micro-manager/mmCoreAndDevices/files/7080097/Problem.Report.txt)

The issue seems related with the communication commands between the microscope and the driver. In particular, the control unit doesn't replay to the command "2POS?" and previously to the command "2LOG IN". I think, the "2LOG IN" command gives computer control to the commands beginning with 2. I haven't found a list of commands for the Olympus microscope, so I cannot confirm that some versions of BX51 do not replay to those commands.

Related to your question about the control of the object turret, I'm able to move the turret manually with a joystick and through serial communication with the command "1OBJ (position)". In addition, the control unit has a light when the specified motorized revolving nosepiece is connected.

JYeste, 2021-08-31:
I understand. The Objective turret code needs the motorized Z drive from the Olympus. My motorized Z stage is from Prior; I can control it with Micro-Manager. Is it possible to adapt the objective turret drive to work with any generic Z drive?

Related to disable the use of the Z drive by the objective turret code, How can I do it? All the Olympus code is in the privateMain branch.

marktsuchida, 2021-09-01:
We would have to modify the Olympus device adapter code to add an option to disable Z escape during objective switching. Would that work for you? There is no way to use the Prior Z drive for this purpose, unfortunately.