micro-manager/mmCoreAndDevices #78: DA Z Stage does not work well with TriggerScope DAC channels

ieivanov, 2021-11-06:
The DA Z Stage does not work well with TriggerScope DAC channels (using TriggerScopeMM device adapter).

The DAC output can be changed using the DA Z Stage-Position slider in the Device Property Browser, however, the new stage position is not updated and stays at 0.

Controlling the stage Stage Control GUI also does not work well. I can move the stage up once from zero position, but not a second time. Moving the stage down throws an "Out of range" error.

Probably a straightforward bug, would appreciate help with it. Thanks!

nicost, 2021-11-08:
I can not reproduce either of these issues.  I have a configuration with the Triggerscope (a V3 with the current Triggerscope V3 firmware), and the Utilities - DA Z stage. Moving the DA-Z stage slider in the Device Property Browser updates the voltage output of the Triggerscope and also updates the Z position indicator in the Image Info line in the main window.  To update the position in the Stage Control plugin, check the "Polling updates" checkbox.

Using the arrows in the Stage Conrol plugin, I can move the Z position within its range (in my case, 0-200 microns).  The Z position in the Info line updates as expected.  However, the Z-Position in the DA Z Stage slider does not update (I have to press the Refresh button to get it to update).  

Which version of MM are you running?

ieivanov, 2021-11-09:
Hm, interesting. This may be a firmware issue, we're using a V4 TriggerScope with the latest V4 firmware and the 2.0 release version of MM from July 13, 2021. Anything I can do to troubleshoot this further?

nicost, 2021-11-09:
There was a commit on July 22 related to the DA Zstage in the TriggerscopeMM device adapter code .  I doubt that the Triggerscope firmware makes much of a difference.  Try updating  to a more recent Micro-Manager binary.

ieivanov, 2021-11-11:
Updating the firmware and the MM version fixed this issue, thanks for your help @nicost!

closed 2021-11-11 by ieivanov