micro-manager/mmCoreAndDevices #79: TriggerScope Falling Edge Sequencing

ieivanov, 2021-11-06:
We have a number of applications where we would like to change the state of the microscope after the camera exposure in a sequenced acquisition. We would set up the microscope state to Channel0 before the acquisition begins and set up sequencing on the falling edge of the camera exposure signal to cycle through Channel1, Channel2, etc. We set the acquisition framerate to the time needed for exposure plus the time needed to change the hardware to the new state.

This is nearly possible using the TriggerScope with sequencing on the Falling edge of the exposure signal. The problem is that for a 3-channel acquisition the acquired images correspond to Channel0-Channel0-Channel1, instead of  Channel0-Channel1-Channel2. The feature that is missing is shifting the sequence of states by 1 position when triggering on the falling edge, as implemented for example in the MCL NanoDrive device adapter.

There may be an argument for always shifting the sequence position by 1 when triggering on the falling edge. As a step further, the microscope could be set to Channel0 through software commands before the sequenced acquisition starts - in case the user did not do that. Are there applications where that's not desirable?

nicost, 2021-11-13:
These are firmware issues.  I implemented what I think is the correct way of doing this in the branch: https://github.com/micro-manager/TriggerScopeMM/tree/V4SignalLEDs.  I have no V4 here to test (plus it is a bit elaborate to set up a good test environment.  When you have a chance, can you load that firmware and see if it behaves as expected?  By the way, this firmware - in conjunction with recent MM builds, also lets you switch off the "Signal LEDs" on the device.  Turns out that some of these take ~0.5ms to switch, which is a bit of a pity.

ieivanov, 2021-11-16:
The V4SignalLEDs branch of the TriggerScopeMM firmware works great! Triggering on the falling edge works properly. Turning the signal LEDs off also removed the ~0.5 ms delay in switching the digital and analog states. Thanks @nicost!

nicost, 2021-11-16:
Merged here: https://github.com/micro-manager/TriggerScopeMM/pull/9

closed 2021-11-16 by nicost