micro-manager/mmCoreAndDevices #83: Sequencing along time does not work when z axis present

ieivanov, 2021-11-16:
With the latest nightly build of MM sequencing works when:
* Acquiring a time series with time internal set to zero (XYT dataset)
* Acquiring a z-stack (XYZ dataset)
* Acquiring XYCT dataset with time interval set to zero
* Acquiring XYZC dataset

but not when acquiring XYZT datasets - sequences are broken up between time points, such that individual z-stacks are acquired in a sequences acquisition, but the sequence does not span multiple time points. 

I'm not sure if this is intentional (XYCT works, but XYZT does not), but if so it would be good to have the option to allow sequencing of XYZT dataset through a flag in the Core.

@lihaoyeh

nicost, 2021-11-16:
These sequence strategies are created by the Acquisition Engine, not the core.  Could you move this ticket to micro-manager/micro-manager?  
Also, the acquisition engine is written in Clojure, and nobody speaks that language.  Henry has an acquisition engine that is slated to replace the existing one.  That work will need to be done first (and maybe Henry engine already has the features you ask for here).

ieivanov, 2021-11-16:
Thanks @nicost! I'll move this ticket over and I'll test the pycromanager acquisition engine for this behavior.

closed 2021-11-16 by ieivanov