micro-manager/mmCoreAndDevices #84: API needed for Camera triggers

nicost, 2021-11-16:
Cameras need a trigger to start exposing (or to start a series of exposures).  Triggers can be provided through software, using a clock internal to the camera (internal trigger), or by an external trigger source.  Each of these triggers can take various forms: software triggers can start a single exposure or a sequence of exposures (driven by the internal clock), external triggers can trigger on the rising edge of the signal or the falling edge, and sometimes exposure continues as long as the trigger is active (sometimes called "bulb" mode).  It would be wonderful to provide and API so that the same call can be used for cameras from different vendors.

In addition, there is some interplay with the SnapImage and StartSequence API calls that the MMDevice interface mandates.  For instance, it does not make sense in the Sequence functions to send a software trigger for each image in the sequence.  Internal triggering (or software-started internal triggering) is much more useful to the end user).  On the other hand, in the SnapImage function, software triggering is almost always desired over internal triggering (since response times using software triggering are lower and much more predictable).  We should offer some kind of guidelines to the device adapter authors what choices work best.

marktsuchida, 2021-12-01:
Here is one possible model for the standardized settings, though I haven't yet tested it against actual camera APIs to make sure it will work with the widest possible range of devices.

- **Start trigger source** = **None** or **External**
- **Frame trigger source** = **Internal** or **External**
- (Possibly) **Start trigger type** = **Edge** or **Level**
- **Frame trigger type** = **Edge** or **Level**
- (Possibly) **Start trigger polarity** and **Frame trigger polarity** = **Rising/active-high** or **Falling/active-low**

Some cameras have multiple trigger input pins that can be mapped to the start/frame trigger. Let's assume that such mapping is done by device-specific properties.

Of the 4 possible combinations of Start and Frame trigger source, the 3 that are more common are (Start = None, Frame = Internal), which is the default, (Start = None, Frame = External), and (Start = External, Frame = Internal). For cameras that can use separate input pins for the start trigger and frame trigger, (Start = External, Frame = External) also makes sense; for cameras with a single input pin, this setting should behave like (Start = None, Frame = External).

An alternative to the "start trigger" vs "frame trigger" distinction would be "trigger" vs "clock", which would be similar to the DAQmx terminology, but this is far from obvious, and "clock" has a different meaning for image sensors, i.e., vertical and horizontal clocks.

I would propose that "Snap" should work identically to a 1-frame sequence acquisition. Any other rule would seem impossible to memorize.

Many cameras support a software frame trigger. It is unfortunate that we don't support this, but it is hard to change this now in Micro-Manager, so I agree that the frame trigger source should be Internal or External only, at least for now. Camera adapters can internally use software triggers to improve Snap performance, as some already do; this is not visible through the MMCore API (and therefore doesn't enter into the model proposed here).

I propose calling the default start trigger source "None", rather than "Software", because our device interface does not have the capability to separate the (slow) acquisition setup from a (fast) software start trigger. But we could imagine adding, in the future, an MMCore function to send such a software trigger, so we should reserve the term "Software". Instead of "None", we could also call it "Automatic" (in the sense that the start trigger is automatically/implicitly sent after setting up the acquisition).

A start trigger type of Level would mean "keep acquiring frames while the trigger is at active level". I'm not sure how common this is; we can probably omit the Start trigger type entirely, and always consider the start trigger to be an edge trigger.

The terms "edge-triggered" and "level-triggered" are used in the context of [flip-flops](https://en.wikipedia.org/wiki/Flip-flop_(electronics)), [processor interrupts](https://en.wikipedia.org/wiki/Interrupt#Triggering_methods), and (less relevant) [OS kernel I/O readiness APIs](https://en.wikipedia.org/wiki/Epoll#Triggering_modes). One could argue that the meaning here differs from the meaning in the flip-flop/latch context, but it is close to the usage in interrupt triggering. There may be a better pair of terms. ("Bulb" is quite correct for a level-triggered frame, but requires familiarity with traditional photography to understand, and we would need a different term for the start trigger type (if supporting); also, it is not clear what the antonym should be.)

For the trigger polarities, I think it makes more sense to let it be device-specific, because they only need to be configured once and will not be switched during an acquisition (unless somebody can think of a counterexample), and therefore rarely require programmatic control. The fewer things we formalize, the lower chance that we encounter a device that doesn't play well with our model.

Aside from the above model and terms, there is the question of whether to implement this as dedicated device member functions (like `SetExposure()`) or as properties with standardized names (like `Exposure`).

There are also additional parameters that could potentially be standardized, such as **start trigger delay** and **frame trigger delay**. Also the timeout for external trigger, which currently causes much grief.

Finally, it might be worth comparing with how these settings are standardized by industry standards such as [GenICam SFNC](https://www.emva.org/wp-content/uploads/GenICam_SFNC_2_2.pdf) (and maybe [1394 IIDC](https://damien.douxchamps.net/ieee1394/libdc1394/iidc/IIDC_1.31.pdf)), although supporting all the possibilities is probably too much, and we have our own backward-compatibility considerations.

tomhanak, 2021-12-03:
I like Mark's proposal. However, I cannot map all PVCAM triggers to it. The very generic design of triggers in GenICam is worth reading, at least.
Here is what PVCAM offers and naming used.

Trigger sources (from camera hardware point of view):
- **Internal** - The camera starts exposing and sending data to host PC as soon as the acquisition is started by PVCAM (i.e. the hardware is internally timed). I'm not sure this means the same as _Internal_ described by Mark.
- **External** - Once PVCAM starts the acquisition, the camera sets its **Trigger ready** pin to high and waits for one or more external/HW triggers to collect some data (i.e. the hardware is externally triggered via 5V TTL signal on **Trigger in** pin).
- **Software** - Once PVCAM starts the acquisition, the camera is immediately ready for SW trigger and waits for one or more PVCAM commands to collect some data (i.e. the hardware is externally triggered by software on host PC).

Trigger types (not all can be used with all trigger sources):
- **Timed** - Only used with **Internal** trigger source, the camera captures frames as fast as possible.
- **Trigger First** - basically _Start trigger_ - When any type of acquisition is started, the camera waits for first rising edge from selected source, then switches to **Internal Timed** mode and captures frames as fast as possible.
- **Edge Trigger** - basically _Frame trigger_ - When any type of acquisition is started, the camera waits for a rising edge from selected source before starting exposure of new frame.
- **Level**, formerly also called **Bulb** - Only used with **External** trigger source. When any type of acquisition is started, the camera waits for a rising edge to start with exposure and keeps exposing the sensor until the signal falls to low, then transfers the frame data to host, sets **Trigger ready** to high again and waits for new trigger for another frame. The exposure time set via PVCAM is ignored.
- **Level Pulsed** - Only used with **External** trigger source. A special case of a **Level** trigger where the first trigger pulse starts the exposure and the next trigger pulse starts the readout of that frame. For N frames, there must be N+1 trigger pulses in the series with individual exposure times being equal to the time interval between the trigger pulses. The exposure time set via PVCAM is ignored.

The statement "a Snap should work identically to 1-frame sequence acquisition" is fully truth in PVCAM. Unfortunately, the Snap doesn't work this way in Micro-Manager. Nico wrote me once that MM expects the camera is always ready for a Snap. That explains why there is no "PrepareForSnap" function in camera API. The consequence is whenever user presses Snap button, or MDA is configured with a delay between frames, the device adapter calls "setup acq." and "start acq." PVCAM functions in order to capture one frame. Although the camera could be almost always ready (if we snap frames with the same setup), the "start acq." PVCAM function e.g. pins memory buffers for DMA that must be done again for every acquisition. So for PVCAM doing 100 Snaps and collecting 100 frames in one sequence acquisition are two completely different things.
A **Software** trigger with a sequence acquisition started once would eliminate the time needed to setup and start each snap acquisition, but that's not possible with current MM API.

jondaniels, 2022-06-20:
Two trigger modes I don't yet see mentioned we use heavily:

- "Synchronous" or "Overlap" where each external trigger indicates the end of one capture and the beginning of the next.  In other words, N+1 edge triggers are sent to capture N images.  Sort of like the "bulb" or "level" mode but with images immediately following each other.  Note that some cameras can be re-triggered during readout (I think PCO and PVCAM) but don't support this mode directly, whereas IIRC Hamamatsu and Andor cannot be triggered during readout but support this "synchronous" mode where a single trigger pulse is both a stop and start signal.
- "Light sheet" or "virtual slit" which really is a special case of another mode and involves camera shuttering too.  So maybe it should be excluded from this API, but in general it would be useful to have an API to use for it.

Having a camera API for setting the trigger mode would be very useful in the context of the MM1.4 ASIdiSPIM plugin and in the MM2 replacement being worked on now.  For the MM1.4 plugin we created our own Java enum to implement the following modes in supported cameras:
- Internal
- Edge
- Level
- Overlap
- Pseudo-overlap (the re-trigger during readout case mentioned above)
- Light sheet (which I would prefer to name "virtual slit" or something else instead since not all light sheet microscopes use it and it has uses other than light sheet)

henrypinkard, 2022-07-08:
I'm going to try and synthesize all this info to produce a draft API, but before I do a couple things:

@jondaniels 

> * "Light sheet" or "virtual slit" which really is a special case of edge trigger mode and involves camera shutter behavior.  So maybe it should be excluded from the camera trigger API, but in general it would be useful to have an API to use for virtual slit operation to make things uniform across cameras.

Could you explain more about what this is? Is this somehow related to rolling shutter or [TDI cameras](https://en.wikipedia.org/wiki/Time_delay_and_integration)?

@marktsuchida 
> I would propose that "Snap" should work identically to a 1-frame sequence acquisition. Any other rule would seem impossible to memorize.

Are you referring to specifically `core.snapImage()` and/or the `snap()` function in the camera API?

> Aside from the above model and terms, there is the question of whether to implement this as dedicated device member functions (like `SetExposure()`) or as properties with standardized names (like `Exposure`).

I lean towards functions like `SetExposure()`, not because it is more naturally expressed like that than in properties, but because I think standardized properties are much less clear from a user perspective. Correct me if I'm wrong, but there's no clear way when using the application to see if a particular property is standardized or device specific, right? So if you're just writing some code and have only one camera on hand, its hard to know when setting properties if those will generalize to to other cameras. There's also the disadvantage that if you're just reading through the API docs there's no mention of property standardization. Am I missing anything here? Is there something in the Core that could be changed to clarify this?

jondaniels, 2022-07-12:
@henrypinkard: regarding "virtual slit" mode: Yes it is related to rolling shutter and conceptually a bit like TDI cameras but definitely not the same.  Many but not all high-end sCMOS cameras support such a "virtual slit" mode.  Only a subset of rows are exposing at any instant, and every so often one row is read out and another row on the other side starts exposing so the number of exposing rows stays constant, but the exposing region moves across the sensor unidirectionally at a speed that is no faster (but optionally slower) than the row readout.

Different companies call it different things but here are some links (in some cases you need to search within the document)
[Hamamatsu Lightsheet Readout Mode](https://www.hamamatsu.com/us/en/product/cameras/cmos-cameras/lightsheet-readout-mode.html)
[pco ightsheet scanning mode](https://www.pco-imaging.com/fileadmin/user_upload/pco-product_sheets/DS_PCOEDGE42BI_V106.pdf)
[Andor  LightScan PLUS including FlexiScan](https://andor.oxinst.com/downloads/uploads/Zyla_hardware_user_guide.pdf)
[Photometrics Programmable Scan Mode](https://www.photometrics.com/learn/exclusive-product-features/programmable-scan-mode)

Some characteristics that could be abstracted include directionality, shutter height in pixels, row interval in microseconds, and delay in microseconds.  These would only apply when the camera is in the appropriate trigger mode which is a special case of per-frame edge trigger.

marktsuchida, 2022-07-12:
> > I would propose that "Snap" should work identically to a 1-frame sequence acquisition. Any other rule would seem impossible to memorize.
> 
> Are you referring to specifically `core.snapImage()` and/or the `snap()` function in the camera API?

Yes. For example, if the trigger mode is set to external start-on-edge (or whatever it may be called), a sequence acquisition of length 1 should acquire one frame at the software-configured exposure, starting when the trigger arrives. I think snap should do the same.

> Correct me if I'm wrong, but there's no clear way when using the application to see if a particular property is standardized or device specific, right?

Yes, at the moment, although such a way could be devised and implemented, if desired. For example, there is no reason why you can't have a property on the device side and a function on the application side (with the property also visible, allowing subscribing to change notifications -- although I don't know how important that is for the trigger settings). It would be important, in that case, to choose a property name that won't be used elsewhere (previously, or inadvertently in the future) -- perhaps with some special prefix -- and also to enforce rules about the type and allowed values at the time of property creation (or have a dedicated function to create the property).

marktsuchida, 2022-07-12:
Yeah, I think an interface based on dedicated functions is reasonable here.

henrypinkard, 2022-07-13:
Here is a potential abstraction. If this covers everything we're trying to do, I can write out a corresponding API.

 To begin, I'll clarify terminology, which I've chosen mostly for consistency with GenICam, not based on the current micro-manager API (as will be described shortly).

# Introduction

The important concepts are **Frames**, **Bursts**, and **Acquisitions**. 

A **Frame** describes the acquisition of a single image, including a reset before exposure, exposure, readout, and optional delays before reset and after readout (often these will be 0):

[image]

A **Burst** is sequence of frames. Shown below they occur one after another, but as will be described below in some cases parts of frames can overlap:

[image]

An **Acquisition** is a sequence of **Bursts**:

[image]

In Micro-Manager's current terminology, **Bursts** are called "Sequence Acquisitions", which are distinct from the higher-level concept in Micro-Manager, which are functionally equivalent to the **Acquisition** described here. Because this is potentially confusing, I'll drop "Sequence Acquisition" and stick with **Burst**.

**Acquisitions** don't really figure into the design of this API--they are handled at a higher level of abstraction.

## Types of triggers
A **Trigger** is a signal to take some action related to **Frames**/**Bursts**. There are 4 triggers defined in this proposed API. Every burst will require a source for each trigger.

**BurstStart** and **BurstEnd** triggers signal the start and end of a burst of frames. 

 
[image]

The **FrameStart** trigger signals the start of a frame. The **ExposureEnd** trigger signals the end of the frame's exposure (but not the end of the frame, since there is still readout).

[image]

## Sources of triggers
Triggers can come from software commands from computer to camera (**Software**), hardware TTL signals (**External**), or can be generated by the camera itself (**Internal**).

## FrameExposureMode 
Separate from the source of the trigger, there are 3 different modes for **FrameStart** and **ExposureEnd** triggers. 

In **Distinct** mode, **FrameStart** and **ExposureEnd** triggers are separate events, and can in some cases come from different sources (e.g. **external**/**internal**).

In **Sustained** mode, which only makes sense with **external** **FrameStart** and **ExposureEnd** triggers, one long TTL pulse is used to signal the interval from frame start to exposure end. (This is also known as "strobe" or "bulb" or "level" triggering).

In **Combined** mode, the **ExposureEnd** trigger for frame N is the same as the **FrameStart** trigger for frame N+1. This covers @tomhanak's "Level pulsed" and @jondaniels' "synchrnous"/"overlap". It seems to me this could make sense with either **external** or **software** **FrameStart** and **ExposureEnd**, though they'd probably need to be the same type (i.e. both **external** or both **software**).

[image]

# Examples of common scenarios

Below is a representative, though not exhaustive, list of example combinations. It would probably be useful to make a table of all possible combinations.

[image]

## Software triggered burst
Make an API call to start a burst, with the interval between frames determined by exposures and/or framerate properties and controlled by the cameras internal clock 

[image]

## Software triggered burst with TTL triggered frames
Make an API call to arm the camera to start a burst, with individual frames started according to received TTL pulses, and exposure end controlled by the camera

[image]

## TTL trigger a burst with internally triggered frames
An external TTL triggers the start of a burst, after which the camera controls frame starts and and ends. The camera has been preprogrammed (through the API) to collect a certain number of frames (here 2).

[image]

## Bulb/Strobe triggers
Software trigger a burst, with exposure controlled by a sustained TTL pulse (i.e. **FrameExposureMode** == **Sustained**)

[image]

## Synchronous

**FrameStart** for next image is identical to next frame's **ExposureEnd**.  (i.e. **FrameExposureMode** == **Combined**)

[image]

# Still left to figure out

- There isn't enough here to cover "Virtual slit" mode. More API calls will be needed to get/compute line readout times
- Querying cameras about what TTL signals they will provide. It could be useful to know for a future acquisition engine and/or synchronization device control the interactions of multiple devices. This might involve more types of triggers than described here (e.g. ExposureStart)
- If this is backwards compatible with current API/how to make it so

marktsuchida, 2022-07-14:
This is good progress.

> If this covers everything we're trying to do, I can write out a corresponding API.

Actually, I feel that it would be a lot easier to answer this question once there is an API design written out. I don't think the functions and their constraints automatically follow from the terminology alone (or maybe I just don't see it).

As I mentioned the other day, it will be especially helpful to show how it corresponds to camera SDK APIs (and their enum constants), for those that have publicly-accessible documentation. And GenICam, which has a good model (even if not covering everything we want?) followed by many of the industrial/vision camera manufacturers.

In fact that could be more important than covering every possible triggering mode, because it's not a huge deal if we don't (initially) support some exotic mode only found in a few vendors. It is a big problem if the typical/common/standard modes take a lot of effort to implement in the camera adapters or have a confusing mapping to our API.

nicost, 2022-07-17:
> // Current API
> int SnapImage();
> // New API
> int SendFrameStartTrigger();

Do you propose to make the current SnapImage into a special case of a Burst? The current  SnapImage function blocks for duration of the exposure to enable a simple way to synchronize light sources with camera exposure, and it also has a dedicated method to retrieve the pixel data ("GetImageBuffer").  How will that work with the proposed API?  Probably more general, how will synchronization with light sources and data retrieval work with this new API?

henrypinkard, 2022-07-18:
> Do you propose to make the current SnapImage into a special case of a Burst?

In this paradigm it could be a burst of 1 frame  or it could be 1 frame in a longer burst. I'm not sure which one it makes sense to map to the default implementation of SnapImage. It sounds like this currently varies by device adapter: @marktsuchida mentioned something about one of of them always waiting for a signal to snap (i.e. a BurstStartTrigger has already been sent), and then switching modes when an internally triggered burst is needed.

> The current SnapImage function blocks for duration of the exposure to enable a simple way to synchronize light sources with camera exposure, and it also has a dedicated method to retrieve the pixel [...]

Upon further consideration, it probably makes sense to not have these be two separate functions:

```c++
int SendFrameStartTrigger();
int SendExposureEndTrigger();
```

I can't think of why you'd ever actually need to send an `ExposureEndTrigger`, so that could be dropped from the API, and `SendFrameStartTrigger()` could block until exposure end. This might actually be an improvement over SnapImage(), because that blocks until after readout/copy of data into an image buffer (I think?), so you could potentially turn off light sources even faster. This change would mean that an image isn't necessarily ready after `SendFrameStartTrigger()` returns, so we'd probably want to also add something into the core function for getting the image that blocks until an image is ready in order to not introduce confusing behavior for people still using SnapImage.

As for readout, I think this would be a good opportunity to rectify the often confusing fact that the snap buffer and circular buffer are separate entities. So I would propose a unified mechanism where any frames acquired through all means are placed in a common buffer, and dedicated thread in the core that whenever a burst is active is constantly polling for new images. Would be curious to hear others thoughts about whether this would lead to problems and if there are better alternatives. This should probably be considered in tandem with  improvements to the circular buffer/multi-camera behavior https://github.com/micro-manager/mmCoreAndDevices/issues/168.

nicost, 2022-07-18:
> This might actually be an improvement over SnapImage(), because that blocks until after readout/copy of data into an image buffer (I think?), so you could potentially turn off light sources even faster. 

The "contract" for the current SnapImage is to block for duration of the exposure (i.e. not for the readout), so - if the SnapImage function is carried out correctly, it is as fast it should be.  

> As for readout, I think this would be a good opportunity to rectify the often confusing fact that the snap buffer and circular buffer are separate entities. 

Agreed, but this will make it a very substantial change to how many things in Micro-Manager operate.  It then makes sense to add the plans for data transfer and shutter synchronization to this API sketch.  I guess we should keep the things that need to implemented by the Camera Device Adapter authors separate from additions/changes to behavior of the Core and user layers, but we should have a clear idea how the whole will function.

marktsuchida, 2022-07-26:
Sorry for the late reply.

I agree with the points that @nicost made. It would be great if the new API could be based on the old one, and interoperate where possible, rather than just ignoring the old API. Ignoring the existing API is easier for the designer, of course, but it puts burden on everybody consuming the API.

For example, what happens if you configure the new-style trigger and then call the existing `snapImage()` or `startSequenceAcquisition()`? Presumably you want that to work if the configured trigger type can be supported using the existing API. Worse, what if one intermixes the new and old prepare/start/trigger functions? Even if you just say that's an error, checking all these cases would require a lot of state management.

Perhaps it would be best to just handle the trigger configuration (and information querying, and sending software triggers) with the new API but make it work with the existing acquisition API first, and then tackle the "better acquisition API" as a separate project (it will be connected to the multi-camera support, too, and should have a good way to segregate it from the existing API).

(It is also totally unclear to me how the device-side interface (MMDevice) will work under this scheme.)

So, getting back to the actual trigger specification part:

- What do we gain by defining our own way to specify triggers instead of just copying GenICam as closely as possible (and possibly adding minor extensions or additional parameters)? (We lose a lot, so it will only be worth it if we gain a lot.)
    - Having our own spec might have made sense if we were only supporting a handful of simple modes, as was my expectation when I first commented on this issue. Now that we want to support so much, and it is clear that the API won't be simple, it seems better to avoid reinventing an incomplete (or complete but gratuitously incompatible) wheel.
    - There could be valid arguments why we cannot copy GenICam (such as backward compatibility with our API, or that their underlying system of features being too complicated -- but if the latter, we probably need to make sure our "simple" way actually works equally well).

- Because the configuration is split across 7 functions, it would seem impossible for user code to tell which combinations of options are supported. For example, let's say I want to set the frame start trigger to software, but `setFrameStartTriggerType` throws an exception. How do I know if that is because the camera doesn't support software trigger at all, or if the camera doesn't support software frame start in combination with the current values of all the other settings? The result may even differ depending on whether I set the frame start trigger first or the burst start trigger first.
    - I suspect (but have not confirmed) that GenICam solves this using their feature graph system, but it might become unusable without such a mechanism. Not to mention that GenICam allows you to query the allowed values for each feature and all that.

- There are 4 triggers and only (up to) 3 types that can be set, so at least two triggers need to be connected to the same input. I think you intended those to be software (the only one that has separate send-trigger functions), but then one presumably needs to send pointless software triggers that the device adapter somehow ignores(?). GenICam allows you to switch trigger on or off.

- I think it makes sense to have a single `sendSoftwareTrigger` function that takes an argument specifying which trigger. That way, we don't need to add functions when a new type is added in the future. Again, see GenICam.

- Many cameras have the concept of frame rate (another thing we poorly support so far). How does the post-frame delay interact with frame rate (currently exposed as a property or not exposed)? Not sure this is critical but it may be.

- I'm not very confident that I'm seeing all possible problems just yet; this could take a lot of work and I haven't seen how this interacts with actual camera SDKs or application code (only took a quick look at the GenICam SFNC as I wrote this).

henrypinkard, 2022-07-27:
> I agree with the points that @nicost made. It would be great if the new API could be based on the old one, and interoperate where possible, rather than just ignoring the old API. Ignoring the existing API is easier for the designer, of course, but it puts burden on everybody consuming the API.

> For example, what happens if you configure the new-style trigger and then call the existing snapImage() or startSequenceAcquisition()? Presumably you want that to work if the configured trigger type [...]

This is a miscommunication. I am not suggesting to ignore the old API

> What do we gain by defining our own way to specify triggers instead of just copying GenICam as closely as possible (and possibly adding minor extensions or additional parameters)? (We lose a lot, so it will only be worth it if we gain a lot.)

This was meant to copy a subset of GenICam as closely as possible, though it may not yet be perfect. GenICam supports many many many more fetures that copied over here. This simplification is essential.

> Because the configuration is split across 7 functions, it would seem impossible for user code to tell which combinations of options are supported. For example, let's say I want to set the frame [...]

I'm not sure what you're mean by the "feature graph" (the word "graph" does not appear anywhere in the document). As far as I can tell, GenICam does not solve this problem

> There are 4 triggers and only (up to) 3 types that can be set, so at least two triggers need to be connected to the same input. I think you intended those to be software (the only one that has [...]

I don't understand what you're talking about here. GenICam "off" is equivalent to "internal" in the spec above. This is for consistency with the language we've used in earlier posts on this thread

> I think it makes sense to have a single sendSoftwareTrigger function that takes an argument specifying which trigger. That way, we don't need to add functions when a new type is added in the future. Again, see GenICam.

I think dedicated API functions for each is more consistent with how MMCore is currently set up. For example, we have `startContinuousSequenceAcquisition`, not `AcquisitionMode="Continous"` like GenICam. So if we copied the genICam way for the new stuff, but aren't going to change the existing API, we'd have a hybrid API where sometimes GenICam is followed and sometimes it's not. Personally, I think programming with the API is a little clearer if it's all consistent, and I also prefer having dedicated method names rather than having to know valid enum values. But I'm fine with implementing whichever

henrypinkard, 2022-07-27:
Other points:

- An alternative to the proposed is to map micro-manager sequences to GenICam acquisitions. Based on my reading of the GenICam spec, as well as the Basler Pylon API, this choice seems to be degenerate--both might be possible. Maybe feedback from camera companies will further inform this
- I think this can be implemented pretty easily without having to make circular buffer changes (as described in document):

I wrote up a document describing this proposal [here](https://github.com/micro-manager/mmCoreAndDevices/blob/main/camera_triggering_API.md)

Closing this issue so that we can open a fresh one and bring new people on to give feedback on this API

marktsuchida, 2022-07-28:
> I'm not sure what you're mean by the "feature graph" (the word "graph" does not appear anywhere in the document). As far as I can tell, GenICam does not solve this problem

So you didn't actually read the GenICam standard. It's called the node graph in the [document](https://www.emva.org/wp-content/uploads/GenICam_Standard_v2_1_1.pdf). I'm pretty sure GenICam does solve this problem using the node graph and other mechanisms.

I'm not saying we should replicate that mechanism (we can't, it is very complex). But we can't just take the superficial spec from the SFNC without having an equivalent underlying structure. We need to somehow come up with a mechanism that ensures all the parts work together, and I'm not seeing that anywhere.

(One idea might involve getting and setting the configuration in bulk, instead of having individual functions for each parameter. Or maybe a set of functions to query capabilities can solve the problem. I don't know what the best solution is, but we need a solution.)

> GenICam "off" is equivalent to "internal" in the spec above.

In GenICam, there is no "internal" as such (except for the advanced counter- and timer-based internal triggers, which we thankfully don't need to support now), but the default start trigger (when trigger mode is off) in MultiFrame is indeed like what we think of as internal. However, the default (off) for the burst start trigger is not internal, because by default there are no triggered bursts. So they are not equivalent once you pay attention to:

> * An alternative to the proposed is to map micro-manager sequences to GenICam acquisitions. Based on my reading of the GenICam spec, as well as the Basler Pylon API, this choice seems to be degenerate--both might be possible. Maybe feedback from camera companies will further inform this

Why is _my_ feedback being ignored? The manufacturers can help with how their APIs work (although it's not like we can't see that in their docs), but they are not experts on the MM API.

GenICam bursts are nothing like MM acquisitions, as I explained on Zoom. See page 165 of the [SFNC](https://www.emva.org/wp-content/uploads/GenICam_SFNC_v2_7.pdf) for an example.

We probably don't need to support bursts (as defined by GenICam) at this time. But I see no reason to gratuitously introduce the term "burst" to mean something totally different from GenICam, when our existing term "sequence acquisition" already matches what GenICam calls a "continuous" or "multiframe" "acquisition". It will not only be utterly confusing (especially if claiming some sort of similarity to GenICam, which certainly threw me off the track), but will also preclude us from supporting bursts in the future using any sort of reasonable terminology.

> I think dedicated API functions for each is more consistent with how MMCore is currently set up. For example, we have `startContinuousSequenceAcquisition`, not `AcquisitionMode="Continous"` like [...]

By this argument, you would need separate functions like `SetBurstStartTriggerTypeExternal()`, which is a little absurd. I think the comparison with `startContinuousSequenceAcquisition()` is a straw man (look at MMDevice, where it's a single function). If you are fine with either, making the trigger type a parameter is better, for future extensibility. Actually, the set-trigger-source functions should probably also be unified into `setTriggerSource(burstStart, external)`, again for the possible future addition of trigger types.

> This was meant to copy a subset of GenICam as closely as possible, though it may not yet be perfect. GenICam supports many many many more fetures that copied over here. This simplification is essential.

Of course simplification is essential, but the more I look at this, the more I realize that it's not even close to GenICam's general model. We can start by eliminating arbitrary differences between the terms used -- because otherwise it's hard to even point out why your design is incompatible with GenICam.

It would be great if we can come up with a subset of GenICam (an actual subset, meaning that it could _in principle_ later be expanded to full compatibility; not a hand-wavy subset). But we may have to weigh this against practicalities, and make a deliberate decision to not be 100% compatible. But that can only be done after fully understanding and comparing the models.

I probably should have commented on your new issue, but then it would have been hard to reply to specific points you wrote above. I think all of the issues above still apply to your new proposal.

nicost, 2022-07-28:
> So you didn't actually read the GenICam standard.

That is a disrespectful comment, and unnecessary.  Please stay constructive

This discussion becomes long, and hard to understand.  Henry provided an API proposal in which he tried to accommodate all input, including everything Mark brought up.   Mark, can you please provide a concrete proposal rather than continue to  point out the negatives?  If we do not move forward, we will never get anywhere.  

At this point, I will invite comments from camera companies to the document provided by Henry.  Mark, if you see room for improvements, can you please provide those in that document?

closed 2022-07-28 by nicost

marktsuchida, 2022-07-28:
> > So you didn't actually read the GenICam standard.
> 
> That is a disrespectful comment, and unnecessary. Please stay constructive

@nicost You are right. @henrypinkard I apologize for that. I shouldn't have used such a tone.

marktsuchida, 2022-07-28:
I beg that you don't ignore the rest of my comment, however, despite perhaps compromising my own right to do so. I think I need to cool down for a day or two before getting back to commenting on the proposal, but the points I raised there still apply to the new proposal, unless I'm missing something.

I hope you will read the rest of this comment as filling in some background, not trying to excuse myself.

> Mark, can you please provide a concrete proposal rather than continue to point out the negatives?

I honestly wish I could. It is always better to show how it's done than to point out flaws. Unfortunately, this is not an easy problem, there are many things to study and consider, and while I can point out some issues that need further thought, actually providing a full solution on my part would take more time than I can find in the short term (plus, I don't want to steal Henry's project).

> This discussion becomes long, and hard to understand.

It is, but it is frustrating when I put hours of work into trying to understand it and comment on it, and then those comments are dismissed as "seems to be degenerate" and whatnot without any real technical explanation why, the issue is closed before I have a chance to clarify my points, and a new document is created that still has most of the same issues (except, I should acknowledge, for the one issue that _was_ addressed, regarding the existing snap and sequence acquisition APIs), and I'm expected to comment again from scratch. It would be a different matter if my comments were acknowledged as unresolved and carried over to the new thread. Not to suggest that any of this is in any way an excuse for the language I used; I really feel bad (and stupid) about that.

henrypinkard, 2022-07-29:
Thank you for apologizing @marktsuchida. I appreciate that you're very passionate about getting this right. And I don't know where we would be without your encyclopedic knowledge and many excellent suggestions!

To be clear, I  do not mean to dismiss any of your previous concerns. I'm using the word "degenerate" in the linear algebra sense--no unique solution. I can see valid reasons to adopt your proposal and valid reasons to stick with mine. For that reason, I think it is important to solicit input from others. You will, of course, have more chances to provide feedback on this. It would be helpful if you could continue workshopping your ideas into more concrete proposals for changes to the API, so that we can decide on whether to adopt them without losing too much time to extensive back-and-forths, since there are many other areas that need attention.

marktsuchida, 2022-07-29:
@henrypinkard Thank you.