micro-manager/mmCoreAndDevices #87: Create a conda-forge package

ianhi, 2021-11-30:
It would be great if this repo was available via conda-forge. This would make a linux computer a viable option for controlling a microscope.

I think that this would likely be a follow up to https://github.com/micro-manager/mmCoreAndDevices/pull/86

But opening now to discuss:

1. Desireability
2. If/how it would be possible to include the `SecretDeviceAdapters`.

I think an example recipe would be libtiff: https://github.com/conda-forge/libtiff-feedstock/blob/master/recipe/build.sh

Some other relevant links:
https://docs.conda.io/projects/conda-build/en/latest/concepts/recipe.html
https://conda-forge.org/docs/maintainer/adding_pkgs.html
https://conda.io/projects/conda-build/en/latest/user-guide/tutorials/build-pkgs.html

Also curious if @tlambert03 has thoughts as you're the only person in microscopy world who I know has made conda-forge pacakges.

tlambert03, 2021-11-30:
i think that would be great, and I can try to help... one thing I can tell you is that they'll want a clear license file in this repository (or whatever source gets used to build).  Where do "releases" go here?  Is there an official tarball source?

ianhi, 2021-11-30:
> If/how it would be possible to include the SecretDeviceAdapters.

Given that the licenses of secret adapters would surely muddy the licensing of the repo as a whole is this maybe a good candidate for a `micro-manager` conda channel rather than conda forge?

tlambert03, 2021-11-30:
just a note there: conda-forge requires all dependencies to be on conda-forge.  So, it's fine to make a new channel, but it's a bit less useful, since no conda-forge package could depend on it.