micro-manager/mmCoreAndDevices #106: Fix TIScam for new msvc version

nanthony21, 2021-12-14:
The TIScam device adapter currently fails to build  with msvc v142 due to the following lines in the SDK that it includes from "3rdParty":
```C++
#elif _MSC_VER > 1800
  #error This compiler was not tested with this library.
#else
  #error Wrong Compiler. This library does only run with Visual C++ 7.1, 8.0, 9.0 and 10.0.
  // other maybe newer compiler ...
#endif
```

If the `#error` is commented out then everything builds fine. I can commit this change to the SVN repo but wanted to use this issue as a place to review the change before commiting.

marktsuchida, 2021-12-14:
It would be pointless if only we can build it with VS2019 (using a modified SDK in a private repo), so I think we need to upgrade the SDK.

nanthony21, 2021-12-14:
This change wouldn't break compatibility with older compilers, it would just remove the restriction on using a compiler that is newer than what the SDK was tested on. Having said that, it is a bit of a hacky fix so if you new SDK is available that may be a better option.

marktsuchida, 2021-12-14:
Indeed. Editing the SDK header would just be pushing the problem down the line. I wouldn't call that a successful migration to VS2019.

Plus, this is a C++ API, so they may have good reason to restrict the compiler version; it could compile yet fail to work correctly if the compiler version is wrong.

It looks like the last 3 available versions of the SDK support Visual Studio 2019:
https://www.theimagingsource.com/support/downloads-for-windows/software-development-kits-sdks/icimagingcontrolcpp/

nanthony21, 2021-12-15:
Ok. I've downloaded 3.5.6 and changed the project to use that instead. It looks like there are some breaking changes that will need to be fixed but once that's done I'll submit a PR and commit 3.5.6 to the SVN repo

marktsuchida, 2021-12-15:
Sounds good!

closed 2021-12-16 by nanthony21