micro-manager/mmCoreAndDevices #129: vs2019 branch: HIDManager required new build of hidapi library

marktsuchida, 2022-01-21:
The hidapi API is C, so HIDManager will likely work as is, but we should upgrade so as not to require the Visual C++ 2010 Redistributables.

```
mmgr_dal_HIDManager.dll
    hidapi.dll (bundled)
        MSVCR100.dll
```

nicost, 2022-02-02:
As far as I understand, we are currently using a patched version of the hidapi code on github.  @marktsuchida's patch to the hidapi code is present upstream, so all that is needed is to clone the hidapi github repository, build the code using vs2019 locally, and put the relevant files (include files, dlls, and .lib files) in a relevant place in 3rdparty/hidapi.  Please advice if this is not correct.  Also, do I remember correctly that we will only build 64-bit code?

nicost, 2022-02-02:
Correction:  It appears that development of hidapi has moved from signal11 to https://github.com/libusb/hidapi.  They even have (windows) releases there.  Is it possible to pull those automatically to the build machine, or is it still better to use the 3rdpartypublic repository?  I'll add something there for now (with a readme), and make sure that the code builds.  Not sure if I can test it (as I have no supported HID device here).

marktsuchida, 2022-02-02:
I self-assigned this having started work on it, but hadn't done a whole lot, so backing out....

We do not currently have any mechanism to download (non-Java) packages for the Windows build currently, so I would rather it stayed that way (if we fetch files, we need to have a mechanism that everybody can run smoothly, which is one of the many reasons to move to Meson; I don't think an ad-hoc script is a good idea, even though that is what we do for the macOS build). I think it is best to place the binaries in `3rdpartypublic` under the current scheme of things.

I don't have a supported HID device, either....

nicost, 2022-02-02:
Sorry, missed the self-assignment, looking for small things I can contribute.  PR about to be submitted.

marktsuchida, 2022-02-04:
One more thing needed: the `build.xml` needs to install the new hidapi DLL. I'm working on a fix now.

marktsuchida, 2022-02-04:
Although we have not tested this with a supported device, HIDManager loads correctly, which implies that it was able to enumerate devices by calling hidapi. So I think we are ready to close this.

closed 2022-02-04 by marktsuchida