micro-manager/mmCoreAndDevices #130: vs2019 branch: PointGrey requires new SDK

marktsuchida, 2022-01-21:
The PointGrey device adapter uses the C++ interface of the (legacy) FlyCapture SDK, which requires a version matched to the compiler.

(I believe the vendor was providing specific downloads for Micro-Manager, which might need to be updated. See https://micro-manager.org/Point_Grey_Research)

```
mmgr_dal_PointGrey.dll
    FlyCapture2_v100.dll
```

nicost, 2022-02-01:
The FlyCapture SDK we are currently using (2.10.3.266) has dlls for the v110 toolset (not included in our repo, but available), but not for v140.  The current release (2.13.61) includes dlls for vs2015.  I will add 2.13.61 to our 3rdparty repo and change the build file to build using that version.

nicost, 2022-02-02:
Taken care of in PR #136.  Once merged, we should update the documentation.

closed 2022-02-02 by nicost

kmdouglass, 2023-07-25:
Hi @nicost and @marktsuchida ,
Sorry to bring up this old issue, but I'm wondering whether the Point Grey device adapter documentation also needs to be updated to state that the FlyCap SDK 2.13.3.61 is required for Micro-Manager versions after February 2, 2022? 

https://micro-manager.org/Point_Grey_Research

The page above still states that FlyCap 2.10.3.266 is required, but I noticed that the Point Grey device adapter will not load in recent MM versions if FlyCap 2.10.3.266 is installed.

For the moment I'm not entirely sure whether it's only the FlyCap version that's preventing us from using the camera in recent MM versions because it takes time to get the download link for FlyCap from Flir. I have a student who's looking deeper into this and can get back to you if you need more info.

marktsuchida, 2023-07-25:
I think you are most likely right; 2.13.3.61 is the current version used in our build. (Or if your camera is supported you might be able to use the Spinnaker device adapter.)
Great if you could let us know if 2.13.3.61 worked (or feel free to create a PR for the device page).

kmdouglass, 2023-08-15:
Fixed in https://github.com/micro-manager/mmCoreAndDevices/pull/364 and https://github.com/micro-manager/micro-manager.github.io/pull/123