micro-manager/mmCoreAndDevices #151: Add threads to demo device

ianhi, 2022-02-18:
The demo devices are very useful for developing software based on this. However, they have an important omission from modern device adapter behavior. That is that the demo adapters don't use threads anywhere (in particular they never trigger a callback from an internal thread). 

Having them do this would have allowed for much earlier catching of issues like: https://github.com/micro-manager/pymmcore/issues/49 and also is a roadblock to being fully confident in something like https://github.com/tlambert03/pymmcore-plus/pull/73.

If given a bit of advice on how to apporach this (and maybe an example from an existing adapter) then I'd be happy to have a go at implementing this.

ianhi, 2022-02-18:
Also if someone can help point me to resources for doing this I'd love to add threads to the TE2000 adapter as I'm taken to believe that this will allow more functions to return immediately instead of blocking

marktsuchida, 2022-02-18:
To add to this, the fundamental problem we have is that the threading model of callbacks was never specified, so that different device adapters (and sometimes the same adapter) do different things. It is also easy to write client code that assumes callbacks are always on device-internal threads, and deadlock when some device issues a synchronous callback (e.g. from within `SetProperty`).

It is theoretically possible for MMCore to convert all synchronous callbacks into asynchronous ones, but I don't think it is worth the added complexity given that GUI applications typically need to reissue these events anyway.

The approach taken by MMStudio is to collect all callbacks in a single location, and schedule them on the UI (event dispatch) thread. I think this approach works well for GUI applications, and I don't think any other approach is really viable in most GUI frameworks.

I would be open to adding helpers to pymmcore if it can simplify callback handling in Python.

But I agree that in any case it would still be nice if the demo devices had at least some examples of asynchronous callbacks.

As for how to modify DemoCamera: my suggestion would be to wait until next week when we expect to enable C++14 (#152), then use [`std::async(std::launch::async, ...)`](https://en.cppreference.com/w/cpp/thread/async) to issue the callback from a property handler (but make sure to read the note regarding the `std::future` destructor blocking -- you'll need to keep the future around as a data member of the device class). You could add a new property `AsynchronousCallbackTest` (or something), following the example of the existing test properties. If you want a delay between the property being set and the callback firing, it's just a matter of sleeping in the asynchronous procedure.

There are no good examples from real device adapters that I can remember (because most of those rely on a thread in the vendor SDK if they issue asynchronous callbacks at all). Except maybe the Leica and Zeiss stand adapters, but those have a different overall architecture from the simple DemoCamera (they have a thread reading incoming serial messages).

ianhi, 2022-02-18:
> I think it is better to do something above the MMCore layer that applies to all devices.

That would be ideal as it would also improve the micromanager experience. As an anecdote one of the things that started me down the road of trying to use python was frustration with micromanager freezing up when using the TE2000.

ianhi, 2022-02-21:
@marktsuchida do you have any advice on how to set up the submodules so that I can compile this. My understanding is that on linux I need to use the micromanager repo to actually do the build, and Ideally I should be able to link to a local git submodule but some brief googling hasn't been super helpful

marktsuchida, 2022-02-21:
@ianhi Do you mean to compile mmCoreAndDevices (including DemoCamera)?

```
git clone --recurse-submodules https://github.com/micro-manager/micro-manager.git
cd micro-manager
./autogen.sh
./configure   # with any needed args
make
```

Let me know if I misunderstood your question; I wasn't sure what you meant by "link to a local git submodule". (If you are talking about avoiding re-downloading Git objects for the submodule, I think you can do a non-recursive clone and then use `git submodule update --init --reference path/to/existing-mmCoreAndDevices-checkout`.)

ianhi, 2022-02-21:
> Let me know if I misunderstood your question; I wasn't sure what you meant by "link to a local git submodule". (If you are talking about avoiding re-downloading Git objects for the submodule, I [...]

I mean that I want to use my local copy of the mmCoreAndDevices repo (where I will be making changes). Currently I have that in a different folder, but maybe thats wrong?

ianhi, 2022-02-21:
I seem to have figured it out. I used `submodule set-url` to change the url to my fork, then i used your command to update it and now I  just work in that repo. thanks!

closed 2022-02-25 by marktsuchida