micro-manager/mmCoreAndDevices #168: Need support for simultanous acquisition of cameras of different sizes

nicost, 2022-03-01:
Micro-manager's system of using a single buffer for camera images limits its ability to work with multiple cameras simultaneously.  The current "Multi-Camera" adapter can only work with cameras that produces images of the same size and is a bit of a hack to begin with.  In addition, it will be very important that camera operations are performant, and that copying of data and metadata is restricted to an absolute minimum.  

One approach would be to make the number of cameras directly supported by the Core configurable, and a circular buffer could be created for each camera.  It may be most useful to start thinking about the api calls that are needed.  I am also very interested what work-arounds others have been using so far.

marktsuchida, 2022-03-02:
I think the best overall direction is to create a new API that is more versatile and, depending on what degree of backward compatibility we want, emulate the existing "Multi Camera" inside MMCore. However, I don't know if every single aspect of the current API (especially the very idiosyncratic `getNBeforeLastImageMD`) can be easily emulated by a cleanly implemented sequence buffer.

Also, before anybody gets confused, the current "circular buffer" is not a circular buffer or ring buffer. It is actually a bounded queue. Except in Live mode (where overflow is allowed), where it is completely weird.

marktsuchida, 2022-03-03:
Please see https://github.com/micro-manager/mmCoreAndDevices/issues/171#issuecomment-1058312997 for some of the things one must be aware of before proposing a solution to this.