micro-manager/mmCoreAndDevices #169: add `getOrGuessChannelGroup` method

ianhi, 2022-03-02:
`getChannelGroup` is a super helpful method, but only if the user has defined it in their config. So in pymmcore-plus we implemented a method `getOrGuessChannelGroup`. Which first checks if it's been set, and then searches through existing configs to see if any of them have names that could sensibly be interpreted as a channel group.

https://github.com/tlambert03/pymmcore-plus/blob/62416c797ec40a7c7614fc16061f1928ced8382d/pymmcore_plus/core/_mmcore_plus.py#L393

By default we check for this regex:

```python
re.compile("(chan{1,2}(el)?|filt(er)?)s?", re.IGNORECASE)
```

which is also settable by the user here: https://github.com/tlambert03/pymmcore-plus/blob/62416c797ec40a7c7614fc16061f1928ced8382d/pymmcore_plus/core/_mmcore_plus.py#L361-L374

From https://github.com/tlambert03/napari-micromanager/issues/97#issuecomment-1056078644 I think that this maybe would be a generally useful thing to have in core. thoughts on this? 

cc @henrypinkard

nicost, 2022-03-02:
Instead of guessing, why not ask the software operator?  You can even store the answer in the configuration file so that you have to ask it only once. Putting such guesses in the Core look very scary to me.

marktsuchida, 2022-03-02:
I agree with @nicost on this one. This is a user interface level feature, and it is best if MMCore doesn't try to be smart.

ianhi, 2022-03-02:
> Instead of guessing, why not ask the software operator?

that is.... very reasonable :thinking:.  I think we even do that if that guessing doesn't return anything.

closed 2022-03-02 by ianhi