micro-manager/mmCoreAndDevices #172: Should `getAllowedPropertyValues` for properties `State` and `Label` of a StateDevice perhaps show the allowed states, labels?

tlambert03, 2022-03-04:
if one uses `core.getAllowedPropertyValues(stateDevice, 'State')` (or `..., 'Label'`) on a state device, you very often get an empty list (not sure if there are some devices that give the proper values, but many don't.  However, if you try to set that property (`core.setProperty(stateDevice, 'State', '1000')`) you may or may not get an error depending on whether you picked a valid State index or label.

would it be a relatively easy fix to ensure that `getAllowedPropertyValues` actually returned the allowed property values for state device `State` and `Label` properties?

marktsuchida, 2022-03-04:
Properties can have one of 3 types of value constraints:
- No constraint (shows up as text box in Device Property Browser)
- Multiple-choice (dropdown)
- Numerical range (slider)

The choice is orthogonal to the property data type (integer, float, or string), except that strings cannot have a range.

In a rather brazen violation of the separation of view from model, the choice is often made based on the preferred UI element. So you will see integer properties with range 0 to 1 (inclusive) because it is easier to click on the arrows of a slider than to toggle using a dropdown.

There is no restriction about which of these 3 options is used by the standard `State` property.

In the API, everything with "allowed values" in the name concerns multiple-choice constraints, and everything with "limits" in the name concerns numerical range constraints. If the `State` has a numerical range (e.g. `0` to `255`), you will get an empty list from `getAllowedPropertyValues()`, because the property is not a multiple-choice property.

So you need to first call `hasPropertyLimits()`, and use `getAllowedPropertyValues()` only in the case that the property has no (contiguous) range. If `hasPropertyLimits()` returns `false` and `getAllowedPropertyValues()` returns an empty list, then the property value is not constrained.

There is definitely room from improvement in the API.

For unconstrained integer properties, the valid integer range is not clearly specified. Due to the unfortunate use of `long` in C++ (which is 64 bits on 64-bit Unix but 32 bits on 64-bit Windows), the behavior is likely different between platforms. Also, many device adapters are written by programmers who have seldom worked with non-Microsoft platforms, so may have integer overflow bugs if passed a 64-bit value even when running on Unix. Avoid setting anything outside the range of `int32_t`, but maybe tolerate values outside of that range when reading (though unlikely in practice).

tlambert03, 2022-03-04:
> So you need to first call hasPropertyLimits(), and use getAllowedPropertyValues() only in the case that the property has no (contiguous) range. If hasPropertyLimits() returns false and getAllowedPropertyValues() returns an empty list, then the property value is not constrained.

what I'm saying though is that it _is_ in many cases still practically constrained with state devices:

here's an example using the demo MMConfig, 

```python
In [8]: core.hasPropertyLimits('Path', 'State')
Out[8]: False

In [9]: core.getAllowedPropertyValues('Path', 'State')
Out[9]: ()

In [11]: core.getPropertyType('Path', 'State')
Out[11]: <PropertyType.Integer: 3>

# from your logic above, it appears that we have an unconstrained integer property here...
# however

In [7]: core.setProperty('Path', 'State', '100')

RuntimeError: Cannot set property "State" to "100"
```

So, in addition to the checks you proposed above (checking hasPropLimits, and AllowedPropValues), one really also needs to check whether `core.getDeviceType()` is a state device and the property name is one of the special 'State' or 'Label' strings

marktsuchida, 2022-03-04:
I forgot to mention that it is also possible to get errors for particular values of an _unconstrained_ property, if the device adapter so chooses. This may be seen with integer properties where a slider would be inappropriate, but where the value has to be, say, non-negative. There is not much we can do in these cases (without updating every device adapter, at least). There are also some cases where the device adapter doesn't know if the value will work until it sends it to the device.

If the property does have a formal constraint, it is enforced by the Core (actually the code might be in `MMDevice`, but conceptually it should be in the Core).

tlambert03, 2022-03-04:
I recognize that there's an independent `core.setState` API... my only point here is that, from the perspective of properties, there essentially is no sequence of commands you can run querying the property type, range, or enumerations...

tlambert03, 2022-03-04:
if it helps to clarify, what I'm saying is that this it seems like this better logic for getAllowedPropertyValues (in python for simplicity)

```python
def getAllowedPropertyValues(dev, prop):
    allowed = core.getAllowedPropertyValues(dev, prop)
    if not allowed and core.getDeviceType(dev) == StateDevice:
        if prop == 'Label':
            return core.getStateLabels(dev)
        if prop == 'State':
            return tuple(str(i) for i in range(core.getNumberOfStates(dev)))
    return allowed
```

this all assumes that the string property labels `Label` and `State` are actually "special" to mmcore in some way (it appears that way, but I haven't verified in the source)

tlambert03, 2022-03-04:
ahh ... those strings are specific to DemoCam, yes?

marktsuchida, 2022-03-04:
First of all, if you are seeing a state device that has a `Label` property with no set of allowed values, it is almost certainly a bug in that particular device.

Second, what is lacking from the `State` property (of the particular device you are looking at, among many others) is not the set of allowed values, but a numerical range. So `getAllowedPropertyValues()` should not change; the candidates for special casing are the `Limit` functions. (Especially in light of devices possibly having very large numbers of states: https://github.com/micro-manager/micro-manager/issues/1418.)

Aside from that, I haven't totally made up my mind what the best solution is, but here are my thoughts so far.

State devices with no constraint on the `State` property appear to be the large majority, I think because a human user was expected to use `Label` instead. The reason we've been able to live with `State`s with an unspecified range so far is because the `Label` property is good to use in the case of most switcher devices; for devices that map `State` to a digital output port (typically 8-bit), users who know how to set a binary mask also know what the range (typically 0-255) is.

If we are to special-case `State` in the Core, we need to override the limits, not allowed values. Something like this pseudocode:
```py
def should_override_property_limits_for_state(device, prop):
    return (device.GetType() == StateDevice and
        not device.HasPropertyLimits(prop) and
        device.GetAllowedValues(prop).empty())

def hasPropertyLimits(devLabel, prop):
    device = GetDevice(devLabel)
    if should_override_property_limits_for_state(device, prop):
        return true
    return device.HasPropertyLimits(prop):

def getPropertyUpperLimit(devLabel, prop):
    device = GetDevice(devLabel)
    if should_override_property_limits_for_state(device, prop):
        return device.GetNumberOfStates() - 1
    return device.GetPropertyUpperLimit(prop)

def getPropertyLowerLimit(devLabel, prop):
    device = GetDevice(devLabel)
    if should_override_property_limits_for_state(device, prop):
        return 0
    return device.GetPropertyLowerLimit(prop)
````

I think this has a good chance of being okay, but I would not do it without reviewing all extant state devices -- there may be a few weird ones. Some concerns are:

- Devices' implementations of `getNumberOfStates()` may not be very well tested (in MMStudio, only the Hardware Configuration Wizard calls this function; almost everything else in MM uses the `State` property).
- Based on actual usage, it is the `State` property that is primary, and functions like `getState()` and `setState()` are rarely used. This special-casing would be a step in the opposite direction.
- To really pay off technical debt (i.e., the non-uniform behavior of `State` between different state devices), the place to fix is the `CStateDeviceBase` code in `MMDevice` (some of which should perhaps live in `MMCore`). But this is a much harder change to make, due to the tighter coupling with device adapter code.
- Hiding the behavior by patching the Core simplifies things for app developers but adds to the already-significant confusion of device adapter developers (probably a minor issue here).

In any case, this is effectively a proposal to also change the UI widgets in MMStudio's Device Property Browser, unless we first do the work to decouple widget choice from constraint type (which I think is doable and would be highly beneficial). Forcing all text box `State`s to sliders is an improvement in the typical case, and I _think_ it is safe for all devices, but I need to spend some time brooding over the device code we have before I can convince myself.

Sorry to be inconclusive -- I might make up my mind if we let this incubate for a few days....

tlambert03, 2022-03-04:
> Sorry to be inconclusive -- I might make up my mind if we let this incubate for a few days....

no apology needed!  very helpful thoughts, thank you.  Definitely not urgent, and I might be inclined to dig through some of the adapter code to see how common this sort of thing is.   I agree there are some possibly beneficial changes to be made in here, but that they're not necessarily "light".  For my part, I have reasonable workarounds

thanks again!

marktsuchida, 2022-03-04:
I compared the number of grep matches of `GetNumberOfPositions` (= 140, approx. number of state devices classes) and grep matches to `SetPropertyLimits(MM::g_Keyword_State` (= 19), although some adapters might hard-code `"State"`. There are 222 matches of `AddAllowedValue(MM::g_Keyword_State`, which might be ~60 devices (very rough count), and 3 matches of `SetAllowedValues(MM::g_Keyword_State`.

So, very roughly, it looks like 15% have a range, 45% have a list, and the rest have neither. But I suspect most of the ones that have a list have no good reason to prefer a list over a range (except possibly for consistency with other properties of the device).

I'm mostly only worried about peculiar devices that don't fall into the typical categories of turrets/wheels or LED/laser wavelength selection, and any that may alter the number of states dynamically (don't know if any exist). That should cut down the actual review work by quite a bit.