micro-manager/mmCoreAndDevices #186: update libusb support

artmeln, 2022-04-20:
I’m writing a device adapter that uses libusb as a library. After looking at 3rdpartypublic, it looks like libusb versions utilized by MM device adapters in the past are rather outdated. Would it be possible to add libusb-1.0 to 3rdpartypublic? It is the current platform independent version of the library.

marktsuchida, 2022-04-28:
I think we can make use of libusb 1.0 without necessarily updating the existing device adapters that need libusb 0.1 (aka libusb-compat). If you contribute a device adapter that uses libusb-1.0, we should be able to add a Windows binary to 3rdpartypublic (and a scripted build for macOS).

Let's not add dependencies until they are actually used, however.

Also, it will be good to make sure that talking directly to libusb is the right thing to do, i.e., that you are not reinventing what USBManager does.

artmeln, 2022-06-24:
libusb (version 1.0.26) has been made available in 3rdpartypublic

closed 2022-06-24 by artmeln