micro-manager/mmCoreAndDevices #207: Projects not in micromanager.sln

marktsuchida, 2022-05-26:
Does anybody know why the following projects are not in micromanager.sln?
- MicroFPGA (missing `.vcxproj.filters`; project needs `Win32` removed)
- MPBLaser (missing `.vcxproj.filters`; project needs `Win32` removed)

I found these by chance while looking at micro-manager/micro-manager.github.io#104. Want to make sure it's not a mistake.

marktsuchida, 2022-05-26:
In case this comes up in the future, here is how to list all projects that are not in micromanager.sln:
```sh
grep '^Project' micromanager.sln |cut -d , -f 2 |cut -d '"' -f 2 |tr \\ / |sort >in-sln
find . -name '*.vcxproj' |sed 's/^\.\///' |sort >not-in-sln
diff in-sln not-in-sln |grep '^>'
```

jdeschamps, 2022-11-06:
Are the filters necessary? I have seen that some other device adapters don't have `vcxproj.filters` (including some of mine).

marktsuchida, 2022-11-07:
Yes, the filters should be version controlled (unlike `.vcxproj.user`). The build works without them but I think the files don't display nicely in VS when they are missing. So adding the missing ones would be nice but not super urgent.

PR #347 (2023-05-11): Add MPBLaser to build
See micro-manager/micro-manager#615.
Closes #207.
closed 2023-05-11 by marktsuchida