micro-manager/mmCoreAndDevices #209: Bug: MM crashed at shutdown with Spinnaker cameras on different interfaces

ieivanov, 2022-05-27:
Micro-manager crashes at shutdown then two Spinnaker cameras on different interfaces (e.g. USB and GigE) are used at the same time. The two cameras initialize and run properly - the crash only happens when Micro-manager is closed. My guess is that the two interface instances are not cleaned up properly. I'm happy to test if someone can help debug this.

Here is an example from the documentation which may help: http://softwareservices.flir.com/Spinnaker/latest/_enumeration_8cpp-example.html

nicost, 2022-06-03:
Problem is that debugging needs access to two such cameras.  If you can lend me or Elliot Steele a set, then we would have a chance to figure out what is happening.

ieivanov, 2022-06-23:
Hi @nicost, sorry for the late reply. I think we can arrange for that. Would you have bandwidth to do this troubleshooting? We can either ship you the cameras or drop them off next time we're nearby.

nicost, 2022-06-27:
Once configured it is probably not too hard.  Can we schedule something for the week of July 11 or so?  Let's take this offline;)

ieivanov, 2022-10-27:
Hi @nicost, sorry I didn't follow up on this in time. Would you still have bandwidth to help us with this issue? We can easily set you up with access to a remote workstation which has two such cameras. Please let me know and we can coordinate offline.