micro-manager/mmCoreAndDevices #210: Question on debugging with new VS

henrypinkard, 2022-05-27:
I've got a fresh device adapter set up with no extra dependencies for now, an dI'm trying to debug it.

When I attach to javaw.exe on the splash screen, my breakpoint says "This breakpoint will not be hit. No symbols have been loaded for this document"

and then when I click ok and load the hardware config in MM, I get:

```
Exception thrown at 0x0000000002E03262 in javaw.exe: 0xC0000005: Access violation reading location 0x0000000000000008.
```

Any ideas on what this means and how to fix?

nicost, 2022-05-27:
You can ignore those.  The JVM throws exceptions, and the debugger catches them, but they are immaterial.  There is an arrow or something similar in that dialog and also an option to ignore this error in the future.

marktsuchida, 2022-05-27:
This is a frequently asked question (not specific to VS2019, but in any situation when using a C or C++ debugger with a JVM process), so here are some more details.

Note that it says "thrown", not "uncaught". The JVM is about to catch and handle this exception.

The setting to ignore these can also be accessed at **Debug** > **Windows** > **Exception Settings**, which gives you a tree view of exceptions for "Break When Thrown". Under **Win32 Exceptions**, uncheck **0xc0000005 Access Violation**.

Microsoft "exceptions" in this context include what Unix would use signals for in addition to C++ exceptions, i.e., problems with a program that the operating system detected and that result in process termination by default unless a special handler is set up (which the JVM does). Microsoft "access violation" means the same thing as Unix "segmentation fault", i.e., trying to access a memory address (in this case `0`) that is not in the process's address space. The JVM, for performance reasons, lets the operating system (ultimately the CPU) detect null-pointer dereferences that happen internally and converts them into Java `NullPointerException`s.

closed 2022-05-27 by marktsuchida

henrypinkard, 2022-05-28:
Thanks for the clarifications. One more question: I've got a shell of a device adapter set up, and can debug it. However, when I start adding in calls to other drivers, it still works on Release, but not Debug. On debug the HCW cannot load the adapter and I get the error:

```[ The module, or a module it depends upon, could not be found (Windows error: The specified module could not be found.) ] ]```

Any ideas what might cause this difference? I should be able to use the same drivers whether the device adapter is debug or release, right? That is, I don't need to switch to a debug version of the driver if I only want to debug the device adapter itself, right?

marktsuchida, 2022-05-28:
If the "other drivers" are C++ libraries (as opposed to C libraries), you would likely need a debug version of them to use with a Debug build (which some vendors provide; many don't).

(Micro-Manager device adapters can be Debug even if MMCore is Release, but that is because the interface is carefully designed to work across different configurations and compiler versions.)

That's the general answer, but there could be other stuff going on depending on ... details.

nicost, 2022-05-31:
To add to this, I regularly debug Realease builds (either when I do not have debug versions of dlls the code relies on, or just because I am lazy).  It usually works well enough, although stepping through the code is a bit less logical (presumably because of code optimization, etc..).

marktsuchida, 2022-05-31:
Good point. You can temporarily turn off optimization in the Release configuration to help with debugging. (The Debug configuration, in addition to not being optimized by default, uses a special runtime (standard library) to catch malloc-related issues, etc., which is one reason why it becomes incompatible with other C++ binaries.)

henrypinkard, 2022-05-31:
Ah good to know thanks for the tips! I managed to get debug copies of dependencies so it is working for now, but always good to get more insight into these things.

> (Micro-Manager device adapters can be Debug even if MMCore is Release, but that is because the interface is carefully designed to work across different configurations and compiler versions.)

Is this just because it is a pure C api, or are there more special things?

marktsuchida, 2022-06-01:
Usually C APIs are preferred for this purpose, but it is neither necessary nor sufficient, strictly speaking. The API must ensure that memory allocated in one DLL is released in the same DLL (so that `malloc` and `free` come from the same runtime), and that no non-[POD](https://en.cppreference.com/w/cpp/named_req/PODType) C++ objects are passed (because their memory layout may change depending on the standard library used). Exceptions cannot be thrown, and run-time type information cannot be used across DLL boundaries. The MMDevice API relies on the fact that the C++ object and vtable layout is stable for all Microsoft compilers. Using C makes it easier to prevent non-experts from breaking these rules, especially in view of the fact that the restricted C++ features that could be used make it look more or less like a C-style API anyway.