micro-manager/mmCoreAndDevices #219: path to 3rd party

henrypinkard, 2022-07-08:
Trying to setup the paths for a device adapter, and when I include `$(MM_3RDPARTYPRIVATE)` it resolves to a path with an extra `\` in `VisualStudio\\`. I think this is inherited from a project default so changing it doesn't seem like the right solution. Any ideas what might be going on?

@marktsuchida @nicost 

[image]

marktsuchida, 2022-07-08:
Does it actually affect the build?
Usually `$(MM_3RDPARTYPRIVATE)\path\to\libdir` works in the context you are showing.

henrypinkard, 2022-07-08:
Whoops, you're right. I guess that was an unrelated error I was seeing. Thanks!

closed 2022-07-08 by henrypinkard