micro-manager/mmCoreAndDevices #221: jvm.pdb not loaded when debugging

henrypinkard, 2022-07-20:
I'm working on a device adapter and this error has started popping up when debugging in VS2019:

[image]

and then subsequently:

[image]

This only started after I updated to the latest nightly build (using the exact same source I've been using with no issue) in order to run the studio code from intellij. Reinstalling the previous nightly build that was working before (5-26) does not fix the issue. It doesn't seem to affect the functioning of the device adapter. If I don't enter debug mode, no detectable error arises.

Any insights?

nicost, 2022-07-20:
What happens when you uncheck the "Break when this exception is thrown" checkbox and continue?

henrypinkard, 2022-07-20:
Just did that and it seems to function normally. Good enough for developing I guess. But still seems weird that this started cropping up...

henrypinkard, 2022-07-20:
Also, not if sure if this is related or not, and I'm not 100% sure about it, but I think previously when I pressed "stop debugging" in VS it would close MM, but now it just stops debugging but keeps MM open

nicost, 2022-07-20:
Mac OS X used to have these.  These are exceptions in the JVM.  It is possible that you use a different JVM, or something similar.  I believe that the JVM uses exceptions in its normal functioning, so these are of no concern. Easiest path seems to be to simply ignore these things.

nicost, 2022-07-20:
I always keeps it open for me (as it should, if you use "attach to process" and the application closes when you detach, that suggests a bug to me.

henrypinkard, 2022-07-20:
Ok good to know. So it sounds like the error message started popping up to alert me that I no longer had a bug :)

In any case, it all seems to work. Thanks!

closed 2022-07-20 by henrypinkard

henrypinkard, 2022-07-20:
Thanks! Sorry for the duplicate. I've committed it to memory this time :)

marktsuchida, 2022-07-21:
No problem, didn't mean to sound like a complaint.