micro-manager/mmCoreAndDevices #222: Add ability to acquire arbitrary data to MM

artmeln, 2022-07-24:
I’d like to open a discussion about adding to MM the ability to retrieve arbitrary data from devices (following up on an earlier [discussion](https://github.com/micro-manager/mmCoreAndDevices/issues/35)). This would allow one to add oscilloscopes, spectrometers, photon counting devices, etc to the family of devices supported by MM.

First, we need to decide how this change would fit into the MM architecture. Based on previous discussions, there are several options:
1. Modifying GenericDevice to give it (optional) ability to acquire data.
2. Adding a new kind of device dedicated to data acquisition.
3. Modifying DeviceBase with data acquisition calls optional for implementation.
4. Adding arbitrary data as a new kind of MM Property (called byteStream).
5. Adding a data acquisition entity that is modeled after MM Property but is distinct.
Personally, I favor the last solution as it would give every device the ability to transfer data, possibly through multiple channels. Below I’m calling this proposed class DataChannel.

Second, we need to decide on specific calls that will implement data acquisition functionality and expose it to the java layer. The raw acquired data will not necessarily be useful to the user as is, but ultimately should result in some sort of graphical display. Therefore it should be processed into arrays that can be used for plotting. I’m proposing the following calls for
1. Controlling acquisition:
    • StartDataChannel – start collection
    • StopDataChannel – stop collection
    • GetDataChannelBufferSize  - returns data size in bytes
    • GetDataChannelBuffer – retrieves data from the hardware and returns the pointer to it
    • GetDataChannelOverflow – overflow flag indicating that data was lost by the hardware before it could be transferred
    • ResetDataChannelOverflow – resets overflow flag
2. Processing acquired data:
    • ProcessBuffer() – transform the obtained data buffer into arrays that can be put on a graph;
    • GetArrayNames() - obtain names of all arrays generated by ProcessBuffer call
    • GetArray(String arrayName)  - obtain specific array by name
    • GetArrayType(String arrayName) - float, int, etc.
    • GetArrayDimensions(String arrayName) - for example, 2 for a XY plot
    • GetArrayShape(String arrayName) - returns integers describing lengths of array dimensions.
    • GetArrayDimensionNames(String arrayName) – returns names for the axes.

Once the best path forward for adding this functionality has been established, I’d be happy to work on implementing it, I have a strong and continued interest in adding this functionality to MM.

marktsuchida, 2022-07-26:
Thanks for moving this forward.

> * Modifying GenericDevice to give it (optional) ability to acquire data.
> * Adding a new kind of device dedicated to data acquisition.
> * Modifying DeviceBase with data acquisition calls optional for implementation.
> * Adding arbitrary data as a new kind of MM Property (called byteStream).
> * Adding a data acquisition entity that is modeled after MM Property but is distinct.
>   Personally, I favor the last solution as it would give every device the ability to transfer data, possibly through multiple channels. Below I’m calling this proposed class DataChannel.

My preference is 2. It fits into the existing way of doing things, is easy to understand, and will not affect any existing code. All the other options will create a special-purpose island of APIs that need special treatment because they don't behave like everything else.

(Whether a single device of this new type can have multiple channels, and whether those channels can be used independently, is up for separate debate.)

You seem to really prefer a solution that can be added to any device, but you haven't described your motivation for that (apologies if I missed it). To me, it is not clear why one would want a "shutter that also acquires arbitrary data" or a "camera that also acquires arbitrary data", especially given the existing hub-peripheral mechanism. Of course it would be more flexible, but flexibility is a bad thing when you are on the side that has to deal with all the possible cases. Can you come up with a concrete reason for wanting every device to potentially acquire streamed data? If you have a TCSPC card, for example, it makes sense to have a single `MM::Device` for it, but it doesn't need to be a camera or shutter or stage at the same time.

Just to be clear, it's not that I like the way MMCore currently does things. But it's better than an API that uses different models for different things for arbitrary reasons. Evolving the whole API would be the alternative, but that's hard and time-consuming.

Some further thoughts that came to mind:

I think processing is best kept out of the device adapter or MMCore, and left to Java and Python, unless it is proven that there are use cases where this is not enough.

There will be a need for finite-length acquisitions (e.g. 1000 samples). So there needs to be an API to tell when it's done.

Overflow detection is important, but it might be simpler to just have two options: stop (with an error) on overflow or ignore and continue. Otherwise I imagine application code duplicating the same logic everywhere.

Getting the buffer pointer and size from separate functions is not viable (one might change in the meantime). Actually getting the pointer is not viable (how long is it valid?).

artmeln, 2022-07-28:
I’m not opposed at all to introducing a new kind of device that will take care of data transfer needs. I definitely see the advantage in keeping things simple. My only argument for giving every MM device this functionality was making things more general but I’m happy to sacrifice this for simplicity. As for multiple channels, one channel per device is probably enough (multiple devices can be created and combined in a hub if necessary).

> I think processing is best kept out of the device adapter or MMCore, and left to Java and Python, unless it is proven that there are use cases where this is not enough.

The motivation was to allow Java/Python layer to treat all data transfer devices in a unified fashion. Different data transfer devices are likely to produce data that looks very different, but if it is to be immediately useful to the user, the result should be plotted. The upper layer of the software can then take on the job of getting dimensions and shapes of the processed data arrays and plotting them, regardless of which data transfer device produced them.

> There will be a need for finite-length acquisitions (e.g. 1000 samples). So there needs to be an API to tell when it's done.

Is it a good idea to do finite length acquisition by allowing data transfer device to be sequenceable?

> Getting the buffer pointer and size from separate functions is not viable (one might change in the meantime). Actually getting the pointer is not viable (how long is it valid?).

The idea was to model this after the camera device API. The device informs the adapter when it is has data for transfer (responds with bufferSize>0) and makes the commitment to not discard this data until it is retrieved through the GetBuffer call.

marktsuchida, 2022-08-22:
Sorry for the lack of reply. I'll definitely get to this this week.

artmeln, 2022-11-02:
I apologize for the long silence on the subject but I’m finally able to dedicate more time to this project. Here is a first (preliminary) stab at implementing the DataStreamer device https://github.com/artmeln/mmCoreAndDevices/tree/data-streamer-1.

At the moment, this new device has the following methods (defined in MMDevice.h): 
* int GetBufferSize(unsigned& dataBufferSiize);
* std::unique_ptr<char[]> GetBuffer(unsigned expectedDataBufferSize, unsigned& actualDataBufferSize);
* int ProcessBuffer(std::unique_ptr<char[]>& pDataBuffer, unsigned dataSize);
* int SetStreamParameters(bool stopOnOverflow, unsigned numberOfBuffers, double durationUs, double updatePeriodUs);
* int GetStreamParameters(bool& stopOnOverflow, unsigned& numberOfBuffers, double& durationUs, double& updatePeriodUs);
* int StartStream();
* int StopStream();
* int IsStreaming(unsigned& isStreaming);
* int SetCircularAcquisitionBufferCapacity(unsigned capacity);
* int GetCircularAcquisitionBufferCapacity(unsigned& capacity);

The first three are implemented by the user at the level of the device adapter, the rest have default implementations in DeviceBase.h and should not be overwritten.
The following methods are exposed to the java layer:
 - setCircularAcquisitionBufferCapacity;
 - getCircularAcquisitionBufferCapacity;
 - setStreamParameters;
 - getStreamParameters;
 - startStream;
 - stopStream;

The logic of DataStreamer operation is as follows. When StartStream is called, two threads are automatically started, one for data acquisition and one for data processing. The acquisition thread calls GetBufferSize with the period specified by updatePeriodUs stream parameter. When GetBufferSize call returns >0, the acquisition thread calls GetBuffer and puts the pointer to the data (along with data size) into a circular buffer. Then GetBufferSize is called again and so on. This cycle continues until the specified number of data buffers are collected, or the measurement time elapses, or the circular buffer overflows, or the GetBuffer call times out (fails to return >0); these parameters are set by  SetStreamParameters.

The acquisition thread only adds data to the circular buffer but never retrieves it. This is done by the processing thread which repeatedly calls ProcessBuffer on the first available entry in the circular buffer. The processing thread stops running only after the acquisition thread has stopped. The simplest example of processing is writing data to the hard drive although I still imagine creating standard data arrays that can be used for plotting. This idea was briefly discussed before, I haven’t tried to implement it yet.

The circular buffer accessed by both threads contains pointers to the data only; the data buffers themselves are supposed to be allocated by GetBuffer which is implemented at the level of the device adapter.

I’ve added an example DataStreamer device to DemoCamera.h that generates some mock data and saves it (through ProcessBuffer calls) to c:\temp\mybinaryfile.txt. The device can be tested from the MM scripting window by calling, for example,

mmc.setStreamParameters("DSimulatedDataStreamer",true,4,1e7,1e5);
mmc.startStream("DsimulatedDataStreamer");

I would very much appreciate everyone’s feedback on the direction of the project. There are additional functionalities I'd like to implement but this felt like a good point to pause and evaluate the progress.

artmeln, 2022-11-12:
I have noticed one thing that isn’t working correctly in this branch: getStreamParameters function. It looks like it is not correctly generated by SWIG, and I can’t figure out why.

In MMCore.h, I have the following prototype compared here to a well behaving function getROI:

void getROI(const char* label, int& x, int& y, int& xSize, int& ySize) throw (CMMError);

void getStreamParameters(const char* label, bool& stopOnOverflow, unsigned& numberOfBlocks, double& durationUs, double& updatePeriodUs) throw (CMMError);

The resulting functions autogenerated by SWIG are

SWIGEXPORT void JNICALL Java_mmcorej_MMCoreJJNI_CMMCore_1getROI_1_1SWIG_11(JNIEnv *jenv, jclass jcls, jlong jarg1, jobject jarg1_, jstring jarg2, jintArray jarg3, jintArray jarg4, jintArray jarg5, jintArray jarg6)

SWIGEXPORT void JNICALL Java_mmcorej_MMCoreJJNI_CMMCore_1getStreamParameters(JNIEnv *jenv, jclass jcls, jlong jarg1, jobject jarg1_, jstring jarg2, jlong jarg3, jlong jarg4, jlong jarg5, jlong jarg6)

It looks like in the first case int& is interpreted as jintArray but in my case it is always jlong. I tried changing the parameter types to int& but that didn’t seem to help. Does anyone have any ideas why this isn’t working?

marktsuchida, 2022-11-14:
Sorry I haven't had a chance to look at your proposal above (will do as soon as I have a chance), but regarding your SWIG question: are you using SWIG 3.x (which is currently unfortunately required for MMCoreJ)?
Also which platform are you testing on?

artmeln, 2022-11-16:
No worries, there is no rush.
I'm testing on Windows 10 and using SWIG 2.
I think I figured out what the problem was. There are special rules in MMCoreJ.i for these kinds of parameters, and after adding them everything is working correctly.