micro-manager/mmCoreAndDevices #224: MicroManager does not parse data channels properly when using MultiCamera adapter

ieivanov, 2022-07-26:
We would like to acquire data using the Multi-Dimensional Acquisition window in 3 channels, two of which are acquired simultaneously using the Multi Camera adapter (e.g. transmission and GFP), and the third is acquired in a separate channel (e.g. RFP). The channel configuration looks something like this:

# Group: Channel
# Preset: TL-GFP
ConfigGroup,Channel,TL-GFP,FilterTurret,State,0
ConfigGroup,Channel,TL-GFP,Shutter,State,0
ConfigGroup,Channel,TL-GFP,Core,Camera,Multi Camera

# Preset: RFP
ConfigGroup,Channel,RFP,FilterTurret,State,1
ConfigGroup,Channel,RFP,Shutter,State,1
ConfigGroup,Channel,RFP,Core,Camera,Camera2

In this example, Multi Camera links Camera1 and Camera2.

The current MM acquisition engine returns data in 2 channels only, corresponding to data acquired by Camera1 in the TL-GFP channels and Camera2 in the RFP channel. As a result z-stack and time series data arrays are also not built properly.

marktsuchida, 2022-07-26:
I know this is not a solution, but does it make any difference if you reorder the two channels in the MDA channel list?

You mention that it "returns data in 2 channels only". Do you mean that there are 3 channels but one is blank, or that there are only 2 channels in the dataset? What do you mean by "data arrays are not built properly"?

It is possible that the acquisition engine makes assumptions about the cameras not changing in certain ways between channels. It might be necessary to use Multi Camera in both channels, with a redundant transmitted light image (if your setup allows this optically).

ieivanov, 2022-07-26:
Hi Mark,

>I know this is not a solution, but does it make any difference if you reorder the two channels in the MDA channel list?

Interestingly, that worked! The acquisition ran and completed. In this case, the MDA created a phantom channel with no image data in it, but that was not a big problem for us.

>You mention that it "returns data in 2 channels only". Do you mean that there are 3 channels but one is blank, or that there are only 2 channels in the dataset?

The dataset contained two channels only, with data acquired by Camera1 in the TL-GFP channels and Camera2 in the RFP channel in the above example.

>What do you mean by "data arrays are not built properly"?

I observed that the acquisition would run for a few timepoints, but then crash. The preview window did not show a slider for the time dimension. The same happened when acquiring z stacks.

Do you think it would be easy to fix this bug in the acquisition engine?

marktsuchida, 2022-07-26:
Glad that trick worked (I thought there was a chance).

It should be fixable, but not quite trivially (and there's the activation energy barrier since that code is written in Clojure, which I don't frequently deal with...). Let's keep this issue open; I'll transfer it to the micro-manager repository where it belongs.