micro-manager/mmCoreAndDevices #248: Cannot compile: type-conversion erros in MMCoreJ_wrap module

FilCo98, 2022-09-27:
I am trying to install the software but compiling fails at the MMCoreJ_wrap module, in particular in the CMMCore.java file. My compiler mostly type-conversion errors:

```
micromanager-git/src/micromanager-git/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:231: error: incompatible types: Long cannot be converted to int
 [mm-javac]          Rectangle r = new Rectangle((int) xs.get(i), (int) ys.get(i),

 [mm-javac] /home/phil/programs/micromanager-git/src/micromanager-git/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:231: error: incompatible types: Long cannot be converted to int
 [mm-javac]          Rectangle r = new Rectangle((int) xs.get(i), (int) ys.get(i),  

[mm-javac] /home/phil/programs/micromanager-git/src/micromanager-git/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:232: error: incompatible types: Long cannot be converted to int
 [mm-javac]                (int) widths.get(i), (int) heights.get(i));

[mm-javac] /home/phil/programs/micromanager-git/src/micromanager-git/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:232: error: incompatible types: Long cannot be converted to int
 [mm-javac]                (int) widths.get(i), (int) heights.get(i));

[mm-javac] /home/phil/programs/micromanager-git/src/micromanager-git/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:248: error: incompatible types: int cannot be converted to Long
 [mm-javac]          xs.add(r.x);

 [mm-javac] /home/phil/programs/micromanager-git/src/micromanager-git/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:249: error: incompatible types: int cannot be converted to Long
 [mm-javac]          ys.add(r.y);

 [mm-javac] /home/phil/programs/micromanager-git/src/micromanager-git/mmCoreAndDevices/MMCoreJ_wrap/gensrc/mmcorej/CMMCore.java:251: error: incompatible types: int cannot be converted to Long
 [mm-javac]          heights.add(r.height);  
```

I was wondering if this was happening to other people or how to fix it.

FilCo98, 2022-09-29:
I have installed the software manually and ensured that it was the one on PATH, but it is still not compiling for the same errors. It may be the way Arch Linux handles the command make. Anyway thank you for your support.

closed 2022-09-29 by FilCo98

marktsuchida, 2022-09-29:
You might need to do a clean rebuild (in this case `make clean` should be sufficient to remove the MMCoreJ previously built by SWIG 4).