micro-manager/mmCoreAndDevices #255: Issue building MMCoreJ wrap

henrypinkard, 2022-10-09:
As I'm making changes to MMDevice/MMCore etc., I need to rebuild MMCoreJ for testing purposes. I'm running into some issue with SWIG that I can't figure out. 

The steps I'm taking are:
 - Build MMCoreJ_wrap.sln in VS2019
 - Run `ant jar` in `\mmCoreAndDevices\MMCoreJ_wrap`

Though there are no errors in this process, the MMCoreJ.jar that is produced contains only JSON and TaggedImage (i.e. not MMCoreJ.java or any of the other classes like StrVector.java).

`MMCoreAndDevices/build/intermediates` looks like this:

```
├───Release
│   └───x64
│       ├───MMCore
│       │   └───MMCore.tlog
│       ├───MMCoreJ_wrap
│       │   ├───MMCoreJ_wrap.tlog
│       │   └───Swig
│       └───MMDevice-SharedRuntime
│           └───MMDevice.B8C95F39.tlog
└───Swig
    └───mmcorej
```

All the expected classes are present in `build/intermediates/Swig/mmcorej` and `build/intermediates\Release\x64\MMCoreJ_wrap\Swig\`

I've tried all the instructions on the website a couple times and various `ant clean`s etc. I ran into the same problem a week ago and somehow got it to work, but I wasn't sure what I changed that did it.

Any ideas @marktsuchida ?

marktsuchida, 2022-10-10:
Hmm, it looks like yet another thing that was left broken when splitting `mmCoreAndDevices` from `micro-manager`. I can reproduce it.

The problem is that when building `MMCoreJ_wrap.vcxproj` in Visual Studio, it now places the SWIG outputs in `mmCoreAndDevices/build/intermediates`, whereas the Ant build for MMCoreJ still expects them in `build/intermediates`.

The reason it works in the automated builds is that MSBuild (the command-line build tool for Visual Studio) is invoked from inside Ant in that case, setting `MM_BUILDDIR` to `build/` (outside of `mmCoreAndDevices`). But building C++ from Ant is inconvenient during development (you can only build the whole solution).

I think I can fix this; working on a PR right now.

henrypinkard, 2022-10-10:
Awesome, thank you!!

I guess I built the entire thing from Ant before without realizing that was the key step

closed 2022-10-10 by marktsuchida