micro-manager/mmCoreAndDevices #284: Remove deprecated functions from MMCore

marktsuchida, 2022-10-28:
Nothing urgent here, but it will be nice to clean up. Among other things, it will get rid of deprecation warnings when compiling MMCore.

- Old-style device adapter path settings (deprecated since 2014)
```c++
   MMCORE_DEPRECATED(static void addSearchPath(const char *path));
   MMCORE_DEPRECATED(static std::vector<std::string> getDeviceLibraries() throw (CMMError));
```

- ImageSynchro (deprecated since Mar 2022 but unused and unmaintained for a decade)
```c++
   MMCORE_DEPRECATED(void waitForImageSynchro() throw (CMMError));
   MMCORE_DEPRECATED(void assignImageSynchro(const char* deviceLabel) throw (CMMError));
   MMCORE_DEPRECATED(void removeImageSynchro(const char* deviceLabel) throw (CMMError));
   MMCORE_DEPRECATED(void removeImageSynchroAll());
   MM::g_CFGCommand_ImageSynchro
```

- PropertyBlock (deprecated since Mar 2022 but unused and unmaintained for a decade)
```c++
   MMCORE_DEPRECATED(void definePropertyBlock(const char* blockName, const char* propertyName,
   MMCORE_DEPRECATED(std::vector<std::string> getAvailablePropertyBlocks() const);
   MMCORE_DEPRECATED(PropertyBlock getPropertyBlockData(const char* blockName));
   MMCORE_DEPRECATED(PropertyBlock getStateLabelData(const char* stateDeviceLabel,
   MMCORE_DEPRECATED(PropertyBlock getData(const char* stateDeviceLabel));
   MM::g_CFGCommand_Equipment
```

- In addition, I propose that we also remove the following functions, pending micro-manager/micro-manager#1534
```c++
   std::string getUserId() const;
   std::string getHostName() const;
   std::vector<std::string> getMACAddresses(void);
```
These never really belonged in MMCore (they are only there because it was a convenient place to get C++ functions wrapped for Java) and these days there are better alternatives in Java and Python. It seems like a good time to remove if we're bumping the MMCore major version anyway. Removing these has a small benefit of simplifying the build config on macOS.

Cc: @nicost, @henrypinkard.

henrypinkard, 2022-10-31:
Yes agreed! 

Maybe also good to consider deprecating or removing in core callback:

```c++
   // autofocus support
   const char* GetImage();
   int GetImageDimensions(int& width, int& height, int& depth);
   int GetFocusPosition(double& pos);
   int SetFocusPosition(double pos);
   int MoveFocus(double v);
   int SetXYPosition(double x, double y);
   int GetXYPosition(double& x, double& y);
   int MoveXYStage(double vX, double vY);
   int SetExposure(double expMs);
   int GetExposure(double& expMs);
   int SetConfig(const char* group, const char* name);
   int GetCurrentConfig(const char* group, int bufLen, char* name);
   int GetChannelConfig(char* channelConfigName, const unsigned int channelConfigIterator);
```

(I haven't yet thoroughly checked if any device adapters use these)

marktsuchida, 2022-11-07:
#295 (Thanks, Henry!).

PR #396 (2023-11-02): MMCore: Remove deprecated functions
Closes #284.

This increments the MMCore API version to 11.0.0, since it removes existing functionality.
All the removed functions have been deprecated for at least 1 year (some for 9 years).

Internal code and build settings related to the removed functions have also been removed.

Removed public functions:
```c++
MMCORE_DEPRECATED(static void addSearchPath(const char *path));
MMCORE_DEPRECATED(static std::vector<std::string> getDeviceLibraries() throw (CMMError));

MMCORE_DEPRECATED(void waitForImageSynchro() throw (CMMError));
MMCORE_DEPRECATED(void assig [... cut]
closed 2023-11-02 by marktsuchida