micro-manager/mmCoreAndDevices #288: Something on Oct 25 commits broke pymmcore

tlambert03, 2022-10-30:
with the nightly release from 20221024, the following code runs fine:

```python
from pymmcore import CMMCore
core = CMMCore()
MM_PATH = 'path/to/micromanager'
core.setDeviceAdapterSearchPaths([MM_PATH])
core.loadSystemConfiguration(f'{MM_PATH}/MMConfig_demo.cfg')
```

but the nightly from 20221025, now throws a segfault.  Haven't had time to dig into it yet
cc @marktsuchida

marktsuchida, 2022-10-31:
Thanks, I'll look into this ASAP.

marktsuchida, 2022-10-31:
My bad. I broke the MMDevice ABI by changing the representation of `MM::MMTime`. Obvious now that I look at it; my head must not have been screwed in right. Will have a fix soon.

marktsuchida, 2022-10-31:
I was hoping to make these changes to `MMTime` without incrementing the device interface version, but somehow at the last minute it slipped my mind that I can't change the data layout of `MMTime` because it is passed from Core to device.

There are two possible ways to fix this:

- Increment the device interface version from 70 to 71.
- Revert `MMTime` to use `long sec_; long uSec_;` instead of `long long microseconds_;`.

I think at this point I prefer incrementing the interface version, because some edge cases will need to be dealt with if we go back to the pair of `long`s (we need to make sure the seconds doesn't overflow a 32-bit integer). Not to mention the fact that the code will be much more complicated and error-prone.

I will make this change this afternoon unless there are other opinions. Cc: @nicost, @henrypinkard.
After that I can release pymmcore 10.3.0.71.0 which should then work with nightly builds starting tonight's.

marktsuchida, 2022-10-31:
BTW This issue does not affect anybody using a single Micro-Manager build without pymmcore, because the MMCore and device adapters packaged together are compatible with each other.

nicost, 2022-10-31:
Agreed.  Better to increase device version.  I do have a practical problem with some vendors building their own version of the device adapter (Gataca, Mightex).  It would also be useful to get the Crest V3 into the repo (but I see their developer did not respond, will ping them). Hopefully I can use this to convince them of the usefulness of putting this code out there (or at least so that they will understand there is more work for them by not adding it to our repos).

tlambert03, 2022-10-31:
thanks for the quick response @marktsuchida!

here's a naive question: would it be hard to setup a CI compile & test workflow here?  Or is it likely to be something that would take a lot of time to setup, only to catch a very rare problem?  If you think it's feasible, I would be willing to take a look at it.

marktsuchida, 2022-10-31:
@tlambert03 Yeah, I was thinking the same thing, but it seems non-trivial to do as unit tests. We need to test that virtual member functions appear in the same order in device classes and the core callback class, and that all their parameters and return values have not changed. In the rare case where a parameter or return value is a struct (as with `MMTime`), we also need to ensure that the struct layout has not changed. This time, the size of `MMTime` changed on macOS (16 -> 8 bytes) but not on Windows (8 -> 8 bytes), so it may not have crashed on Windows, but it would have wildly misbehaved when computing delays, perhaps resulting in behavior like snaps taking forever (or just random durations).

What could be done is to scan `MMDevice.h` with libclang (or equivalent) and check that certain things have not changed relative to the reference `MMDevice.h` for that device interface version. But I don't know if that level of effort is worth putting into this (perhaps yes if we find ourselves making more frequent changes going forward).

We could also just have a sort of smoke test where we load a reference DemoCamera DLL and exercise it a bit. That would have caught the problem this time, though only by luck (because DemoCamera happens to use `MMTime` in certain ways). It's hard to say what percentage of the time inadvertent changes will be caught.

A third approach would be to create mock devices against a reference `MMDevice.h` and make sure that every function and callback produces the expected result. But I don't know if there is a way, using a C++ mocking framework, to build the mocks and the tests against different copies of `MMDevice.h` and still be able to check that the mocks were called as expected. The direct comparison of AST using libclang seems simpler to me.

And in all cases, when a deliberate change to the ABI _is_ made, we also need to check that it follows certain rules: for example, all parameters and return values need to be C-style plain-old data (to slightly oversimplify).

I think it can be said that this was a pretty subtle case, where I was deliberately making source-incompatible changes that were nonetheless meant to be binary-compatible (until I changed my plans, forgetting the constraint). And I duped myself and possibly others by making the claim of binary compatibility. This kind of change has been very rare; in most cases changes to MMDevice involve adding (or, less commonly, removing) a function or class and the need to increment the version is evident.

marktsuchida, 2022-10-31:
Well, it turns out that I did not merge the `MMTime` struct layout change (#281) until 20221027. So I still need to find out what went wrong on 20221025. The pull requests affecting MMDevice, MMCore, or DemoCamera that were merged on that day are #274 and #277. I'm not seeing how these could have broken the ABI, so more investigation is needed (including for possible non-ABI bugs).

Let's go forward with the version increment anyway, because it is needed, and keep looking if anything else is broken.

tlambert03, 2022-10-31:
> We could also just have a sort of smoke test where we load a reference DemoCamera DLL and exercise it a bit. 

yeah, I guess that's what I was thinking... but I definitely appreciate that it would have only caught it by luck this time.  I guess with the heterogeneity across device adapters, that may always be the case though, so it's better than nothing?

For what it's worth, the pymmcore-plus tests might provide a reasonable pre-built smoke test; they run a number of standard MDA style operations on the demo adapter (I noticed this problem in the first place because our tests started failing there).

> I think it can be said that this was a pretty subtle case ... This kind of change has been very rare; 

Yep, I definitely hear you here!  Probably something to put on the back burner :)  but do ping me later if you think I could be helpful in setting up even a crude smoke test.

marktsuchida, 2022-10-31:
Thanks, @tlambert03. Perhaps a great time to look at this will be after the first 2-3 steps of our [build system migration plan](https://github.com/micro-manager/micro-manager/issues/1392#issuecomment-1156956436) are done, so that CI operations on MMCore and MMDevice are easier to set up. (Right now each platform will need a different script.)

marktsuchida, 2022-10-31:
Okay, `git bisect` tells me that the problem starts with this commit: 8ba45cde4436f8331f26a1adfac858ba08d51731 (#277).
Upon further testing, it looks like the culprit was the removal of the empty destructor `~MMTime() {}` -- if only that change is reverted, the crash is averted.

I don't know how to explain this, even after searching around a little. Even if calls to `~MMTime()` were not inlined by the compiler (unlikely), MMDevice is statically linked to both DemoCamera and MMCore, so each should just call their own copy of `~MMTime()` just fine. And both versions (explicitly empty or implicitly generated) do nothing in this case, so it's hard to see how they could crash. There must be something quite subtle going on here.

In any case, a pymmcore `main` build (with MMCore from after all the `MMTime` changes) does not crash with DemoCamera built from the above commit (or tip of `main`), so #289 should be a sufficient fix.

closed 2022-10-31 by marktsuchida

marktsuchida, 2022-10-31:
For CI detection of ABI break, tools like [this](https://lvc.github.io/abi-compliance-checker/) might also be worth looking into.

marktsuchida, 2022-10-31:
I figured out why removing `~MMTime()` caused the ABI to break, so recording for future reference. It turns out to be something I had heard about, once I connected the dots.

The compiler-provided destructor (when `~MMTime()` is not explicitly defined) is a [trivial destructor](https://en.cppreference.com/w/cpp/language/destructor#Trivial_destructor) in this particular case. But if `~MMCore() {}` is explicitly user-defined, it is not considered trivial.

Other conditions that hold with `MMTime` mean that whether the destructor is trivial determines whether `MMTime` is a [trivially copyable class](https://en.cppreference.com/w/cpp/language/classes#Trivially_copyable_class), and therefore whether it is a [trivial class](https://en.cppreference.com/w/cpp/language/classes#Trivial_class).

The Itanium C++ ABI, upon which the macOS and Linux x86_64 ABIs are based, specifies different calling conventions for trivial vs non-trivial return values. Therefore the return value of `GetCurrentMMTime()` (a function implemented in MMCore but called from devices) is placed in different locations. Specifically, the version with the empty user-defined destructor places the `MMTime` on the stack, whereas the version with no explicit destructor (or, equivalently, with a `= default` destructor) places it in a pair of registers (`rax` and `rdx`).

The difference can be seen in a [simplified example](https://godbolt.org/z/Eo19E4qfh).

Issues like these should not arise if we limit ourselves to passing only POD between DLLs. I think `GetCurrentMMTime()` returning `MMTime` is the only exception to that rule in MMDevice, so hopefully we don't have to worry about these things that much normally.

kasliwalr, 2022-12-18:
@marktsuchida I am having problems try to run this [pymmcore example](https://micro-manager.org/Using_the_Micro-Manager_python_library) on Ubuntu. It is throwing an error - seems its unable to file the demo camera shared libraries.
```
2022-12-17T15:44:27.693674 tid140069159765824 [IFO,LogManager] Enabled logging to stderr
2022-12-17T15:44:27.693693 tid140069159765824 [IFO,LogManager] Switching primary log level from info to trace
MMCore version 10.1.1
Device API version 70, Module API version 10
Traceback (most recent call last):
  File "/home/optics10x/repos/optics-10x/mm_integration/python/scripts/load_demo_camera.py", line 21, in <module>
    run_demo_camera(mmc)	
    ^^^^^^^^^^^^^^^^^^^^
  File "/home/optics10x/repos/optics-10x/mm_integration/python/scripts/load_demo_camera.py", line 7, in run_demo_camera
    mmc.loadDevice('Camera', 'DemoCamera', 'DCam')
RuntimeError: Failed to load device "DCam" from adapter module "DemoCamera"
2022-12-17T15:44:27.704103 tid140069159765824 [IFO,LogManager] Switched primary log level from info to trace
2022-12-17T15:44:27.704215 tid140069159765824 [dbg,Core] Will load device DCam from DemoCamera
2022-12-17T15:44:27.712947 tid140069159765824 [dbg,Core] Will unload all devices
2022-12-17T15:44:27.712952 tid140069159765824 [IFO,Core] Did unload all devices
2022-12-17T15:44:27.712967 tid140069159765824 [IFO,Core] System reset
2022-12-17T15:44:27.713268 tid140069159765824 [IFO,Core] Core session ended
```

I did a full installation of micro-manager (working off the main branch), after that I have been trying the following
```
# Modify Makefile.am
# Modify configure.ac
> cd micro-manager
> ./autogen.sh
> ./configure --without-java
> make fetchdeps
> make -j4
> sudo make install
```
 I can see that demo camera shared libraries end up in `/usr/local/lib/micro-manager/`, but when I try to run the following script, I get the above error.
```
import pymmcore
#import pymmcore-plus

def run_demo_camera(mmc: pymmcore.CMMCore):
	mmc.setDeviceAdapterSearchPaths('/usr/local/lib/micro-manager')
	#mmc.loadSystemConfiguration('<path_to_micro-manager>/micro-manager/bindist/any-platform/MMConfig_demo.cfg')
	mmc.loadDevice('Camera', 'DemoCamera', 'DCam')
	mmc.initializeAllDevices()
	mmc.setCameraDevice('Camera')
	mmc.snapImage()
	img = mmc.getImage()
	print(type(img))

if __name__=="__main__":
	mmc = pymmcore.CMMCore()  # Instance micromanager core
	# mmc = pymmcore_plus.CMMCore.instance()  # Instance micromanager core
	mmc.enableStderrLog(True)
	mmc.enableDebugLog(True)
	print(mmc.getVersionInfo()) # gives: 'MMCore version 2.3.2'
	print(mmc.getAPIVersionInfo()) # gives: 'Device API version 59, Module API version 10'
	run_demo_camera(mmc)	
```

However, using the beanshell interface in micro-manager to load demo camera works. 
```
core = mmc;  
core.unloadAllDevices();       
core.loadDevice("Camera", "DemoCamera", "DCam");
core.initializeDevice("Camera");
```

PR #289 (2022-10-31): Increment device interface version (70 -> 71) for MMTime change
Belatedly, for the MMTime struct layout change in #281 (specifically, commit 179b5fc86e133b243d4663b744f1f27789ad052c).

Closes #288.

If we merge this today (as I plan to), it will mean that the following nightly builds had incorrect device interface version (70) despite having actual binary interface 71:
20221030, 20221028, 20221027
In addition, 20221026 and 20221025 (since #277 commit 8ba45cde4436f8331f26a1adfac858ba08d51731) may have yet another incompatible interface (true at least [on macOS](https://github.com/micro-manager/ [... cut]