micro-manager/mmCoreAndDevices #295: Investigate and maybe remove "Autofocus support" in CoreCallback

henrypinkard, 2022-11-07:
It's probably bad practice for devices to be calling these methods, because it presumes a certain type of microscope, an assumption that will be removed in a future version. Its also preferable to not have devices horizontally controlling eachother without going through higher level code (except maybe in hub devices). Thus, it warrantes investigation as to how easily the following methods can be removed. This will require updating device interface version.

```c++
   // autofocus support
   const char* GetImage();
   int GetImageDimensions(int& width, int& height, int& depth);
   int GetFocusPosition(double& pos);
   int SetFocusPosition(double pos);
   int MoveFocus(double v);
   int SetXYPosition(double x, double y);
   int GetXYPosition(double& x, double& y);
   int MoveXYStage(double vX, double vY);
   int SetExposure(double expMs);
   int GetExposure(double& expMs);
   int SetConfig(const char* group, const char* name);
   int GetCurrentConfig(const char* group, int bufLen, char* name);
   int GetChannelConfig(char* channelConfigName, const unsigned int channelConfigIterator);
```

 - [ ] Investigate their usages (FakeCamera uses at least one)

marktsuchida, 2022-11-07:
It turns out to be FakeCamera only, and just 1 usage. I checked full builds on Windows and macOS.

henrypinkard, 2022-11-07:
Hooray! Should be easy then

marktsuchida, 2022-11-07:
Also we can remove the following already deprecated methods at the same time, to avoid an extra bump of device interface version:
```c++
MM_DEPRECATED(virtual void NextPostedError(int& /*errorCode*/, char* /*pMessage*/, int /*maxlen*/, int& /*messageLength*/)) = 0;
MM_DEPRECATED(virtual void PostError(const int, const char*)) = 0;
MM_DEPRECATED(virtual void ClearPostedErrors(void)) = 0;
```
None of them are used in device adapters in our codebase, and I don't see how to make this interface work (we need a completely new design if we want asynchronous error reporting).

PR #710 (2025-08-15): MMDevice: Delete most deprecated functions
And remove corresponding CoreCallback implementations (and leftover functions from previously deprecated ones).

Closes #295.

This leaves a few deprecated functions in MMDevice because they are still used by a few device adapters:

```cpp
MM_DEPRECATED(virtual int GetFocusPosition(double& pos)) = 0;
MM_DEPRECATED(virtual int GetXYPosition(double& x, double& y)) = 0;
MM_DEPRECATED(virtual MM::SignalIO* GetSignalIODevice(const MM::Device* caller, const char* deviceName)) = 0;
```