micro-manager/mmCoreAndDevices #299: 89 North LDI device adapter display bug

fadero, 2022-11-11:
The device adapter for the 89 North LDI gets "out of sync" with Micro-Manager. The cause of the bug is unknown, but MMCore will ask the LDI to verify the intensities of its illumination lines, typically starting with 405. The LDI responds with "SLEEP=0" likely indicating that it is not asleep. The problem with this first response is that MMcore parses the last digit, "0" as the intensity of the 405 line, then it displays the 405 line as an intensity value of 0 (even though the LDI is correctly illuminating). Next, MMcore asks for the 470 line's intensity, and the LDI responds with the intensity of the 405 line. Thus, all the lines' intensities are displayed in the GUI out-of-sync with how they actually are at the LDI.

marktsuchida, 2022-11-11:
If I'm guessing correctly that you are getting this description by observing the CoreLog, could you include a full log that includes these interactions and what came before it? The timestamps on each message will also be of interest. And it would just generally provide much better context for understanding the situation, given that I'm not familiar with the protocol.

Also, what user action triggered this series of events?

My first guess from what you describe is that is sounds like the `SLEEP=0` is either being sent by the device unsolicited, or is being missed despite being the response to a previous command (such as `SLEEP?`, which, based on looking a the code, queries the value for the `Sleep Timer (Minutes)` property) sent by the device adapter. I would need more context to determine which is the case and why.

Aside from that, I would also be interested in whether this happens under the default settings (sleep timer is 30 min). The Problem Report (Help menu) can be used to record this and all of the above information if this is easy to reproduce deliberately.

nicost, 2022-11-16:
Things seem to go wrong here:

`2022-11-15T14:05:16.944032 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: SLEEP?
2022-11-15T14:05:16.944032 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SLEEP?
2022-11-15T14:05:16.973952 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: SLEEP=0
2022-11-15T14:05:16.977984 tid4004 [IFO,Core] Did update system state cache
2022-11-15T14:05:17.017874 tid4004 [IFO,App] Updating GUI; config pad = true; from cache = true
2022-11-15T14:05:17.017874 tid4004 [IFO,Core] Default autofocus set to PFS
2022-11-15T14:05:17.018872 tid4004 [IFO,App] Rebuilding config group table
2022-11-15T14:05:17.018872 tid4004 [IFO,App] Finished rebuilding config group table
2022-11-15T14:05:17.019869 tid4004 [IFO,App] Finished updating GUI
2022-11-15T14:06:07.910967 tid4004 [IFO,dev:CameraLeft] Device adapter requests invalid values ( 0, 0) for property: CircularBufferFrameCount
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: SET?405
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SET?405
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: ok
2022-11-15T14:06:07.911965 tid4004 [ERR,Core:dev:89 North Laser Diode Illuminator] Cannot get value of property "405 Intensity" [ Error in device "89 North Laser Diode Illuminator": Failed to get current intensity (405nm) setting from LDI!
                                   [                                             ] ok (108901) ]
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: SHUTTER?405
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SHUTTER?405
2022-11-15T14:06:07.939889 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: SET:405=0
2022-11-15T14:06:07.939889 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: TTL_INVERT?405`

marktsuchida, 2022-11-18:
I think debug logging would show the SerialManager logging, which I would know better how to interpret than what I see here (and I would trust it more for completeness).

It does look like the `ok` response is causing issues, and the device adapter code does not look like it expects `ok` except as a response to `FAULT?`. Given that the timestamp of the `ok` is (probably) too close to be in response to `SET?405`, it is probably a result of something earlier: either the `SLEEP?` or something not visible here. (Note that the `RECEIVED` log entries are logged after the device adapter calls `GetSerialAnswer`, not when the actual response arrived at the PC.)

nicost, 2022-11-18:
I think I now get it.  The LDI has an interlock.  When the interlock loop is open, it will report a fault (at least when asked for it).  When the interlock the closes, it appears that the LDI sends "ok".  That means the device adapter should either refrain from communicating with the LDI when it is in a FAULT state until it receive the "ok", or is should be prepared to receive "ok" at any point in time.  I believe that Elliot Steels wrote the adapter (pinging @elliotsteele).  The slightly longer log that gives a bit more of a clue below:

```
2022-11-15T14:05:16.824352 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: SPECKLE=ON
2022-11-15T14:05:16.824352 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: FAULT?
2022-11-15T14:05:16.824352 tid4004 [IFO,dev:89 North Laser Diode Illuminator] FAULT?
2022-11-15T14:05:16.854272 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: FAULT:Interlock - Remote Open
2022-11-15T14:05:16.854272 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: F_MODE?
2022-11-15T14:05:16.854272 tid4004 [IFO,dev:89 North Laser Diode Illuminator] F_MODE?
2022-11-15T14:05:16.884192 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: F_MODE=IDLE
2022-11-15T14:05:16.884192 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: INT_MODE?
2022-11-15T14:05:16.884192 tid4004 [IFO,dev:89 North Laser Diode Illuminator] INT_MODE?
2022-11-15T14:05:16.914112 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: INT_MODE=PC
2022-11-15T14:05:16.914112 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: SH_MODE?
2022-11-15T14:05:16.914112 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SH_MODE?
2022-11-15T14:05:16.944032 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: SH_MODE=EXT
2022-11-15T14:05:16.944032 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: SLEEP?
2022-11-15T14:05:16.944032 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SLEEP?
2022-11-15T14:05:16.973952 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: SLEEP=0
2022-11-15T14:05:16.977984 tid4004 [IFO,Core] Did update system state cache
2022-11-15T14:05:17.017874 tid4004 [IFO,App] Updating GUI; config pad = true; from cache = true
2022-11-15T14:05:17.017874 tid4004 [IFO,Core] Default autofocus set to PFS
2022-11-15T14:05:17.018872 tid4004 [IFO,App] Rebuilding config group table
2022-11-15T14:05:17.018872 tid4004 [IFO,App] Finished rebuilding config group table
2022-11-15T14:05:17.019869 tid4004 [IFO,App] Finished updating GUI
2022-11-15T14:06:07.910967 tid4004 [IFO,dev:CameraLeft] Device adapter requests invalid values ( 0, 0) for property: CircularBufferFrameCount
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: SET?405
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SET?405
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: ok
2022-11-15T14:06:07.911965 tid4004 [ERR,Core:dev:89 North Laser Diode Illuminator] Cannot get value of property "405 Intensity" [ Error in device "89 North Laser Diode Illuminator": Failed to get current intensity (405nm) setting from LDI!
                                   [                                             ] ok (108901) ]
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: SHUTTER?405
2022-11-15T14:06:07.911965 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SHUTTER?405
2022-11-15T14:06:07.939889 tid4004 [IFO,dev:89 North Laser Diode Illuminator] RECIEVED: SET:405=0
2022-11-15T14:06:07.939889 tid4004 [IFO,dev:89 North Laser Diode Illuminator] SENDING: TTL_INVERT?405
```

nicost, 2022-11-29:
Fixed in #304

closed 2022-11-29 by nicost