micro-manager/mmCoreAndDevices #306: Lumencor SOLA light engine shutter timing

fadero, 2022-11-29:
While running the SOLA V-N Light Engine shutter through MM software (not through BNC triggering), the first image acquired after MM tells the SOLA shutter to open is unevenly illuminated (the "top" quarter of the image is dimmer than the "bottom" three-quarters of the image). The second-through-nth images have even illumination. This problem occurs regardless of camera exposure time, but it is exacerbated at lower exposure times. 

We have hacked a solution by taking one more slice than we need in a z-stack, leaving the shutter open during the z-stack, and ignoring the first image. But this is sub-optimal from a timing and light dosage perspective. Is there a way to change the timing on the sola shutter opening?

Attached image: a gif of a 3D stack where the first slice is unevenly illuminated and the 2nd-14th slices are evenly illuminated.
Attached file: metadata from the stack.

[image]

[Info for 100x_300nm_1_MMStack_A1-Site_4.ome.txt](https://github.com/micro-manager/mmCoreAndDevices/files/10115842/Info.for.100x_300nm_1_MMStack_A1-Site_4.ome.txt)

fadero, 2022-11-30:
@marktsuchida There's been a few emails exchanged between @nicost and Nenad Amodaj on this topic. I'll post them here:

On 11/29/2022 1:07 PM, Nico Stuurman <[nstuurman@altoslabs.com](mailto:nstuurman@altoslabs.com)> wrote:

Hi Nenad,

It would be awesome to add a delay for the Sola.  It probably is fine to hard code it.  Let's start at 5 ms, if we encounter issues, we can change the value;)

Thanks!

Nico

On 11/29/2022 10:03 AM, Nenad Amodaj wrote:
Hi Nico

I'll check that out today. Busy should always return false for all models except for Sola. That's an oversight on my part. I can fix that by adding a fixed delay of 5 ms to on/off; that would be the easiest and would mimic how the other light engines respond to commands. They block until the command is executed. Or I can do some threading stuff to keep Busy on for about 5 ms. Let me know. Also, do you think we should introduce a property to control the delay interval? Sola does not have handshaking for commands and works at 9600 baud. The command is 4 bytes. 5ms is going to work, but maybe you can do 3 ms. Not sure if that matters to you.

BTW I have a meeting with Ravi this week to get started. I'll send you a separate email about it.

Best
Nenad

On Tue, Nov 29, 2022 at 9:53 AM Nico Stuurman <[nstuurman@altoslabs.com](mailto:nstuurman@altoslabs.com)> wrote:
Hi Nenad,

Hope you are well, and looking forward to seeing you around here!

We are running into an issue with the Sola.  Especially with short 
exposure times, the light comes on only after camera exposure has 
started (using the MM autoshutter functionality).  Looking at the device 
adapter code, the Busy function always returns false, so I take this to 
mean that the Sola itself does not immediately respond after receiving 
the message that the LEDs should be turned on.  The difference is 
probably only a few milliseconds, but makes our images unusable.

Would it make sense to add a delay to the device adapter?  Any other way 
we can mitigate this problem?

Best,

Nico

fadero, 2022-11-30:
Hi @marktsuchida- @nico and Nenad are working on a solution. This was meant to be a landing page for them to reference the issue. Effectively, Nenad is going to add in a hard-coded 5ms delay after the SOLA turns on before the camera starts exposing.

marktsuchida, 2022-11-30:
@fadero Ah! Sounds good.

closed 2022-12-05 by fadero