micro-manager/mmCoreAndDevices #311: pymmcore fails to load Demo camera

kasliwalr, 2022-12-18:
Attention: @marktsuchida I am having problems try to run this [pymmcore example](https://micro-manager.org/Using_the_Micro-Manager_python_library) on Ubuntu. It is throwing an error - seems its unable to load the demo camera shared libraries.
```
2022-12-17T15:44:27.693674 tid140069159765824 [IFO,LogManager] Enabled logging to stderr
2022-12-17T15:44:27.693693 tid140069159765824 [IFO,LogManager] Switching primary log level from info to trace
MMCore version 10.1.1
Device API version 70, Module API version 10
Traceback (most recent call last):
  File "/home/<user>/repos/mm_integration/python/scripts/load_demo_camera.py", line 21, in <module>
    run_demo_camera(mmc)	
    ^^^^^^^^^^^^^^^^^^^^
  File "<path_to_micro-manager>/mm_integration/python/scripts/load_demo_camera.py", line 7, in run_demo_camera
    mmc.loadDevice('Camera', 'DemoCamera', 'DCam')
RuntimeError: Failed to load device "DCam" from adapter module "DemoCamera"
2022-12-17T15:44:27.704103 tid140069159765824 [IFO,LogManager] Switched primary log level from info to trace
2022-12-17T15:44:27.704215 tid140069159765824 [dbg,Core] Will load device DCam from DemoCamera
2022-12-17T15:44:27.712947 tid140069159765824 [dbg,Core] Will unload all devices
2022-12-17T15:44:27.712952 tid140069159765824 [IFO,Core] Did unload all devices
2022-12-17T15:44:27.712967 tid140069159765824 [IFO,Core] System reset
2022-12-17T15:44:27.713268 tid140069159765824 [IFO,Core] Core session ended
```

I did a full installation of micro-manager (working off the main branch), after that I have been trying the following
```
# Modify Makefile.am
# Modify configure.ac
> cd micro-manager
> ./autogen.sh
> ./configure --without-java
> make fetchdeps
> make -j4
> sudo make install
```
 I can see that demo camera shared libraries end up in `/usr/local/lib/micro-manager/`, but when I try to run the following script, I get the above error.
```
import pymmcore
#import pymmcore-plus

def run_demo_camera(mmc: pymmcore.CMMCore):
	mmc.setDeviceAdapterSearchPaths('/usr/local/lib/micro-manager')
	#mmc.loadSystemConfiguration('<path_to_micro-manager>/micro-manager/bindist/any-platform/MMConfig_demo.cfg')
	mmc.loadDevice('Camera', 'DemoCamera', 'DCam')
	mmc.initializeAllDevices()
	mmc.setCameraDevice('Camera')
	mmc.snapImage()
	img = mmc.getImage()
	print(type(img))

if __name__=="__main__":
	mmc = pymmcore.CMMCore()  # Instance micromanager core
	# mmc = pymmcore_plus.CMMCore.instance()  # Instance micromanager core
	mmc.enableStderrLog(True)
	mmc.enableDebugLog(True)
	print(mmc.getVersionInfo()) # gives: 'MMCore version 2.3.2'
	print(mmc.getAPIVersionInfo()) # gives: 'Device API version 59, Module API version 10'
	run_demo_camera(mmc)	
```

However, using the beanshell interface in micro-manager to load demo camera works. 
```
core = mmc;  
core.unloadAllDevices();       
core.loadDevice("Camera", "DemoCamera", "DCam");
core.initializeDevice("Camera");
```
Micro-manager version
```
MM Studio version: 2.0.x dev
MMCore version 10.4.0
Device API version 71, Module API version 10
User: <user>
```

marktsuchida, 2023-01-03:
> print(mmc.getAPIVersionInfo()) # gives: 'Device API version 59, Module API version 10'

If this is really correct, something is very wrong (pymmcore was never released for device interface version `59`). If it's (by any chance) a typo for `69`, then you need to upgrade pymmcore to the latest, which has device interface version 71, matching your Micro-Manager installation.

> mmc.setDeviceAdapterSearchPaths('/usr/local/lib/micro-manager')
> 	#mmc.loadSystemConfiguration('<path_to_micro-manager>/micro-manager/bindist/any-platform/MMConfig_demo.cfg')
> 	mmc.loadDevice('Camera', 'DemoCamera', 'DCam')
> 	mmc.initializeAllDevices()

The DemoCamera devices are intended to be used with the "hub" device `DHub`. Loading `DCam` by itself may not work (I don't remember for sure). It's probably safer to use a `.cfg` file created by the MM app (or `MMConfig_demo.cfg`).

kasliwalr, 2023-01-06:
@marktsuchida, thanks for taking a look into this....It was indeed a typo, but that did not solve the problem..

I installed the latest `pymmcore` from [here](https://github.com/micro-manager/pymmcore/releases/download/v10.3.0.71.0/pymmcore-10.3.0.71.0-cp37-cp37m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl), the device API version now matches the Micro-manager device API version. But am still getting the same error (even when using `DHub` with/without `MMConfig_demo.cfg`) 

Ran the following script
```
import pymmcore

def run_demo_camera(mmc: pymmcore.CMMCore):
        mmc.setDeviceAdapterSearchPaths('/usr/local/lib/micro-manager')
        mmc.loadSystemConfiguration('<path_to_micro-manager>/micro-manager/bindist/any-platform/MMConfig_demo.cfg')  # fails here.
        mmc.loadDevice('Camera', 'DemoCamera', 'DHub')
        mmc.initializeAllDevices()
        mmc.setCameraDevice('Camera')
        mmc.snapImage()
        img = mmc.getImage()
        print(type(img))

if __name__=="__main__":
        mmc = pymmcore.CMMCore()  # Instance micromanager core
        mmc.enableStderrLog(True)
        mmc.enableDebugLog(True)
        print(f'Pymmcore version: {mmc.getVersionInfo()}') # gives: 'MMCore version 2.3.2'
        print(f'Device API version: {mmc.getAPIVersionInfo()}') # gives: 'Device API version 59, Module API version 10'
        run_demo_camera(mmc)    
```
And the result was
```
2023-01-05T17:52:13.178566 tid139794291077504 [IFO,LogManager] Enabled logging to stderr
2023-01-05T17:52:13.178581 tid139794291077504 [IFO,LogManager] Switching primary log level from info to trace
Pymmcore version: MMCore version 10.3.0
Device API version: Device API version 71, Module API version 10
Traceback (most recent call last):
  File "load_demo_camera.py", line 19, in <module>
    run_demo_camera(mmc)	
  File "load_demo_camera.py", line 5, in run_demo_camera
    mmc.loadSystemConfiguration('<path_to_micro-manager>/micro-manager/bindist/any-platform/MMConfig_demo.cfg')
OSError: Line 7: Device,DHub,DemoCamera,DHub
Failed to load device "DHub" from adapter module "DemoCamera" [ Failed to load device adapter "DemoCamera" [ Failed to load module "libmmgr_dal_DemoCamera.so.0" [ libmmgr_dal_DemoCamera.so.0: cannot open shared object file: No such file or directory ] ] ]

2023-01-05T17:52:13.188912 tid139794291077504 [IFO,LogManager] Switched primary log level from info to trace
2023-01-05T17:52:13.189083 tid139794291077504 [dbg,Core] Will set Core property: Initialize = 0
2023-01-05T17:52:13.189088 tid139794291077504 [dbg,Core] Will unload all devices
2023-01-05T17:52:13.189090 tid139794291077504 [IFO,Core] Did unload all devices
2023-01-05T17:52:13.189104 tid139794291077504 [dbg,Core] Did set Core property: Initialize = 0
2023-01-05T17:52:13.189113 tid139794291077504 [dbg,Core] Will load device DHub from DemoCamera
2023-01-05T17:52:13.189478 tid139794291077504 [IFO,Core] Unloading all devices after failure to load system configuration
2023-01-05T17:52:13.189480 tid139794291077504 [dbg,Core] Will unload all devices
2023-01-05T17:52:13.189481 tid139794291077504 [IFO,Core] Did unload all devices
2023-01-05T17:52:13.189490 tid139794291077504 [IFO,Core] Now rethrowing original error from system configuration loading
2023-01-05T17:52:13.197112 tid139794291077504 [dbg,Core] Will unload all devices
2023-01-05T17:52:13.197114 tid139794291077504 [IFO,Core] Did unload all devices
2023-01-05T17:52:13.197127 tid139794291077504 [IFO,Core] System reset
2023-01-05T17:52:13.198231 tid139794291077504 [IFO,Core] Core session ended
```

I checked that the built `DemoCamera` library is at the expected location as well. 
```
> ls /usr/local/lib/micro-manager/
libmmgr_dal_DemoCamera.la  libmmgr_dal_DemoCamera.so.0
```
pymmcore version also seems to be correct
```
> conda list | grep pymmcore
pymmcore                  10.3.0.71.0              pypi_0    pypi
```

On a side note: Some of the documentation in the following links is not up to date
1. [This tutorial](https://micro-manager.org/Using_the_Micro-Manager_python_library) recommends the use of `DCam` .
2. Also tried installing pymmcore directly using `conda install -c conda-forge pymmcore` from conda env based on python > 3.7, it installs pymmcore with device API version 71. Btw, if you use python version 3.7 it install device API version 70.

marktsuchida, 2023-01-09:
> `mmc.setDeviceAdapterSearchPaths('/usr/local/lib/micro-manager')`

I missed this earlier. This needs to be `mmc.setDeviceAdapterSearchPaths(['/usr/local/lib/micro-manager'])`.
You can call `mmc.getDeviceAdapterSearchPaths()` to confirm that setting the path worked as intended.

This is a gotcha due to the Python string being interpreted as a sequence of characters, which are themselves strings in Python:
```
>>> mmc.setDeviceAdapterSearchPaths('/usr/local/lib/micro-manager')
>>> mmc.getDeviceAdapterSearchPaths()
('/', 'u', 's', 'r', '/', 'l', 'o', 'c', 'a', 'l', '/', 'l', 'i', 'b', '/', 'm', 'i', 'c', 'r', 'o', '-', 'm', 'a', 'n', 'a', 'g', 'e', 'r')
```
(Not sure if there is an easy fix to prevent this at the moment, but suggestions are always welcome.)

> * [This tutorial](https://micro-manager.org/Using_the_Micro-Manager_python_library) recommends the use of `DCam` .

Thanks, I removed that example (a correct example is probably too much for an intro page on pymmcore, and not what we want to direct new users to).

> * Also tried installing pymmcore directly using `conda install -c conda-forge pymmcore` from conda env based on python > 3.7, it installs pymmcore with device API version 71. Btw, if you use python version 3.7 it install device API version 70.

Not sure why, but conda-forge seems to only build 3.8+ currently; you probably got the last version that was built for 3.7. PyPI (pip) still has 3.7 binaries, though I plan to drop it later this year when 3.7 reaches end-of-life.

kasliwalr, 2023-01-11:
> I missed this earlier. This needs to be mmc.setDeviceAdapterSearchPaths(['/usr/local/lib/micro-manager']).
This was it! now working.

closed 2023-01-11 by kasliwalr