micro-manager/mmCoreAndDevices #335: Andor Integrated Laser Engine device adapter doesn't work

ieivanov, 2023-03-10:
The current version of the Andor Integrated Laser Engine device adapter doesn't work with the distributed version of `AB_ALC_REV64.dll` from the Andor ALC SDK.

The device adapter is written for version 2.5 of the ALC SDK ([link](https://github.com/micro-manager/mmCoreAndDevices/blob/8c2081a134fb088e1ce59ab2a50afcf161d6224f/DeviceAdapters/IntegratedLaserEngine/IntegratedLaserEngine.vcxproj#L87)), whereas the version distributed with Micro-Manager is older. Version 2.5 of the SDK is also not available through Andor - the most recent version available is 2.4: https://andor.oxinst.com/downloads/view/driver:-andor-alc-sdk-2.4.

Please update the version of `AB_ALC_REV64.dll` distributed with Micro-Manager, and provide access to v2.5 of the SDK through the Andor support website.

CC: @cmessaouik

marktsuchida, 2023-03-11:
The `AB_ALC_REV64.dll` currently distributed with MM is for use with the older `AndorLaserCombiner` device adapter, which I now see can be a source of confusion. Perhaps `IntegratedLaserEngine` could detect the old DLL (say, by missing functions) and provide a better error message.

We don't have a copy of the 2.5 DLL (we only have the headers), so Andor will need to provide that. (Whether we redistribute the new DLL is up to Andor -- we'd need their permission to do so.)

ieivanov, 2023-03-11:
Thanks for the clarification. As far as I can tell, Andor's recent policy has been to ask the users to download the needed driver packs and install them in the Micro-manager folder (e.g. https://micro-manager.org/Andor_SDK3). In that case, it would be great if Andor can provide access to v2.5 of this SDK, along with documentation on how to use this device adapter. I'll also contact their support team directly about this issue. 

It is true that the `AB_ALC_REV64.dll` distributed with MM present a point of confusion. It will be good to document this issue.