micro-manager/mmCoreAndDevices #337: Compilation error PIGCSCommandsDLL.cpp

fabiengagne, 2023-03-20:
I was pretty disappointed that I couldn't find a Linux installation package for micro-manager, or even better be present in the Ubuntu app store. 

I tried to compile from sources as directed by https://github.com/micro-manager/micro-manager/blob/main/doc/how-to-build.md under **Ubuntu Quickstart**

At the step `make -j4` I got the following error :

```
PIGCSCommandsDLL.cpp: In member function ‘int PIGCSCommandsDLL::LoadDLL(const string&, PIController*)’:
PIGCSCommandsDLL.cpp:133:21: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_GcsCommandset’ {aka ‘int (*)(int, const char*)’}
  133 |    GcsCommandset_ = static_cast<FP_GcsCommandset>(LoadDLLFunc ("GcsCommandset"));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PIGCSCommandsDLL.cpp:134:20: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_GcsGetAnswer’ {aka ‘int (*)(int, char*, int)’}
  134 |    GcsGetAnswer_ = static_cast<FP_GcsGetAnswer>(LoadDLLFunc ("GcsGetAnswer"));
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PIGCSCommandsDLL.cpp:135:24: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_GcsGetAnswerSize’ {aka ‘int (*)(int, int*)’}
  135 |    GcsGetAnswerSize_ = static_cast<FP_GcsGetAnswerSize>(LoadDLLFunc ("GcsGetAnswerSize"));
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PIGCSCommandsDLL.cpp:136:20: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_ConnectRS232’ {aka ‘int (*)(int, int)’}
  136 |    ConnectRS232_ = static_cast<FP_ConnectRS232>(LoadDLLFunc ("ConnectRS232"));
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PIGCSCommandsDLL.cpp:137:15: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_Connect’ {aka ‘int (*)(int)’}
  137 |    Connect_ = static_cast<FP_Connect>(LoadDLLFunc ("Connect"));
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PIGCSCommandsDLL.cpp:138:19: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_IsConnected’ {aka ‘int (*)(int)’}
  138 |    IsConnected_ = static_cast<FP_IsConnected>(LoadDLLFunc ("IsConnected"));
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PIGCSCommandsDLL.cpp:139:23: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_CloseConnection’ {aka ‘int (*)(int)’}
  139 |    CloseConnection_ = static_cast<FP_CloseConnection>(LoadDLLFunc ("CloseConnection"));
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PIGCSCommandsDLL.cpp:140:20: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_EnumerateUSB’ {aka ‘int (*)(char*, int, const char*)’}
  140 |    EnumerateUSB_ = static_cast<FP_EnumerateUSB>(LoadDLLFunc ("EnumerateUSB"));
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PIGCSCommandsDLL.cpp:141:18: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_ConnectUSB’ {aka ‘int (*)(const char*)’}
  141 |    ConnectUSB_ = static_cast<FP_ConnectUSB>(LoadDLLFunc ("ConnectUSB"));
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PIGCSCommandsDLL.cpp:142:20: error: invalid ‘static_cast’ from type ‘void*’ to type ‘PIGCSCommandsDLL::FP_ConnectTCPIP’ {aka ‘int (*)(const char*, int)’}
  142 |    ConnectTCPIP_ = static_cast<FP_ConnectTCPIP>(LoadDLLFunc ("ConnectTCPIP"));
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/fabien/micro-manager/mmCoreAndDevices/DeviceAdapters/../MMDevice/MMDevice.h:49,
                 from /home/fabien/micro-manager/mmCoreAndDevices/DeviceAdapters/../MMDevice/DeviceBase.h:30,
                 from PIController.h:29,
                 from PIController.cpp:25:
/home/fabien/micro-manager/mmCoreAndDevices/DeviceAdapters/../MMDevice/ImageMetadata.h:325:58: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
  325 |    MetadataSingleTag GetSingleTag(const char* key) const throw (MetadataKeyError)
      |                                                          ^~~~~
/home/fabien/micro-manager/mmCoreAndDevices/DeviceAdapters/../MMDevice/ImageMetadata.h:332:56: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
  332 |    MetadataArrayTag GetArrayTag(const char* key) const throw (MetadataKeyError)
      |                                                        ^~~~~
mv -f .deps/PI_GCS_2.Tpo .deps/PI_GCS_2.Plo
make[2]: *** [Makefile:565 : PIGCSCommandsDLL.lo] Erreur 1
make[2]: *** Attente des tâches non terminées....
mv -f .deps/PIController.Tpo .deps/PIController.Plo
mv -f .deps/PIGCSCommands.Tpo .deps/PIGCSCommands.Plo
make[2] : on quitte le répertoire « /home/fabien/micro-manager/mmCoreAndDevices/DeviceAdapters/PI_GCS_2 »
make[1]: *** [Makefile:664 : all-recursive] Erreur 1
make[1] : on quitte le répertoire « /home/fabien/micro-manager/mmCoreAndDevices/DeviceAdapters »
make: *** [Makefile:516 : all-recursive] Erreur 1
(micro-manager) fabien@fabien-XPS-15-7590:~/micro-manager$ 
```

I compiled from my Ubuntu 22.04 LTS.

closed 2023-03-20 by marktsuchida

fabiengagne, 2023-03-21:
I confirm that compilation is now successful with #338

marktsuchida, 2023-03-22:
@fabiengagne Thanks for reporting and confirming!

PR #338 (2023-03-20): PI_GCS_2: Fix compilation on Linux
Closes #337.