micro-manager/mmCoreAndDevices #344: Use TUCam under Linux

JiangXL, 2023-04-26:
The TUCam has linux sdk and share the same api with windows version. But current code only compile for windows.
How can I adapt micro-manager code to apply it?

Thanks!

https://www.tucsen.com/download-software/page/2/

marktsuchida, 2023-05-01:
Usually the process looks something like this:
- Install the Linux SDK.
- Try compiling the device adapter manually (`g++` with appropriate flags for library and include directories), to make sure it compiles. Fix any issues and add alternatives for any uses of Windows-only API.
- Add logic to `DeviceAdapters/configure.ac` to detect the camera SDK and set variables for library and include directories. Also, build of the device adapter should only be enabled when the SDK is available.
- Add a `Makefile.am` for the device adapter.

JiangXL, 2023-05-01:
Thanks, I will try.

JiangXL, 2023-05-14:
@fandayu Hi, could we have your help?