micro-manager/mmCoreAndDevices #354: Possible problem with Hamamatsu camera

ilyasdc, 2023-06-21:
We have a Hamamatsu C11440-36U which sometimes irrevocably crashes and requires a hard restart (turned on and off) irrespectively of the restart of MM or even the PC with this kind of message displayed:  

[image]

A script as simple as that can reproduce the issue by repeateadly taking images: 
```java
import java.util.Date;
import java.text.SimpleDateFormat;

public class NowString {
    public static String getTime() { 
          SimpleDateFormat df = new SimpleDateFormat("yyyy/MM/dd HH:mm:ss.SSS");//Set the date format
          String date = df.format(new Date());//new Date() is to get the current system time
          return date;
    }
}

mm.getScriptController().clearMessageWindow();
mm.scripter().message("Starting...");

mmc.setAutoShutter(false); // disable auto shutter 
mmc.setShutterOpen(true);  // keeps the shutter open
mm.live().setLiveMode(false);

long i = 0;
for (;;i++) {
		image = mm.live().snap(true).get(0);
		print(i+"\t"+NowString.getTime());
}

```

It can trigger the bug more or less randomly (might take hours, might take days or even weeks).

After discussion with Hamamatsu, it appears that something /might/ be off on micro-manager side (not sure but that's what /they/ claim). I provide the decoded camera log captured by hamamatsu software (don't remember the name on top of my head but thyey directly provided it during the debug process) 
: [230617_camera_logs_crash.pdf](https://github.com/micro-manager/mmCoreAndDevices/files/11817354/230617_camera_logs_crash.pdf)

According to the engineer, when capturing an image with `snapImage()` a call to `dcamcap_start` is made on the DCAM-API side but is not followed by `dcamcap_stop`which /could/ be the source of the problem. I'm not sure that's totally true since it can be that the buffer only has 1 image, or maybe not ? Anyway, since there's no public access to the camera device adapter I cannot look further.

[image]

The real question here, maybe: is this a normal usecase to use this snap() function repeatedly ? Any alternative ? it doesn't seem to do anything else than to call `MMCore.snapImage()` in our case (single channel image)

Thanks  !

marktsuchida, 2023-06-26:
> The real question here, maybe: is this a normal usecase to use this snap() function repeatedly ? Any alternative ? it doesn't seem to do anything else than to call `MMCore.snapImage()` in our case (single channel image)

Yes, that usage should not pose a problem, but it is not the most typical either (more often, there will be a fixed time interval between repeated snaps). If you think the crash is caused because of it, you could try inserting a brief sleep (say, 10 ms) between snaps, or use a sequence acquisition (or live mode, if displaying the images is all that is needed) instead.

ilyasdc, 2023-07-06:
Hello @nicost, is it possible to reopen the thread ? I know that you have more important bugs to fix but I want to be sure it's not a hardware problem. 
As I stated previously, adding a delay doesn't fix the problem. However I have to say that since I don't have access to the camera adapter nor have the necessary knowledge to dig into it, I'm not sure I can fix it without your help. If you can have a look at this camera adapter code, can you tell me if, when using MM `snapImage()` method, the `dcam_start` uses the argument `DCAM_START_SNAP`,  would be a huge help !

palmada, 2023-07-06:
Have you considered using the camera sequence function from the core object instead? That starts the camera acquiring frames in a continuous sequence. You then have to extract them from the buffer and add them to an acquisition so a tiny bit more code but should do what you need without any issues.

ilyasdc, 2023-07-06:
> 

Hello @palmada and thank you for your follow-up on this issue. Is setting a 1 frame `sequenceAcquisition` (https://valelab4.ucsf.edu/~MM/doc-2.0.0-gamma/mmcorej/mmcorej/CMMCore.html#startSequenceAcquisition-int-double-boolean-)  different from using the `snapImage` from the device adapter point of view ? 

For context I am computing sharpness of a z-stack, like the `AutoFocus` plugin, but with an imposed delay (100ms) since we have also problem with the stage (see https://github.com/micro-manager/micro-manager/commit/f9c6e7a69bd3d63f8c2764f38a83e718b64596ae#commitcomment-95719958 ).

palmada, 2023-07-06:
Yes it is indeed different, but not immensely so. If you want to use a delay, then it's not appropriate, as it starts a stream with no delay. 

One trick that might work for you instead is calling the `StopSequenceAcquisition` core method ([here](stopSequenceAcquisition)), which may call stop after a snap. A wild  guess though as the device adapter code isn't open source, last I checked.

ilyasdc, 2023-07-06:
> Yes it is indeed different, but not immensely so. If you want to use a delay, then it's not appropriate, as it starts a stream with no delay.
>
> One trick that might work for you instead is calling [...]
Ok I will try to check if the call to that method is shown in the logs of this usb monitoring tool that I was provided with.
> A wild guess though as the device adapter code isn't open source, last I checked.
It's not, indeed.

nicost, 2023-07-06:
Your best bet is to contact Hamamatsu support.  Hamamatsu maintains the code to the device adapter.  The bug is surely either in the device adapter, the Hamamatsu driver/SDK, or the camera firmware, and there is nothing we can (or should) do about any of those.  If you can use a sequence acquisition (as suggested by @palmada), that is surely a good workaround the problem.