micro-manager/mmCoreAndDevices #405: How to use MMCore

zc123123123, 2023-11-14:
Hello, I am not deep in programming, I want to use MMcoreAndDevices to control the camera acquisition and post-processing, but I seem to have a problem with the compiled file, I would like to ask if there is a clearer description of the use of the c++ libraries.
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1120	189 unresolved externals	tutorialjavatoc	D:\projects\micro-manager\mmCoreAndDevices\Release\tutorialjavatoc.exe	1	
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2001	unresolved external symbol "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl MMCorePrivate::GetPathOfThisModule(void)" (?GetPathOfThisModule@MMCorePrivate@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)	tutorialjavatoc	D:\projects\micro-manager\mmCoreAndDevices\tutorialjavatoc\PluginManager.obj	1

zc123123123, 2023-11-15:
The description of my question above may not be very clear, what I want to ask is, if I want to call MMcoreAndDevices directly in c++ to write a GUI interface to control my microscope, what should be the whole process of configuring it, although the description of the MMcore API is clear enough, but for a beginner in programming like me, not having a C++ usage example is a bit overwhelming for me.

zc123123123, 2023-11-15:
The python library call on micro-manager seems to be very simple and the tutorial is very clear, can I find somewhere a c++ call that is as clear as pycromanager, similar to how I should configure my library files and which ones to configure

marktsuchida, 2023-11-17:
The only well-supported way to use MMCore currently is from Python or Java. To use from C++, you will unfortunately need to do some work to build it in a way that works for you (there are plans to improve this).

On Windows the MMCore.vcxproj project builds a static library MMCore.lib (easiest to build if you open micromanager.sln and right-click on MMCore to build). You will want to link to this from your C++ project by setting the following project properties:
- Add the location of MMCore.lib to AdditionalLibraryDirectories (under Link)
- Add "MMCore.lib" to AdditionalInputs (under Link > Input)
- Add the mmCoreAndDevices/MMCore directory to AdditionalIncludeDirectories (under C/C++)

For the device adapters, I'd recommend getting them from a nightly build of Micro-Manager.

zc123123123, 2023-11-19:
> The only well-supported way to use MMCore currently is from Python or Java. To use from C++, you will unfortunately need to do some work to build it in a way that works for you (there are plans to [...]

Thanks a lot, I've understood how to link static libraries into my project, but what does it mean for device management to get them from a nightly build of Micro-Manager.

zc123123123, 2023-11-19:
Finally, I wrote a C++ program following the java tutorial, I first ran the virtual camera demo, and then added the header files of my own camera PVCAM to the include to generate a .dll file, and then successfully ran my own camera, but I don't know what you mean by a nightly build of Micro-Manager. What is this nightly build of Micro-Manager you are referring to, do you mean importing the device directly on the GUI like pycromanager?

marktsuchida, 2023-11-20:
https://micro-manager.org/Micro-Manager_Nightly_Builds, i.e., the binary distribution of Micro-Manager. You don't need to use the Java app, but you can get the device adapters (`mmgr_dal_*.dll`) from there -- if you find it hard to build any of them yourself.