micro-manager/mmCoreAndDevices #422: Stop requiring the CRTP type parameter to device base class templates

marktsuchida, 2023-12-19:
Device adapters currently derive from a base class in DeviceBase.h, such as `CGenericBase<U>`. All of these base class templates currently have a template parameter `U`, which is supposed to be set to the leaf (most-derived) class ([CRTP](https://en.cppreference.com/w/cpp/language/crtp)):

```c++
class MyGeneric : public CGenericBase<MyGeneric> { ... };
```

Because of this, device adapters that want to share part of the implementation between device classes sometimes need to resort to using template template parameters (examples in ASITiger, SequenceTester, UserDefinedSerial) or other methods that add complexity. (Preferring composition over inheritance is a good idea, but not always easy due to the need to call base class member functions.)

@IvoVellekoop made the observation in #414 that the type parameter `U` is actually not fundamental to the operation of the base classes. This is true in principle, and it would be good if we could drop it.

Unfortunately things are not so simple, because _the_ main way that `U` is used is in the definition of the types `CPropertyAction` and `CPropertyActionEx` (both members of `CDeviceBase<T, U>`), which are used in the vast majority of existing device adapters.

On the other hand, the new `ActionLambda` (#411) facility means that `CPropertyAction[Ex]` is no longer essential when writing new device adapters (`BH_DCC_DCU` in fact does not use `CPropertyAction[Ex]` even though it is an ordinary device adapter).

So we should be able to add a default argument to the parameter `U` (setting it to an internal tag type), so that one can write

```c++
class MyGeneric : public CGenericBase<> { ... };
```

provided that `MyGeneric` does not use `CPropertyAction[Ex]`. We can enforce the latter (with a custom error message) using `static_assert` inside `MM::Action[Ex]<U>`.

The only use of `U` other than `CPropertyAction[Ex]` is in the method `CDeviceBase::CreatePropertyWithHandler()`, which is only used in several device adapters and which should simply be removed (the method merely saves typing and is not worth keeping).

marktsuchida, 2023-12-19:
There is also the possibility of completely eliminating `CPropertyAction[Ex]` by mechanically translating all its uses to `MM::Action`. Then the device bases won't need to be templates at all.

```c++
// Somewhere in class MyDevice
// before
new CPropertyAction(this, &MyDevice::OnMyProperty)
// after (C++14)
new MM::Action<MyDevice>(this, &MyDevice::OnMyProperty)
// after (C++17 or later)
new MM::Action(this, &MyDevice::OnMyProperty)
```

henrypinkard, 2024-01-16:
I've always found the current way things are set up non-intuitive and confusing, and I'd definitely be in favor of simplifying it in one of the ways you suggest

marktsuchida, 2024-01-16:
> There is also the possibility of completely eliminating `CPropertyAction[Ex]` by mechanically translating all its uses to `MM::Action`. Then the device bases won't need to be templates at all.

Taking another look at this, this second approach is actually not that great, because `MyDevice` needs to be the most derived class and `CPropertyAction` may be called in a base class (so we can't do a mechanical replacement on existing code). But we can still make the template argument optional.