micro-manager/mmCoreAndDevices #426: How are com-port functions in SerialInstance.cpp implemented?

zhouchunyuan, 2024-01-04:
If we trace a serial function in MM, we end up in the SerialInstance.cpp
[https://github.com/micro-manager/mmCoreAndDevices/blob/main/MMCore/Devices/SerialInstance.cpp](url)

But I can not find and detailed information of how they are realized.

Take the Purge() as an example: 
`int SerialInstance::Purge() { RequireInitialized(__func__); return GetImpl()->Purge(); }`

If we use right-click menu in VS2022 to "Go To definition", it links back only to itself.

Can anyone please tell me how to find the reallization of those functions there: 
GetImpl()->SetCommand(command, term);  
GetImpl()->GetAnswer(txt, maxChars, term);
GetImpl()->Write(buf, bufLen);
GetImpl()->Read(buf, bufLen, charsRead);
GetImpl()->Purge();

Thanks and best regards
Chunyuan

nicost, 2024-01-04:
Implementations are in device adapters that implement the CSerialBase interface.  There are several (i.e, for serial, USB, TCP/IP, and HIC interfaces).  You probably are most interested in SerialManager: https://github.com/micro-manager/mmCoreAndDevices/tree/main/DeviceAdapters/SerialManager

closed 2024-01-04 by nicost

zhouchunyuan, 2024-01-05:
Thanks Nico, this not only solves my problem, but also helps better understand the logic inside core/deviceAdapter.