micro-manager/mmCoreAndDevices #442: `core.getExposure()` incorrect with HamamatsuHam camera (but `getProperty(..., "Exposure")` correct)

tlambert03, 2024-02-06:
I'm not exactly sure at what level this issue is being caused, but I think there's an issue with `core.getExposure` on HamamatsuHam devices.  Here's what I observe using pymmcore, and I see the same result with beanshell in MMStudio:

micro-manager nightly: 2.0.3_20240110
Windows 10
DCAM-API for USB: 23.12.6736 with Flash 4.0 V2+ (but I see a similar thing with ORCA R2 and 1394 IIDC driver) 
Device API version 71
Module API version 10

```python
In [1]: from pymmcore import CMMCore

In [2]: core = CMMCore()

In [4]: core.setDeviceAdapterSearchPaths([...])

In [5]: core.loadDevice("C", "HamamatsuHam", "HamamatsuHam_DCAM")

In [6]: core.initializeDevice("C")

In [7]: core.setCameraDevice("C")

In [8]: core.getExposure()
Out[8]: 33.32570940256119                  # apparent starting exposure 

In [9]: core.getProperty("C", "Exposure")
Out[9]: '33.3250'                          # returns the same thing

In [10]: core.snapImage(); core.getImage().mean()
Out[10]: 64063.15238046646                 # lens cap is off, here's the mean intensity at 33ms 

In [11]: core.setExposure(8)

In [12]: core.snapImage(); core.getImage().mean()
Out[12]: 18958.371970176697                # mean intensity suggests exposure WAS changed

In [13]: core.getExposure()
Out[13]: 1000.0                            # <------------- INCORRECT value reported

In [14]: core.getProperty("C", "Exposure")
Out[14]: '8.0000'                          # CORRECT value reported

In [15]: core.setProperty("C", "Exposure", "16")      # set it a different way

In [16]: core.snapImage(); core.getImage().mean()     # mean intensity suggests exposure WAS changed
Out[16]: 37373.14380955696

In [17]: core.getExposure()
Out[17]: 1000.0                            # <------------- incorrect value reported

In [18]: core.getProperty("C", "Exposure")
Out[18]: '16.0000'                         # correct value reported
```

tlambert03, 2024-02-06:
@marktsuchida mentioned that the Hamamatsu Device adapter received an update in December of 2023, and I can confirm that using a nightly build from late november (20231128) fixes this issue (`getExposure()` reports the correct value).

marktsuchida, 2024-02-10:
Hamamatsu is kindly helping investigating this.

tlambert03, 2024-02-10:
thank you Hamamatsu!

harth1026, 2024-11-07:
I know this is an old claim, but I am not sure if the original poster is aware of the work that has been done here.  We (Hamamatsu) have tested the claim as stated and have been able to reproduce the problem.  A newer version of the Hamamatsu adapter has been uploaded around 6/19/24.  This new version fixes an issue setting function values.  After testing this claim against the new adapter, the new adapter appears to have corrected this issue.  If you have experienced any problems similar to the original claim, please check against the newer adapter and see if the problem persists.  And let me know if you see any additional issues.

tlambert03, 2024-11-09:
thanks @harth1026.  Indeed I wasn't aware that the adapter was updated again in June. (I forget, is there a good way for us to know when the secret third party adapters are updated?).  We can close this then.  I haven't tried on a real camera but I'll take your word for it 👍

closed 2024-11-09 by tlambert03