micro-manager/mmCoreAndDevices #456: Poor Acquisition Performance on Different Camera

Radem130, 2024-04-02:
I apologize if I'm missing something obvious here, but I'm a bit lost with things to try. I would greatly appreciate any input.

Briefly, I am using a microscope setup with dyed samples, a laser, and a shutter. During acquisitions, my interval between frames and camera exposure are different. So, to provide a precise laser exposure to the sample, I have been using the pycromanager hooks to open and close the shutter only during the exposure of the camera. Recently, I have swapped a Photometrics primebsi camera to a Hamamatsu C13440 camera, and it seems that the exposure/processing of each image is significantly slower.

This is demonstrated in the time difference between post_camera_hook_fn and post_hardware_hook_fn calls. I have the exposure set to 500ms, with 1s intervals, and here are the time differences:
[image]

The post_camera_hook_fn call seems to have an extra 370ms beyond the exposure of the camera. Am I misunderstanding the use of the hooks here, or is this an issue with the data transfer from the camera?

A few other potentially useful notes:
- pycromanager version: 0.29.9
- micro-manager version: 2.0.3_20240105
- On the previous camera this timing was quite precise, within a couple millliseconds
- The Hamamatsu camera appears to acquire normally when not using hooks. I'm not sure how to systematically verify this in pycromanager, but I know it works fine in micro-manager.

Here is the code I used to take the acquisition (note the small 4x4 ROI):

```
import matplotlib.pyplot as plt
import pycromanager
import datetime
from constants import mm_path, cfg_ham_camera # containing absolute paths
import numpy as np

class AcquisitionHandler:
    def __init__(self, num_frames, exposure, time_interval):
        save_path = "video_tests"
        pycromanager.start_headless(mm_path, cfg_ham_camera, port=4827)
        core = pycromanager.Core(port=4827)
        
        ROI = [4,4]
        core.set_roi(int(2048/2-ROI[0]/2),int(2048/2-ROI[1]/2),ROI[0],ROI[1])
        core.set_exposure(exposure)
        
        self.hook_times = {
            "pre_hard":[],
            "post_hard":[],
            "post_cam":[]
        }
        
        with pycromanager.Acquisition(directory=save_path,
                            name=f"hamtest_new", 
                            show_display=False, 
                            pre_hardware_hook_fn=lambda event: self.hook(event, "pre_hard"),
                            post_hardware_hook_fn=lambda event: self.hook(event, "post_hard"),
                            post_camera_hook_fn=lambda event:self.hook(event, "post_cam")
                            ) as acq:
            events = pycromanager.multi_d_acquisition_events(num_time_points=num_frames,time_interval_s=time_interval)
            acq.acquire(events)

    def hook(self, event, hook_type):
        self.hook_times[hook_type].append(datetime.datetime.now())
        return event

acquisition_handler = AcquisitionHandler(num_frames = 50, exposure = 500, time_interval = 1)

# calculate difference in each event: post_cam - post_hard
times = acquisition_handler.hook_times
diff = [(times['post_cam'][i] - times['post_hard'][i]).total_seconds() for i in range(len(times['post_cam']))]
print(f'avg:{np.average(diff)}, stdev: {np.std(diff)}')

plt.plot(diff,'o')
plt.xlabel("Frame")
plt.ylabel("Diff (s)")
plt.show()
```

marktsuchida, 2024-04-08:
I agree with @henrypinkard. Depending on the camera, its driver, and how its Micro-Manager device adapter is implemented, there will be variability in the time it takes to set up an acquisition for a single frame. This is not something that can be fixed in general, and even in particular cases can be quite tricky to do anything about (if possible at all).

If you need accurate timing, the best way would be to use some form of hardware (TTL) synchronization. For example, you could use the camera's exposure output signal to drive the shutter. If it is a problem if your shutter starts opening after the exposure starts, then you will need to use some other way (perhaps put the camera in external trigger mode and use something like an Arduino to generate a sequence of triggers to both the shutter and the camera).

Radem130, 2024-04-11:
@marktsuchida @henrypinkard Thank you for the responses, I just wanted to ensure I wasn't missing anything with my understanding in how pycromanager is implemented. For now, I have hacked together a solution with the current limitations (this extra delay is actually okay for my purposes), and if I need to implement faster timing I will look into TTL synchronization.

closed 2024-04-11 by Radem130