micro-manager/mmCoreAndDevices #458: CSUW1 device adapter Dichroic misses a position

CellKai, 2024-04-09:
Hi, 

I am trying to setup MM CSUW1 dual camera system but struggle to find a necessary device.

the "Dichroic" device adapter should have two positions, one for the main dichroics that separates excitation from emission light, and another position for the secondary dichroic which guides the light to camera 1, or camera 2, or both at the same time, see the CSU-W1 scanhead scheme from the Yokogawa page below:

[image]

I looked at [the code](https://github.com/micro-manager/mmCoreAndDevices/blob/cd24fa7286b4d8792499f0fb208cc8fd00cdfa1c/DeviceAdapters/CSUW1/CSUW1.h#L110) (hopefully at the right place) and thought that maybe the function `GetNumberOfPositions` does indeed return 2, but this selection is afterwards not exposed in the "action interface", and hence when setting up MM it can not be selected. Maybe this choice (also) needs to be added in t[he .cpp file](https://github.com/micro-manager/mmCoreAndDevices/blob/cd24fa7286b4d8792499f0fb208cc8fd00cdfa1c/DeviceAdapters/CSUW1/CSUW1.cpp#L441).

For comparison, the class "FilterWheel" works fine, when adding this device in MM the dialogue allows me to select filter wheel 1 or filter wheel 2. For the dichroic, this option does not exist see screenshots from MM.

[image]

ping @nicost @marktsuchida  @nanthony21 

crossposte on the micro-forum: https://forum.microlist.org/t/micro-manager-csu-w1-dual-camera-setup-cant-address-secondary-dichroic/2489?u=cellkai

nicost, 2024-04-09:
The Image Spplitting dichroic in the W1 is either the "CSUW1-Port", or the "CSUW1-FRAP" device.  Let us know which one.

`GetNumberOfPositions` is used to establish the number of positions of a State Device (such as a dichroic switcher or a filter wheel), not the number of State Devices of this kind.

closed 2024-04-09 by nicost

CellKai, 2024-04-10:
thank you @nicost !

its the "CSUW1-Port".