micro-manager/mmCoreAndDevices #465: Support Aravis more widely

marktsuchida, 2024-06-14:
Recently added in #445, right now it requires Aravis in a hard-coded location and only works on Linux.

[Aravis](https://aravisproject.github.io/aravis/aravis-stable/) itself supports also Windows and macOS, so it would be nice to support those.

Also, it would be nice for the Linux `./configure` to find a system-installed Aravis. (Ubuntu 24.04 has packages for Aravis 0.8.30; it has not been confirmed if that is a new enough version.)

Because Aravis is LGPL licensed, shipping binaries is an option on Windows and macOS.

I intend to work on this at some point but anybody else is also welcome to look into this.

HazenBabcock, 2026-10-06:
For Windows, I have a build that works in CI:

- **Adapter:** the [`aravis-windows`](https://github.com/HazenBabcock/mmCoreAndDevices/tree/aravis-windows) branch of my fork adds `DeviceAdapters/Aravis/AravisCamera.vcxproj`, an entry in `micromanager.sln`, and a `build.xml` that installs the DLLs. The project is named `AravisCamera` so that the DLL is `mmgr_dal_AravisCamera.dll`, the same name as on Linux. The adapter source needed no changes.
- **Aravis binaries:** the project expects them in `3rdpartypublic/aravis/aravis-0.8.36-bin`: `include/aravis-0.8`, `include/glib-2.0`, and `x64/` with 12 DLLs (Aravis; GLib, GObject, GIO and GModule; libffi, libintl, libiconv, PCRE2, libxml2, libusb and zlib) plus the four import libraries the adapter links. I built them with vcpkg (a pinned release, MSVC v143, release-only DLLs) in [HazenBabcock/mm-aravis-windows-ci](https://github.com/HazenBabcock/mm-aravis-windows-ci). The directory is in [this release](https://github.com/HazenBabcock/mm-aravis-windows-ci/releases/tag/aravis-0.8.36-2) as a zip that unpacks into `3rdpartypublic`; it includes each package's license and a `BUILD-INFO.txt` recording how it was built. Everything else the DLLs load comes from Windows or the Visual C++ 2015-2022 runtime that the installer already installs.
- **Testing:** the CI builds the `AravisCamera` project from `micromanager.sln` with MSBuild against that directory, loads the adapter with pymmcore 12.5.0.75.0, and snaps an image from Aravis's fake GigE camera on the loopback interface ([latest run](https://github.com/HazenBabcock/mm-aravis-windows-ci/actions/runs/37412227487)). I haven't yet tried it in an installed nightly, or with real cameras on Windows.

Some questions for you:

1. Is this approach and layout OK for 3rdpartypublic? If so, could you commit the contents of the zip there? Once `3rdpartypublic-revision` points at it I'll open the PR from `aravis-windows`; merging it before then would break the Windows build.
2. Eight of the twelve DLLs are LGPL (Aravis, the four GLib DLLs, libintl, libiconv and libusb). Their license files are in the zip under `licenses/`, but as far as I can tell the installer doesn't ship third-party license files. Do you want these installed alongside the DLLs?
3. The DLLs install into the Micro-Manager directory under their own names (`z.dll`, `libusb-1.0.dll`, `libxml2.dll`, `iconv-2.dll`, `intl-8.dll` and so on). None of them clash with what the public adapters install, but I can't check SecretDeviceAdapters.

One thing users will run into: on Windows, Aravis reaches USB3 Vision cameras through libusb, which needs the WinUSB or libusbK driver bound to the camera (for example with Zadig) in place of the vendor's driver. So on Windows the adapter will mostly be useful for GigE cameras unless the user switches drivers. GigE cameras may also need a firewall exception for the image stream.

marktsuchida, 2026-10-06:
Of course I didn't think about how complex the dependency chain would be when I wrote this issue :)

I'm going to have to think about this a little. Typically we don't like to ship non-adapter DLLs when we can avoid it, preferring static linking for open source third party libraries. (Among other things, we don't want to force all adapters to depend on the same version of a given library. Also, static linking saves on space if only a portion of a library is used, and only by 1 adapter.) But it's not always avoidable and I'm not sure what the feasibility is in this case.

Do these DLLs get built by MSVC or GCC? I guess MSVC given you use vcpkg?

HazenBabcock, 2026-10-06:
Yes they are built by MSVC from a pinned vcpkg release.

I asked Claude about why we had to compile all of these DLLs ourself and if it couldn't find anything online and it said that "No one publishes a ready-made Windows build of Aravis that fits Micro-Manager" or at least not for MSVC. However, "MSYS2 is the one real prebuilt option: mingw-w64-aravis 0.8.33, built with the MinGW toolchain (GCC or Clang), not Microsoft's compiler", with the following list of why it didn't think this was a good idea:
1. It ships MinGW import libraries, so we'd have to generate MSVC .lib files ourselves.
2. GLib's configuration header glibconfig.h is generated for the compiler that built GLib. Using a MinGW-generated one under MSVC is a known source of trouble, which libdc1394's simple headers don't have.
3. It brings extra MinGW runtime DLLs (libgcc_s_seh/libwinpthread, or libc++).
4. It's 0.8.33, behind the 0.8.36 you test with on Linux. MSYS2 is also a rolling distribution, so pinning a version means fetching archived package files.

HazenBabcock, 2026-10-06:
Here is a proposal for static linking:

- Build Aravis and its dependencies (GLib, libffi, libintl, libiconv, PCRE2, libxml2, libusb and zlib) with vcpkg as static libraries against the dynamic C runtime (`/MD`), and link them into `mmgr_dal_AravisCamera.dll`. The adapter would then be a single DLL that depends only on Windows and the Visual C++ runtime, with nothing else added to the installation directory and no `build.xml`.
- 3rdpartypublic would hold the headers and the static `.lib` files, plus the licenses and a record of how they were built.

GLib supports static builds on Windows, and the adapter's source wouldn't change, only its project file. The LGPL libraries would be statically linked into a BSD-licensed adapter whose source is public, which as far as I can tell is the same situation as the macOS build, where libusb and libdc1394 are already linked statically.

One question: would you want debug builds of the static libraries as well (like OpenCV's `d`-suffixed libraries), or are release-only libraries enough? A Debug build of the adapter linked against release C libraries may give linker warnings.

If this sounds reasonable, I'll try it in CI and report back the size of the adapter DLL and what it depends on.

marktsuchida, 2026-10-06:
Yes, that sounds great, thanks!

Debug libraries are not a requirement. Not critical, but perhaps in your BUILD-INFO.txt you could also include the Git hash of your repo (i.e., not just the Actions run link). Vcpkg hash and triplet should be a pretty good way to reproduce (or build debug libs if needed in the future).

As for the directory layout in 3rdpartypublic, let's use capitalized `Aravis/` at the toplevel (just to match the majority of other "vendors" and the Aravis project's own prose name). Otherwise I'm fine with what you have. The important thing (for legacy build and workflow reasons) is that we can add a new version while keeping the old version in the same Subversion revisions, and the `aravis-0.8.36-bin` directory accomplishes that.

HazenBabcock, 2026-10-06:
Claude got static linking to work:

- **Aravis files:** [release aravis-0.8.36-3](https://github.com/HazenBabcock/mm-aravis-windows-ci/releases/tag/aravis-0.8.36-3) has a zip that unpacks into 3rdpartypublic as `Aravis/aravis-0.8.36-bin`, with `include/`, the static libraries in `x64/`, the licenses, and a `BUILD-INFO.txt` that records the commit of my CI repo it was built from, the vcpkg commit and triplet, the compiler, and the exact list of libraries to link.
- **Adapter:** the [`aravis-windows`](https://github.com/HazenBabcock/mmCoreAndDevices/tree/aravis-windows) branch now links those libraries statically and no longer has a `build.xml`. `mmgr_dal_AravisCamera.dll` is 4.7 MB and loads only Windows and Visual C++ runtime DLLs. CI builds it from `micromanager.sln`, checks its imports, loads it with pymmcore and snaps from Aravis's fake GigE camera ([run](https://github.com/HazenBabcock/mm-aravis-windows-ci/actions/runs/37535352197)). I haven't tried the Debug configuration.

One issue is that the static library can only be linked by a toolset at least as new as the compiler that built it, so these need MSVC 14.44 or newer to link the adapter. Which Visual Studio 2022 version do the Windows build machines have? The static libraries are built with v143 (MSVC 14.44, VS 2022 17.14).

On size: the static libraries come to 80 MB (26 MB zipped). Most of that is debug information that vcpkg embeds in release builds (about 80% of the libraries' code and data). If that is too much for 3rdpartypublic I can look at leaving it out.

If this looks right, could you commit the contents of the zip to 3rdpartypublic? Then I will open the PR for `aravis-windows`.

marktsuchida, 2026-10-08:
Wonderful. The build machine has an older version of 2022 right now, but I can update it. I think 80 MB is okay (was fearing much worse :). Not a huge fraction of the whole.

Please allow me a bit of time to update the MSVC version on the build machine (which currently requires manually updating a machine image). I'd like to combine that work with a few other pending updates to the build infrastructure that I was planning for this month.