micro-manager/mmCoreAndDevices #470: DahengGalaxy snap fails

marktsuchida, 2024-06-20:
The camera worked fine using the Live acquisition. However, the Snap button is greyed out in 'Live' mode, and when I tried to click on the 'Snap' button in the main interface, the following error appeared:

> Error: Failed to snap image.
> Camera image buffer read failed. 

It works fine when I use the demo camera. How should I resolve this issue? Thanks.

_Originally posted by @pchiang5 in https://github.com/micro-manager/mmCoreAndDevices/issues/455#issuecomment-2180263364_

nicost, 2024-06-24:
What is the reason to want to Snap while running live?  The Snap button is greyed out for a reason.  If you want a Snap, simply stop Live mode, and there is your latest "Snap".

pchiang5, 2024-06-24:
Thanks @nicost 

`If you want a Snap, simply stop Live mode, and there is your latest "Snap".` The error below jumped out if I clicked the latest `Snap` by your suggestion:

> Error: Failed to snap image.
> Camera image buffer read failed.

The temporary workaround was to replace the jre folder with a fresh installation (whenever I made changes to Hardware & Devices).