micro-manager/mmCoreAndDevices #481: Can’t locate mmgr_dal_NI-DAQmx.dll

JoeB-NIH, 2024-07-24:
Hi 

@fadero @nicost

we downloaded the latest release of MM v2 and could not find the adapter for NI-DAQmx in the wizard. the associated dll file is not found in the MM folder
[image]

nicost, 2024-07-24:
That is so weird!  I just installed the latest nightly build (20240714), and it is there.  Can you try that one?

[image]

nicost, 2024-07-25:
Oh, I get it.  You downloaded the latest "release".  You always want to download the latest nightly build from https://download.micro-manager.org/nightly/2.0/Windows/.  The latest "release" is about 3 years out of date.  Nightly builds are very stable, and have many features not found in the release.

closed 2024-07-25 by nicost

JoeB-NIH, 2024-07-25:
hey Nico

I will take a look at this tomorrow

appreciate the quick response

This is Joe Brzostowski btw from the NIH!

cheers

JoeB-NIH, 2024-07-25:
Nico

I just want to make sure that there isn’t a specific uninstall procedure
for uManager other than deleting the folder (I thought this all we have to
do)

yesterday when working with my IT person (I lost my admin privileges -
getting back hate the gov that way) we ran the the “unins000” but that
didn’t delete the uManager folder

just curious

joe

JoeB-NIH, 2024-07-25:
Nico

dont bother responding

figured out

the uninstall didn’t work yesterday because uManger was not in Programs
file

long story as to why

joe

nicost, 2024-07-25:
No worries Joe,

Micro-Manager does not install files outside of the MIcro-Manager folder
where it installs, so you can safely delete it (or rename, or whatever).  I
even often install into the same folder to upgrade to a newer version.
Usually there are no issues (only in rare cases where you also have to
install other files into the Micro-Manager folder to make things work).

Hope they will give you admin privileges again!

Best,

Nico